Edzelf/Esp-radio #82: Integrating IHeartRadio into ESP-Radio Arduino Code

NonaSuomy, 2017-05-02:
I started attempting to integrate IHeartRadio XML Parsing into ESP-Radio, but hit a bump in the Parsing code works outside of ESP-Radio but doesn't seem to be able to trigger mp3client inside ESP-Radio code base.

[Diff](https://www.diffchecker.com/hQtU7A7s) Line 1070 on the right of the Diff is where it tries to open the stream of data for XML content parsing.

XMLParsingTest.ino:
[GIST .ino](https://gist.github.com/NonaSuomy/3f35d98e9a0b5cb0054cec44923ee427)
Working...

Testing in ESP-Radio:
[https://github.com/NonaSuomy/Esp-radio/blob/NonaSuomy-XML-Parser/Esp_radio.ino](https://github.com/NonaSuomy/Esp-radio/blob/NonaSuomy-XML-Parser/Esp_radio.ino)

Gets stuck at line 1070
```
  if (mp3client.connect (ihrhost, xmlPort ) ) {
```

I tried to make a second client called xmlclient instead of mp3client and got the same results (nothing).

Any help would be appreciated!

NonaSuomy, 2017-05-03:
Enabled Debug Core+WiFi over serial and got some further information...

Connecting to url http://playerservices.streamtheworld.com:80/api/livestream?version=1.5&mount=CIMXFMAAC&lang=en
[hostByName] request IP for: playerservices.streamtheworld.com
[hostByName] Host: playerservices.streamtheworld.com lookup error: -5!
Not Connected to IHR
D: STOP requested
D: Song stopped correctly after 10 msec
D: Connect to new host 
D: Connect to  on port 80, extension /
[hostByName] request IP for: 
[hostByName] Host:  lookup error: -12!
D: Request  failed!

Edzelf, 2017-05-03:
Line 1070 is called during the "setup()" phase.  At that moment there is no Internet connection, so a DNS lookup will always fail.
The function "iheartradio" is called from "analyzeCmd".  Within "analyzeCmd" no timely actions are permitted as it will run mostly asynchronously to the main loop.

NonaSuomy, 2017-05-03:
Thanks for the tip!

[Show Diff](http://www.mergely.com/ay1A6BS8/)
[GitHub .ino ESP-Radio with XML-Parser](https://github.com/NonaSuomy/Esp-radio/blob/NonaSuomy-XML-Parser/Esp_radio.ino)

```
D: Command: xml with parameter IHR_TRAN
D: STOP requested
D: Stopping client
D: Song stopped correctly after 0 msec
D: Connect to new host: IHR_TRAN
D: Connecting to url: http://playerservices.streamtheworld.com:80/api/livestream?version=1.5&mount=IHR
[hostByName] request IP for: playerservices.streamtheworld.com
[hostByName] Host: playerservices.streamtheworld.com IP: 208.92.55.74
D: Connected!
D: XML parser processing...
D: Found: 18073.live.streamtheworld.com:3690/IHR_TRANAAC_SC
D: Closing XML connection.
D: Stopping client
D: Connect to new host 18073.live.streamtheworld.com:3690/IHR_TRANAAC_SC
D: Connect to 18073.live.streamtheworld.com on port 3690, extension /IHR_TRANAAC_SC
[hostByName] request IP for: 18073.live.streamtheworld.com
[hostByName] Host: 18073.live.streamtheworld.com IP: 208.92.55.40
D: Connected to server
D: Expires: Thu, 01 Dec 2003 16:00:00 GMT
D: Cache-Control: no-cache, must-revalidate
D: Pragma: no-cache
D: Access-Control-Allow-Methods: GET, HEAD
D: Access-Control-Allow-Origin: *
D: Access-Control-Allow-Credentials: true
D: Set-Cookie: uuid=e42b6045-278b-4d4d-9b9d-db3cc87d4921; expires=Sat, 22-Dec-2029 21:30:41 GMT; path=
D: Content-Type: audio/aacp
D: Connection: close
D: icy-br: 48
D: icy-description: Non-stop Trance and Progressive 24/7
D: icy-genre: 
D: icy-name: iHeart Trancid
D: icy-url: 
D: icy-metaint: 16000
D: Server: MediaGateway 4.2.5-003
D: Switch to DATA, bitrate is 48
D: First chunk:
D: FF F1 5E 40 23 FF FC 01
D: 44 36 2C 91 56 54 09 44
D: 82 11 9D F0 3B 8D 4A AB
D: 90 32 CC 91 08 05 DE 4A
D: Metadata block 48 bytes
D: Streamtitle found, 34 bytes
D: StreamTitle='Tiesto - Flight 643';
D: Command: xml with parameter CIMXFM
D: STOP requested
D: Stopping client
D: Song stopped correctly after 10 msec
D: Connect to new host: CIMXFM
D: Connecting to url: http://playerservices.streamtheworld.com:80/api/livestream?version=1.5&mount=CIM
[hostByName] request IP for: playerservices.streamtheworld.com
[hostByName] Host: playerservices.streamtheworld.com IP: 208.92.55.74
D: Connected!
D: XML parser processing...
D: Found: 14163.live.streamtheworld.com:3690/CIMXFMAAC_SC
D: Closing XML connection.
D: Stopping client
D: Connect to new host 14163.live.streamtheworld.com:3690/CIMXFMAAC_SC
D: Connect to 14163.live.streamtheworld.com on port 3690, extension /CIMXFMAAC_SC
[hostByName] request IP for: 14163.live.streamtheworld.com
[hostByName] Host: 14163.live.streamtheworld.com IP: 208.80.52.67
D: Connected to server
D: Expires: Thu, 01 Dec 2003 16:00:00 GMT
D: Cache-Control: no-cache, must-revalidate
D: Pragma: no-cache
D: Access-Control-Allow-Methods: GET, HEAD
D: Access-Control-Allow-Origin: *
D: Access-Control-Allow-Credentials: true
D: Set-Cookie: uuid=771e18c8-824a-4751-97d5-f59143a1e91c; expires=Sat, 22-Dec-2029 21:30:41 GMT; path=
D: Content-Type: audio/aacp
D: Connection: close
D: icy-br: 40
D: icy-description: 
D: icy-genre: Rock
D: icy-name: 
D: icy-url: http://www.89xradio.com
D: icy-metaint: 16000
D: Server: MediaGateway 4.2.5-003
D: Switch to DATA, bitrate is 40
D: First chunk:
D: FF F1 5E 40 1D 7F FC 01
D: 2E 15 20 80 18 28 06 0A
D: B2 41 B0 4C 8F 97 AF A6
D: BE 5E C6 75 BF 1E 7E 78
D: Metadata block 48 bytes
D: Streamtitle found, 33 bytes
D: StreamTitle='Muse - Dead Inside';
```
:tada: 

Does it look copacetic?

NonaSuomy, 2017-05-03:
Added a channel switcher Test to the HTML code: [index_html.h](https://github.com/NonaSuomy/Esp-radio/blob/NonaSuomy-XML-Parser/index_html.h) 
I'll try to figure out adding them to the radio.ini and load them from there later hopefully! :smile_cat:

Edzelf, 2017-05-04:
Looks good!  I'm including this feature in my version.  Testing now.

Edzelf, 2017-05-04:
iHeartRadio is implemented now.
mountpoints can be specified like normal presets, just prefix it with "ihr/", see documentation.
Is there a list somewhere of working iHeartRadio stations?

NonaSuomy, 2017-05-04:
Great work with the prefix, I'll give it a test when I get home.

Just to note I forgot to add XML to analyze cmd comment section inside the code I noticed it is in the PDF but not in the source. [.ino source](https://github.com/Edzelf/Esp-radio/blob/master/Esp_radio.ino#L2647)
I would add the line url from the documentation but it's in the PDF format and Github doesn't support that nicely.

I was trying to figure out a way to pull a list from their site but the API is undocumented, my priority was to get it playing first.

There are some tips here on how to find callsigns from the HTML source on the website: [issues/70](https://github.com/Edzelf/Esp-radio/issues/70)

Essentially the live radio stations are usually in this format XXXXFM for an FM broadcast or XXXXAM for an AM broadcast, so you could look up the callsign letters of a station online and then check to see if IHeartRadio manages their stream then just type ihr/CIMXFM etc, the IHR_ streams are outside of that normality as they only exist on their service. I found those by playing the station on their website then clicking the history option in their player than looking at the source of that page with the log of the songs they last played on it. usually, you can just search for the word "callsign" in the source and you should find a tag with it inside.

You can read their history to find out most of the stations they support in the [wikipedia iHeartRadio](https://en.wikipedia.org/wiki/IHeartRadio)

NonaSuomy, 2017-05-04:
Updated the [README.md ](https://github.com/NonaSuomy/Esp-radio/blob/master/README.md) -
 [Change](https://github.com/NonaSuomy/Esp-radio/commit/41d480abb426c1eb5e7ad0359a0e794f4c9c42f1#diff-04c6e90faac2675aa89e2176d2eec7d8)

koskee, 2017-06-12:
@NonaSuomy would you by any chance have a tiny bit of free time to help me figure out why I can't get any of the IHR stations working? Edzelf has already implied that he will be busy for the next week(s) and won't be available to do so in that time.

 Having these stations functioning is one of the main points of leverage I will have to utilize to get my boss and coworkers to agree to evaluate the internet radio concept to be adopted in the shop where I work, so it is somewhat vital that I am able to get to the bottom of why they aren't playing properly. (the local station that they are stuck on just so happens to be an ihr station) 

Any help would be appreciated, and I figure you're probably the only other person that understands the code well enough to help me work through the  issue. I am not sure if I'm doing something wrong, or if there's a bug or what, but i'm really at a loss when it comes to what approach to take next in resolving this. 

I opened an issue under the esp32 version that explains where I'm at, and referenced it above. As far as I am aware, the code for both versions should be pretty much the same, although I haven't done a side by side comparison to legitimately make this claim. 
Regardless, I believe you wrote (at least) the original code for the implementation, so If youd be so gracious as to go over it and put in your two cents on my problem, It'd be very much appreciated. 

Let me know if there's anything I can do to make it easier on you as well.. 

Thanks in advance, 
Koskee

NonaSuomy, 2017-06-13:
The callsign is indeed ihr/CHBDFM case sensitive.

Are you able to stream any of the shoutcast stations ip:port/mp3 etc ?

koskee, 2017-06-13:
Umm. I am uncertain. What I do recall is that it seemed that in addition to ihr not working, that any of the example stations that were addressed with a raw ip address were also not working, so I replaced them in the preset list almost immediately. I was unaware that these stations had anything in common (ie. Shoutcast) since they had no common domain name and hence didn't seem related. 
I had assumed they were just random IPs from around the internet that had been broadcasting at one point or another, but had since shut down. 

So those were Shoutcast then? They didn't work before, but it's been a long time since I've attempted to use them. Would it help if I gave them a go?

koskee, 2017-06-14:
Nevermind, I guess some of those IP addressed stations do work.

On another note, here is the debug output I'm getting from the `ihr/IHR_IEDM` station. Notice how by almost any measure, it appears to be working, yet for whatever reason, there is absolutely no audio output.

```
`rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:160
load:0x40078000,len:10632
load:0x40080000,len:252
entry 0x40080034

D: Starting ESP32-radio Version Wed, 31 May 2017 12:05:00 GMT...  Free memory 194672
D: GPIO0 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO17 is HIGH
D: GPIO22 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW
D: GPIO35 is LOW
D: GPIO0 will execute uppreset = 1
D: GPIO12 will execute upvolume = 2
D: GPIO13 will execute downvolume = 2
D: Added SSID Excide to list of networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 4
D:  1 - Excide                    Signal: -41 dBm, Encryption WPA2_PSK, Acceptable
D:  2 - 2WIRE891a                 Signal: -45 dBm, Encryption WPA2_PSK, 
D:  3 - SASKTEL848                Signal: -91 dBm, Encryption WPA_WPA2_PSK, 
D:  4 - JCW WiFi                  Signal: -93 dBm, Encryption WPA2_PSK, 
D: End of list
D: Command: mqttbroker with parameter none
D: Command: mqttport with parameter 1883
D: Command: mqttuser with parameter none
D: Command: mqttpasswd with parameter ******
D: Command: volume with parameter 100
D: Command: toneha with parameter 2
D: Command: tonehf with parameter 6
D: Command: tonela with parameter 3
D: Command: tonelf with parameter 10
D: Command: preset with parameter 10
D: Slow SPI,Testing VS1053 read/write registers...
D: Fast SPI, Testing VS1053 read/write registers again...
D: endFillByte is 28
D: Connect to WiFi
D: Connecting to access point: Excide
D: Connected to Excide
D: IP = 192.168.1.10
D: Start server for commands
D: SD Card Mount Failed!
D: New preset/file requested (10/0) from ihr/IHR_IEDM
D: Connect to new iHeartRadio host: IHR_IEDM
D: GET /api/livestream?version=1.5&mount=IHR_IEDMAAC&lang=en
D: Connected!
D: XML parser processing...
D: Found: 15323.live.streamtheworld.com:3690/IHR_IEDMAAC_SC
D: Closing XML connection.
D: Stopping client
D: Connect to new host 15323.live.streamtheworld.com:3690/IHR_IEDMAAC_SC
D: Connect to 15323.live.streamtheworld.com on port 3690, extension /IHR_IEDMAAC_SC
D: Connected to server
D: Switch to HEADER
D: Expires: Thu, 01 Dec 2003 16:00:00 GMT
D: Cache-Control: no-cache, must-revalidate
D: Pragma: no-cache
D: Access-Control-Allow-Methods: GET, HEAD
D: Access-Control-Allow-Origin: *
D: Access-Control-Allow-Credentials: true
D: Set-Cookie: uuid=448d9fba-2137-4cca-8743-3b02464eb272; expires=Sat, 22-Dec-2029 21:30:41 GMT; path=/; domain=.live.streamtheworld.com
D: Content-Type: audio/aacp
D: audio/aacp seen.
D: Connection: close
D: icy-br: 48
D: icy-description: All Things Dance
D: icy-genre: 
D: icy-name: Evolution
D: icy-url: 
D: icy-metaint: 16000
D: Server: MediaGateway 4.2.6-001
D: Switch to DATA, bitrate is 48, metaint is 16000
D: First chunk:
D: FF F1 5E 40 22 BF FC 01
D: 52 36 2A 0C C5 22 0D 82
D: 82 50 A1 04 22 EE CC D0
D: C5 37 57 8A 01 12 A4 02
D: Metadata block 64 bytes
D: Streamtitle found, 60 bytes
D: StreamTitle='Yellow Claw / Dj Snake / Elliphant - Good Day';
D: Metadata block 64 bytes
D: Streamtitle found, 48 bytes
D: StreamTitle='David Guetta / Justin Bieber - 2u';`
```

Edzelf, 2017-06-14:
Contents type is audio/aacp.  Shouldn't that bee audio/mp3?  Please compare it with a working station.  I am not able to test it here.

koskee, 2017-06-16:
Sorry, spent the last couple evenings catching up on some other stuff and eradicating a virus/worm that snuck onto my personal network. All is back to normal now. I will see what further data I can dig up later tonight. 

So,  it may be deciphering the wrong content type, does that mean these mp3 boards do not support aac, or would it be misinterpreting the data stream? 
It seems odd that the stream works for others but not on mine, which (presumably)  has the same or nearly the same firmware. 

Not sure if anyone else noticed the extra leading dot (.) near the end of this line, would that cause it to parse a poorly formatted url? :
`D: Set-Cookie: uuid=448d9fba-2137-4cca-8743-3b02464eb272; expires=Sat, 22-Dec-2029 21:30:41 GMT; path=/; domain=.live.streamtheworld.com`

NonaSuomy, 2017-06-18:
Nope AAC is correct and these boards support it. It defiantly looks like it`s working, just seems like the volumes turned down though it says 100 in the log maybe make it play a shoutcast station then play the ihr stream.

koskee, 2017-06-18:
I did a bit of further poking around, it seems that any mpeg stream is working (fantastically I might add) and anything that comes up with a content type of aac(p) refuses to play. 

I found this (Dutch?) webstore that sells them: http://shop.egnite.de/de/embedded-entwicklung/audio-ics/66/vlsi-vs1053b-l?c=127

After auto-translation, it says: 

> A license for the MPEG Layer-3 decoder from Technicolor is included in the price. For a WMA licensing, please contact Microsoft directly. Customers who wish to use AAC must purchase a license from Via Licensing.

The datasheet also talks about something about being able to have an embedded serial number from the factory for "digital rights management" purposes. Could it be that the board I have is not properly licenced for aac decoding? (After all it did come from China so I would not be surprised if it was either a counterfeit board or a slimmed down version that did not have full features.. Which  the sellers opted for as a penny pinching tactic)

NonaSuomy, 2017-06-18:
Unusual maybe try a sketch that only plays audio files from sd

I got mine from china ebay sellers http://www.ebay.ca/itm/VS1053-MP3-Module-with-SD-card-slot-VS1053B-Ogg-real-time-recording-for-Arduino-/262866917974?hash=item3d3416da56:g:QysAAOSwdGFYrwHo

maybe raw chips are not setup properly like the breakout boards?

NonaSuomy, 2017-06-18:
Here's the datasheet in english https://www.sparkfun.com/datasheets/Components/SMD/vs1053.pdf

Edzelf, 2017-06-18:
I used this one: [https://www.aliexpress.com/item/1PCS-New-VS1053-MP3-Module-Development-Bard-on-board-recording-function/32354143509.html?spm=2114.13010608.0.0.OFjB1u](url)

koskee, 2017-06-19:
Well, that's a bit frustrating, but I guess that would make sense - I had a closer look and figured out that the board I have has a VS1003B chip instead of a VS1053B, despite being sold as the latter. The datasheet for the 1003 makes no mention of the AAC format on first glance. 

Guess I have to wait another 2 months for a new board to arrive >:( 
It's really lucky that I hadn't started fab/etching the custom board I had designed for this. I had just finalized the layout yesterday, but the new board looks like it has a different pinout so I suppose some revisions are in order. 

I'll close the issue on the other side, but this one doesn't belong to me. Thanks for your help everyone @Edzelf @NonaSuomy

NonaSuomy, 2017-06-23:
Glad I could help you solved the mystery!

Just to note I would double check the pin layout as I believe a bunch of the VS series are drop in replacements for each other without too much fuss.

koskee, 2017-06-24:
I was hoping they would be, although based on the pictures of the module (bought the one recommended by @Edzelf)  it appears that the pins are arranged in a double row pattern instead of the single row of 9 pins on the one I have. Obviously a different design, but I wanted to be sure that I got a genuine module this time, to avoid waiting additional months to get it all built. 

Since I already designed for this board (and since they obviously do make 1053 versions with the same pinout - i'm guessing because this  board is labeled vs1003/VS1053 breakout or something like that) I will likely try to incorporate headers for the new board into the design I already made for dual compatibility. I'm trying to avoid making it a dual layer design for simplicity's sake so I guess we shall see. 

On that note, can anyone confirm that the pinout is 2 x __ on those modules? Also if somebody has a caliper and one of those boards and doesn't mind taking a few measurements for me, it would save me a good chunk of time not having to wait for the new one to arrive. (or a datasheet would probably suffice if it includes a diagram with measurements)

NonaSuomy, 2017-06-24:
Maybe you could modify this board https://github.com/dsaltas/WiFi-WebRadio

koskee, 2017-06-25:
Oh I guess i should have been more clear, I would need a datasheet showing the measurements of this module (ie. the pin spacing and relative localisation between the board edges and the aforementioned pins):
 (https://www.aliexpress.com/item/1PCS-New-VS1053-MP3-Module-Development-Bard-on-board-recording-function/32354143509.html?spm=2114.13010608.0.0.OFjB1u)
I'd ask the seller, but in my experience they often don't have any usable information on the products they sell. 

> 
> Me: can you send me the datasheet (like it says you have available on the item listing?) 
> AsiaHappy4USellnow: dear sir, we are sorry you haven't received our package. Most apology. Patient is a kind lamb... 
> Me: no no, the "datasheet" - information, manual, documentation... 
> AsiaHappy4USellnow: dear sir, we are sorry you haven't received our package. Most apology. Patient is a kind lamb. Can you provide your full name & address for me checking? 
> Me: I don't think you understand me.  DA-TA-SHEET. Comprehend svp? 
> AsiaHappy4USellnow: dear sir, we are sorry you haven't received our package. You want refund or resend? Hope for your soon reply sir. 
> Me: *facepalm*

(this actually happened to me pretty recently)

Mantikor, 2017-06-25:
**koskee**, datasheet for this module is unlikely to exist in nature, probably you should guess to type in google.com datasheet vs1053, and not to laugh for Chinese, who does not know English and responds with a learned phrase.

koskee, 2017-06-25:
Was merely an example to illustrate the fact that the sellers often cannot communicate on a functional level. If anything,  I'd say I'm frustrated with the fact that I bought a VS1053 module and got something different. Any laughter would be because there's not much else I can do about it. Just have to realize,  that's how it goes sometimes, as irritating as it is waiting 3 months for stuff to arrive.  

I agree that such a datasheet is unlikely to find its way into my hands, if it exists. That is why I am asking for someone to measure theirs with a caliper, if they could as it would save a lot of time and hassle searching for information that may or may not exist..