Edzelf/Esp-radio #83: Long time on

udavst, 2017-05-05:
After power on, immediately connects to Wifi, and does not respond 1-5 minutes, neither on ping, nor on the web, does not play. In the logs - connected to WIFI (last line), to buttons (A0) reacts - outputs to log. After 1-5 minutes, it starts to play itself and everything starts to work quickly and stably. Because of what there can be such a delay at turning on?

P.S. The version without LCD, ESP - Wemos D1 mini is used.

Edzelf, 2017-05-05:
There is no logging for "connected to Wifi".  The log should show something like:
D: Selected network: XXXXX               
D: Try WiFi XXXXXX
D: IP = 192.168.2.9
D: Start server for commands.
The radio tries to connect to the preset defined by your radio.ini file.  If the first station does not react it will try the next one and so on.  This may take some time.  But all of this must be visible in the logging.

udavst, 2017-05-05:
Yes, for a very long time it hangs on Try WiFi xxxxxxx. On the router it is already connected, and the address is issued, but ESP-Radio continues to wait for something in this place.

Edzelf, 2017-05-05:
Show me the complete logging

udavst, 2017-05-05:
`D: endFillByte is 0
D: Selected network: udavst@gmail.com         
D: Try WiFi udavst@gmail.com
D: FS Total 2949250, used 2008
D: /radio.ini                       -    1516
D: Added SSID 00 = udavst@gmail.com to acceptable networks
D: Added SSID 01 = udavst@gmail.com_5g to acceptable networks
D: * Scan Networks *
D: Number of available networks: 13
D:  1 - ARMENIA.AM                Signal: -70 dBm Encryption WPA2  
D:  2 - MTSRouter_83B0E6          Signal: -60 dBm Encryption WPA2  
D:  3 - zakroma                   Signal: -86 dBm Encryption WPA2  
D:  4 - UPVEL                     Signal: -92 dBm Encryption Auto  
D:  5 - Home-40                   Signal: -91 dBm Encryption Auto  
D:  6 - MTSRouter_8423A6          Signal: -69 dBm Encryption WPA2  
D:  7 - Home_Tatiana              Signal: -66 dBm Encryption WPA2  
D:  8 - MTSRouter-010979          Signal: -70 dBm Encryption WPA2  
D:  9 - MTSRouter_84D8C2          Signal: -86 dBm Encryption WPA2  
D: 10 - Spark-Home                Signal: -88 dBm Encryption Auto  
D: 11 - udavst@gmail.com          Signal: -64 dBm Encryption WPA2  Acceptable
D: 12 - BelMet                    Signal: -83 dBm Encryption None  
D: 13 - MTSRouter-237CE2          Signal: -60 dBm Encryption WPA2  
D: --------------------------------------
D: Command: mqttbroker with parameter broker.hivemq.com
D: Command: mqttport with parameter 1883
D: Command: mqttuser with parameter No
D: Command: mqttpasswd with parameter No
D: Command: mqtttopic with parameter espradio
D: Command: mqttpubtopic with parameter espradioIP
D: Command: wifi_00 with parameter udavst@gmail.com/00000000oo
D: Command: wifi_01 with parameter udavst@gmail.com_5g/00000000oo
D: Command: volume with parameter 82
D: Command: toneha with parameter 0
D: Command: tonehf with parameter 0
D: Command: tonela with parameter 5
D: Command: tonelf with parameter 12
D: Command: preset with parameter 2
D: Command: preset_00 with parameter 109.206.96.34:8100
D: Command: preset_01 with parameter ep128.streamr.ru
D: Command: preset_02 with parameter s2.radioheart.ru:8005/nonstop
D: Command: preset_03 with parameter 192.240.102.3:9042/stream
D: Command: preset_04 with parameter 81.30.54.74:8000/radio3
D: Command: preset_05 with parameter stream3.radiostyle.ru:8003/radioacca
D: Command: preset_06 with parameter s07.radio-tochka.com:6455/
D: Command: preset_07 with parameter s7.radioheart.ru:8003/live
D: Command: preset_08 with parameter 16bit.fm/play/16bit.fm_cafe_128.m3u
D: Command: preset_09 with parameter 94.23.66.155:8106
D: Command: preset_10 with parameter ic3.101.ru:8000/c10_4?setst=-1
D: Command: preset_11 with parameter radio.mixadance.fm:8000/relax
D: Command: preset_12 with parameter beloeradio.ru:8080/whiteradio.mp3
D: Starting ESP Version Wed, 04 May 2017 10:45:00 GMT...  Free memory 13696
D: Sketch size 341056, free size 704512
D: Reset VS1053...
D: End reset VS1053...
D: Slow SPI,Testing VS1053 read/write registers...
D: Fast SPI, Testing VS1053 read/write registers again...
D: endFillByte is 0
D: Selected network: udavst@gmail.com         
D: Try WiFi udavst@gmail.com
**DELAY**
D: IP = 192.168.1.11
D: Start server for commands
D: Connecting to MQTT broker.hivemq.com, port 1883, user No, password No...
D: MQTT Connected to the broker broker.hivemq.com, session is not present
D: Subscribing to espradio at QoS 2, packetId = 1 
D: Publishing IP 192.168.1.11 to topic espradioIP
D: MQTT Subscribe acknowledged, packetId = 1, QoS = 2
D: MQTT Publish acknowledged, packetId = 2
D: STOP requested
D: Song stopped correctly after 0 msec
D: New preset/file requested (-1/0) from s2.radioheart.ru:8005/nonstop
D: Connect to new host s2.radioheart.ru:8005/nonstop
D: Connect to s2.radioheart.ru on port 8005, extension /nonstop
D: Connected to server
D: Accept-Ranges: none
D: Content-Type: audio/mpeg
D: icy-br:128
D: ice-audio-info: bitrate=128
D: icy-description:NONSTOP
D: icy-genre:Club
D: icy-name:Index_Radio
D: icy-pub:0
D: icy-url:http://club_inex.myrh.ru/
D: Server: Icecast 2.3.3-kh11
D: Cache-Control: no-cache, no-store
D: Pragma: no-cache
D: Access-Control-Allow-Origin: *
D: Access-Control-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type
D: Access-Control-Allow-Methods: GET, OPTIONS, HEAD
D: Connection: close
D: Expires: Mon, 26 Jul 1997 05:00:00 GMT
D: icy-metaint:16000
D: Switch to DATA, bitrate is 128
D: First chunk:
D: FF FB 90 64 EC 80 02 C5
D: 25 DB 69 E7 1D 20 47 C2
D: 2B AF 24 E9 65 0C 24 95
D: 73 E7 A4 4F 41 60 8C 6D
D: Metadata block 48 bytes
D: Streamtitle found, 39 bytes
D: StreamTitle='Q-Q Biko - Picture Of Me';
D: list request from browser
`

udavst, 2017-05-05:
D: endFillByte is 0
D: Selected network: udavst@gmail.com         
D: Try WiFi udavst@gmail.com
D: FS Total 2949250, used 2008
D: /radio.ini                       -    1516
D: Added SSID 00 = udavst@gmail.com to acceptable networks
D: Added SSID 01 = udavst@gmail.com_5g to acceptable networks
D: * Scan Networks *
D: Number of available networks: 13
D:  1 - ARMENIA.AM                Signal: -70 dBm Encryption WPA2  
D:  2 - MTSRouter_83B0E6          Signal: -60 dBm Encryption WPA2  
D:  3 - zakroma                   Signal: -86 dBm Encryption WPA2  
D:  4 - UPVEL                     Signal: -92 dBm Encryption Auto  
D:  5 - Home-40                   Signal: -91 dBm Encryption Auto  
D:  6 - MTSRouter_8423A6          Signal: -69 dBm Encryption WPA2  
D:  7 - Home_Tatiana              Signal: -66 dBm Encryption WPA2  
D:  8 - MTSRouter-010979          Signal: -70 dBm Encryption WPA2  
D:  9 - MTSRouter_84D8C2          Signal: -86 dBm Encryption WPA2  
D: 10 - Spark-Home                Signal: -88 dBm Encryption Auto  
D: 11 - udavst@gmail.com          Signal: -64 dBm Encryption WPA2  Acceptable
D: 12 - BelMet                    Signal: -83 dBm Encryption None  
D: 13 - MTSRouter-237CE2          Signal: -60 dBm Encryption WPA2  
D: --------------------------------------
D: Command: mqttbroker with parameter broker.hivemq.com
D: Command: mqttport with parameter 1883
D: Command: mqttuser with parameter No
D: Command: mqttpasswd with parameter No
D: Command: mqtttopic with parameter espradio
D: Command: mqttpubtopic with parameter espradioIP
D: Command: wifi_00 with parameter udavst@gmail.com/00000000
D: Command: wifi_01 with parameter udavst@gmail.com_5g/00000000
D: Command: volume with parameter 82
D: Command: toneha with parameter 0
D: Command: tonehf with parameter 0
D: Command: tonela with parameter 5
D: Command: tonelf with parameter 12
D: Command: preset with parameter 2
D: Command: preset_00 with parameter 109.206.96.34:8100
D: Command: preset_01 with parameter ep128.streamr.ru
D: Command: preset_02 with parameter s2.radioheart.ru:8005/nonstop
D: Command: preset_03 with parameter 192.240.102.3:9042/stream
D: Command: preset_04 with parameter 81.30.54.74:8000/radio3
D: Command: preset_05 with parameter stream3.radiostyle.ru:8003/radioacca
D: Command: preset_06 with parameter s07.radio-tochka.com:6455/
D: Command: preset_07 with parameter s7.radioheart.ru:8003/live
D: Command: preset_08 with parameter 16bit.fm/play/16bit.fm_cafe_128.m3u
D: Command: preset_09 with parameter 94.23.66.155:8106
D: Command: preset_10 with parameter ic3.101.ru:8000/c10_4?setst=-1
D: Command: preset_11 with parameter radio.mixadance.fm:8000/relax
D: Command: preset_12 with parameter beloeradio.ru:8080/whiteradio.mp3
D: Starting ESP Version Wed, 04 May 2017 10:45:00 GMT...  Free memory 13696
D: Sketch size 341056, free size 704512
D: Reset VS1053...
D: End reset VS1053...
D: Slow SPI,Testing VS1053 read/write registers...
D: Fast SPI, Testing VS1053 read/write registers again...
D: endFillByte is 0
D: Selected network: udavst@gmail.com         
D: Try WiFi udavst@gmail.com
**DELAY**
D: IP = 192.168.1.11
D: Start server for commands
D: Connecting to MQTT broker.hivemq.com, port 1883, user No, password No...
D: MQTT Connected to the broker broker.hivemq.com, session is not present
D: Subscribing to espradio at QoS 2, packetId = 1 
D: Publishing IP 192.168.1.11 to topic espradioIP
D: MQTT Subscribe acknowledged, packetId = 1, QoS = 2
D: MQTT Publish acknowledged, packetId = 2
D: STOP requested
D: Song stopped correctly after 0 msec
D: New preset/file requested (-1/0) from s2.radioheart.ru:8005/nonstop
D: Connect to new host s2.radioheart.ru:8005/nonstop
D: Connect to s2.radioheart.ru on port 8005, extension /nonstop
D: Connected to server
D: Accept-Ranges: none
D: Content-Type: audio/mpeg
D: icy-br:128
D: ice-audio-info: bitrate=128
D: icy-description:NONSTOP
D: icy-genre:Club
D: icy-name:Index_Radio
D: icy-pub:0
D: icy-url:http://club_inex.myrh.ru/
D: Server: Icecast 2.3.3-kh11
D: Cache-Control: no-cache, no-store
D: Pragma: no-cache
D: Access-Control-Allow-Origin: *
D: Access-Control-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type
D: Access-Control-Allow-Methods: GET, OPTIONS, HEAD
D: Connection: close
D: Expires: Mon, 26 Jul 1997 05:00:00 GMT
D: icy-metaint:16000
D: Switch to DATA, bitrate is 128
D: First chunk:
D: FF FB 90 64 EC 80 02 C5
D: 25 DB 69 E7 1D 20 47 C2
D: 2B AF 24 E9 65 0C 24 95
D: 73 E7 A4 4F 41 60 8C 6D
D: Metadata block 48 bytes
D: Streamtitle found, 39 bytes
D: StreamTitle='Q-Q Biko - Picture Of Me';
D: list request from browser

udavst, 2017-05-05:
The router issues the address immediately, but esp is still waiting for something. Tried another ESP, the same behavior. Another ESP is working on sending the weather - it does not have problems with the connection.

thermo2016, 2017-05-05:
Could it be this?:
D: Added SSID 01 = udavst@gmail.com_5g to acceptable networks
that is a 5g wifi but the esp is 2.4ghz only. 
If the 5g is a stronger signal it will try this first. 
Try removing the reference to the 5g wifi.

udavst, 2017-05-05:
ESP does not see the same 5g networks, I thought it did not matter what I put in there. I removed the second network completely, and switched the router to 20MHz (it was 20/40 auto). Everything began to work perfectly. Thank you!

closed 2017-05-23 by Edzelf