Edzelf/Esp-radio #85: abs() avoid doing work inside brackets?

NonaSuomy, 2017-05-09:
[ref](https://www.arduino.cc/en/reference/abs)
```
Warning

Because of the way the abs() function is implemented, avoid using other functions inside the brackets, it may lead to incorrect results.

abs(a++);   // avoid this - yields incorrect results

abs(a); 
a++;          // use this instead - keep other math outside the function
```

Original
```
//******************************************************************************************
//                                  A N A G E T S W                                        *
//******************************************************************************************
// Translate analog input to switch number.  0 is inactive.                                *
//******************************************************************************************
uint8_t anagetsw ( uint16_t v )
{
  int      i ;                                    // Loop control
  int      oldmindist = 1000 ;                    // Detection least difference
  int      newdist ;                              // New found difference
  uint8_t  sw = 0 ;                               // Number of switch detected (0 or 1..3)

  if ( v > analogrest )                           // Inactive level?
  {
    for ( i = 0 ; i < NUMANA ; i++ )
    {
      newdist = abs ( analogsw[i] - v ) ;          // Compute difference
      
      if ( newdist < oldmindist )                  // New least difference?
      {
        oldmindist = newdist ;                     // Yes, remember
        sw = i + 1 ;                               // Remember switch
      }
    }
  }
  return sw ;                                      // Return active switch
}
```

Should it be something like this instead?
```
//******************************************************************************************
//                                  A N A G E T S W                                        *
//******************************************************************************************
// Translate analog input to switch number.  0 is inactive.                                *
//******************************************************************************************
uint8_t anagetsw ( uint16_t v )
{
  int      i ;                                    // Loop control
  int      aval ;                                    // Switch value - Analog value
  int      oldmindist = 1000 ;                    // Detection least difference
  int      newdist ;                              // New found difference
  uint8_t  sw = 0 ;                               // Number of switch detected (0 or 1..3)

  if ( v > analogrest )                           // Inactive level?
  {
    for ( i = 0 ; i < NUMANA ; i++ )
    {
      aval = analogsw[i] - v
      newdist = abs ( aval ) ;          // Compute difference
      
      if ( newdist < oldmindist )                  // New least difference?
      {
        oldmindist = newdist ;                     // Yes, remember
        sw = i + 1 ;                               // Remember switch
      }
    }
  }
  return sw ;                                      // Return active switch
}
```

NonaSuomy, 2017-05-09:
Changing this seemed to fix my analog button issue of it detecting wrong button presses.

Edzelf, 2017-05-09:
Fixed.

NonaSuomy, 2017-05-09:
Thanks @Edzelf

closed 2017-05-09 by NonaSuomy