Edzelf/Esp-radio #87: ESP-Radio MQTT seems to lose connection with broker?

NonaSuomy, 2017-05-10:
Having some issue right now, where the esp seems to lose its connection with the broker and then will never send update information to it again. 

On first boot I see in MQTTLens, I'm subscribed to espradio/# and I see: 
```
# Time      Topic              QoS
1 9:51:53  espradio/ip     0
Message: 10.13.37.65

# Time      Topic                            QoS
2 9:51:57  espradio/nowplaying    0
Message: LOUNGE-RADIO.COM - St. Germain - Easy To Remember
```
It will keep updating espradio/nowplaying for a while, and then randomly stop updating.

Last, publish before connection lost:
```
Metadata block 48 bytes
D: Streamtitle found, 41 bytes
D: StreamTitle='Zedd / Alessia Cara - Stay';
D: MQTT Publish acknowledged, packetId = 33
etc...
```

Connection Lost:
```
D: Streamtitle found, 39 bytes
D: StreamTitle='Twenty One Pilots - Ride';
D: MQTT Publish acknowledged, packetId = 34
D: MQTT Disconnected from the broker, reason 0, reconnecting...
D: No data input
D: Recovered from dataloss
D: No data input
D: No data input
D: Trying other station/file...
D: STOP requested
```

After connection lost, still sending messages to topics but alas no connection so the go to limbo.
```
D: 10 minutes over
D: Publishing IP 10.13.37.65 to topic espradio
D: 10 minutes over
D: Publishing IP 10.13.37.65 to topic espradio
D: 10 minutes over
D: Publishing IP 10.13.37.65 to topic espradio
D: 10 minutes over
D: Publishing IP 10.13.37.65 to topic espradio

```

I first thought it may be the broker but it's online and I can send messages to topics fine:

```mosquitto_pub -h 10.13.37.42 -u <username> -P <password> -t espradio/test -m  40```

```
# Time      Topic              QoS
3 9:39:57  espradio/test  0
Message: 40
```

Wondering if it breaks on sending an invalid character or something to that nature, and then never tries to reconnect again, been testing it and it seems like if null values are sent it kills the ESP-Radio MQTT connection maybe...

I think to fix the publishing with no connection you can do a
```if ( mqttclient.connected() ) { ... }```

and then in the 10 min check do a
```if ( !mqttclient.connected() ) { mqttclient.connect(); }```

Though the library seems like it supposed to keep retrying on disconnect.

hmm...
Added some debug seems it caught one nothing value then it got a message of 40 for bitrate and that killed it added a message for the connection tries and it seems to only try once.

```
D: Streamtitle found, 43 bytes
D: StreamTitle='Bleeker - Where's Your Money';
D: Error: Tried to publish null message  to topic espradio/icy/name
D: Publishing message Bleeker - Where's Your Money to topic espradio/icy/streamtitle
D: Publishing message 40 to topic espradio/icy/bitrate
D: MQTT Disconnected from the broker, reason 0, reconnecting...
D: MQTT reconnection attempt: 1
```

Seems the debug code caught the discconect as well no IP got published after 10 mins because of the client crashed message:

```
D: StreamTitle='Positively Fourth Street - Bob Dylan ';StreamUrl='';
D: 10 minutes over
D: Metadata block 64 bytes
```

Here's the code so far [GIST](https://gist.github.com/NonaSuomy/e50cfdf079fe1bbbcac30a381e34fd93/fbedc19cca5332bbced38ecdfa8af6793fe27de4)

This is what makes stuff like [OpenHAB ESP-Radio Control](https://github.com/Edzelf/Esp-radio/issues/77) possible.

NonaSuomy, 2017-05-10:
Guessing that  ```';StreamUrl='';``` bit off more than it could chew from the icy header.

NonaSuomy, 2017-05-10:
We might want to check for a wifi connection as well before reconnecting to the MQTT broker.

Edzelf, 2017-05-10:
If the wifi-connection fails, the ESP will reset after some time.

NonaSuomy, 2017-05-10:
Ok well, I guess I know that it's not disconnecting from wifi, just getting disconnected from the MQTT broker from a bad message sent as it's still playing music just no longer updating the broker with messages.

Edzelf, 2017-05-10:
Maybe it is a bad idea to send MQTT messages asynchronously.  It is called from a Ticker event.
Better to set a flag and data to be sent and handle it within the "loop()".

NonaSuomy, 2017-05-10:
That's what I was doing here: [Line:1178](https://gist.github.com/NonaSuomy/e50cfdf079fe1bbbcac30a381e34fd93/revisions?diff=unified#diff-3a95bed2f96cf5026c57bb207a78cf1eR1178) alas I think it has to do with sending some bad data to the broker that causes the disconnect, what do you think?

NonaSuomy, 2017-05-11:
So even though ```sprintf ( message, "%d", bitrate ) ;``` should have been fine it was killing the MQTT server, I changed it to:

```
      msg = String(bitrate) ;
      msg.toCharArray(mesg, 250) ;
      //message = char(bitrate) ;
      sprintf ( message, "%s", mesg ) ;
      if ( strlen(message) ) // Prevent sending null data to mqtt server.
      {
        mqttclient.publish ( topic, 1, false, message ) ;
        dbgprint ( "Publishing message %s to topic %s", message, topic ) ;
      }
      else
      {
        dbgprint ( "Error: Tried to publish null message %s to topic %s", message, topic ) ;
      }
```

Seems to be working so far MQTT has yet to bomb out.

NonaSuomy, 2017-05-11:
```
D: Metadata block 80 bytes
D: Streamtitle found, 76 bytes
D: StreamTitle='Physics - City Lights';StreamUrl='http://www.lounge-radio.com';
D: Publishing message LOUNGE-RADIO.COM - swiss made to topic espradio/icy/name
D: Publishing message Physics - City Lights to topic espradio/icy/streamtitle
D: Publishing message audio/mpeg to topic espradio/icy/contenttype
D: Publishing message Lounge, Downtempo, Chill, Ambient, Easy Listening to topic espradio/icy/genre
D: Publishing message http://www.lounge-radio.com to topic espradio/icy/icyurl
D: Publishing message 128 to topic espradio/icy/bitrate
D: MQTT Publish acknowledged, packetId = 24
D: MQTT Publish acknowledged, packetId = 25
D: MQTT Publish acknowledged, packetId = 26
D: MQTT Publish acknowledged, packetId = 27
D: MQTT Publish acknowledged, packetId = 28
D: MQTT Publish acknowledged, packetId = 29
D: MQTT Publish acknowledged, packetId = 30
```

Wonder if packetId will overflow.

NonaSuomy, 2017-05-11:
Lost the connection sending a content-type and I added the reconnect to the 10min timer and it reconnected after it died as the other reconnect only tries once.
```
D: Metadata block 48 bytes
D: Streamtitle found, 41 bytes
D: StreamTitle='Weezer - Feels Like Summer';
D: Warning: Tried to publish null message  to topic espradio/icy/name
D: Publishing message Weezer - Feels Like Summer to topic espradio/icy/streamtitle
D: Publishing message audio/aacp to topic espradio/icy/contenttype
D: Publishing message Rock to topic espradio/icy/genre
D: Publishing message http://www.89xradio.com to topic espradio/icy/url
D: Publishing message 40 to topic espradio/icy/bitrate
D: MQTT Publish acknowledged, packetId = 1109
D: MQTT Publish acknowledged, packetId = 1110
D: MQTT Publish acknowledged, packetId = 1111
D: MQTT Publish acknowledged, packetId = 1112
D: MQTT Publish acknowledged, packetId = 1113
D: MQTT Publish acknowledged, packetId = 1114
D: Metadata block 48 bytes
D: Streamtitle found, 38 bytes
D: StreamTitle='Catherine Wheel - Crank';
D: Warning: Tried to publish null message  to topic espradio/icy/name
D: Publishing message Catherine Wheel - Crank to topic espradio/icy/streamtitle
D: Publishing message audio/aacp to topic espradio/icy/contenttype
D: Publishing message Rock to topic espradio/icy/genre
D: Publishing message http://www.89xradio.com to topic espradio/icy/url
D: Publishing message 40 to topic espradio/icy/bitrate
D: MQTT Publish acknowledged, packetId = 1115
D: MQTT Disconnected from the broker, reason 0, reconnecting...
D: MQTT reconnection attempt: 1
D: No data input
D: No data input
D: Trying other station/file...
D: STOP requested
D: Song stopped correctly after 0 msec
D: New preset/file requested (1/0) from us2.internet-radio.com:8050
D: Connect to new host us2.internet-radio.com:8050
D: Connect to us2.internet-radio.com on port 8050, extension /
D: Connected to server
D: icy-name:CLASSIC ROCK MIA WWW.SHERADIO.COM
D: icy-genre:Classic Rock Album Light Rock 60s 70s 80s Oldies
D: icy-url:http://www.ClassicRockmiami.online
D: content-type:audio/mpeg
D: icy-pub:1
D: icy-metaint:32768
D: icy-br:256
D: Switch to DATA, bitrate is 256
D: First chunk:
D: 57 97 A1 6B 25 18 D4 90
D: 9D 78 10 18 05 AC A2 CB
D: D0 02 85 E0 E9 08 9E 48
D: 34 13 D0 08 E4 B6 48 7E
D: Metadata block 64 bytes
D: Streamtitle found, 53 bytes
D: StreamTitle='Santana - Victory is won ';StreamUrl='';
D: Recovered from dataloss
D: Metadata block 80 bytes
D: Streamtitle found, 65 bytes
D: StreamTitle='Rod Stewart - Cigarettes and Alcohol ';StreamUrl='';
D: 10 minutes over
D: MQTT Connected to the broker 10.42.42.52, session is not present
D: Subscribing to espradio at QoS 2, packetId = 1 
D: Publishing IP 10.13.38.65 to topic espradio/ip
D: MQTT Subscribe acknowledged, packetId = 1, QoS = 2
D: MQTT message arrived [espradio], length = 57, LOUNGE-RADIO.COM - swiss made - Zero Cult - Back To E
D: Command: lounge-radio.com - swiss made - zero cult - back to earth with parameter 0
D: Esp-radio called with illegal parameter: lounge-radio.com - swiss made - zero cult - back to earth
D: MQTT Publish acknowledged, packetId = 2
D: Metadata block 80 bytes
D: Streamtitle found, 65 bytes
D: StreamTitle='The Steve Miller Band - Jet Airliner ';StreamUrl='';
D: Publishing message CLASSIC ROCK MIA WWW.SHERADIO.COM to topic espradio/icy/name
D: Publishing message The Steve Miller Band - Jet Airliner  to topic espradio/icy/streamtitle
D: Publishing message audio/mpeg to topic espradio/icy/contenttype
D: Publishing message Classic Rock Album Light Rock 60s 70s 80s Oldies to topic espradio/icy/genre
D: Publishing message http://www.ClassicRockmiami.online to topic espradio/icy/url
D: Publishing message 256 to topic espradio/icy/bitrate
D: MQTT Publish acknowledged, packetId = 3
D: MQTT Publish acknowledged, packetId = 4
D: MQTT Publish acknowledged, packetId = 5
D: MQTT Publish acknowledged, packetId = 6
D: MQTT Publish acknowledged, packetId = 7
D: MQTT Publish acknowledged, packetId = 8
D: MQTT Publish acknowledged, packetId = 9
```

NonaSuomy, 2017-05-11:
Still pondering what could cause the drop...

```
D: Metadata block 48 bytes
D: Streamtitle found, 38 bytes
D: StreamTitle='Catherine Wheel - Crank';
D: Warning: Tried to publish null message  to topic espradio/icy/name
D: Publishing message Catherine Wheel - Crank to topic espradio/icy/streamtitle
D: Publishing message audio/aacp to topic espradio/icy/contenttype
D: Publishing message Rock to topic espradio/icy/genre
D: Publishing message http://www.89xradio.com to topic espradio/icy/url
D: Publishing message 40 to topic espradio/icy/bitrate
D: MQTT Publish acknowledged, packetId = 1115
D: MQTT Disconnected from the broker, reason 0, reconnecting...
D: MQTT reconnection attempt: 1
```

It skips the first message as it has a blank topic
```
D: Warning: Tried to publish null message  to topic espradio/icy/name
```
so the next message it sent was
```
D: Publishing message Catherine Wheel - Crank to topic espradio/icy/streamtitle
and then
D: Publishing message audio/aacp to topic espradio/icy/contenttype
```
Which it then reported:
```
D: MQTT Publish acknowledged, packetId = 1115
D: MQTT Disconnected from the broker, reason 0, reconnecting...
D: MQTT reconnection attempt: 1
```
1115 would be streamtitle, could be either that or contenttype caused the quit but it never got the debug message out for content-type which I believe comes first.

Edzelf, 2017-05-11:
If you send me a private email, I will give you credentials for my broker.  Then I can see the logging.

Edzelf, 2017-05-23:
ESP32 version is released.  Performes better with MQTT connections.

NonaSuomy, 2017-05-26:
Thanks @Edzelf updated the ESP8266 with the ESP32 code [GIST](https://gist.github.com/NonaSuomy/e50cfdf079fe1bbbcac30a381e34fd93) Stable so far :crossed_fingers:

NonaSuomy, 2017-05-26:
nope...

```
c112
~ld
|C⸮D: FS Total 2949250, used 11797
D: /mommyradio.mp3                  -    9639
D: /radio.ini                       -    1443
D: Added SSID 00 = All Your Base to acceptable networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 5
D:  1 - homenet5024               Signal: -87 dBm, Encryption WPA2,  
D:  2 - DIANE                     Signal: -94 dBm, Encryption Auto,  
D:  3 - dlink-2348                Signal: -92 dBm, Encryption Auto,  
D:  4 - All Your Base             Signal: -42 dBm, Encryption WPA2,  Acceptable
D:  5 - public                    Signal: -42 dBm, Encryption WPA2,  
D: End of list
D: Command: mqttbroker with parameter 10.42.42.52
D: Command: mqttport with parameter 1883
D: Command: mqttuser with parameter OpenHAB
D: Command: mqttpasswd with parameter ******
D: Command: mqtttopic with parameter espradio
D: Command: mqttprefix with parameter espradio
D: Command: mqttpubtopic with parameter espradio
D: Command: wifi_00 with parameter All Your Base/faceface01
D: Command: volume with parameter 100
D: Command: toneha with parameter 0
D: Command: tonehf with parameter 0
D: Command: tonela with parameter 0
D: Command: tonelf with parameter 0
D: Command: preset with parameter 0
D: Command: preset_00 with parameter 10.42.42.52:8080/static/mommyradio.mp3
D: Command: preset_01 with parameter ihr/CIMXFM
D: Command: preset_02 with parameter ihr/CKMXAM
D: Command: preset_03 with parameter 178.32.167.206:80
D: Command: preset_04 with parameter us2.internet-radio.com:8050
D: Command: preset_05 with parameter us1.internet-radio.com:8105
D: Command: preset_06 with parameter ihr/IHR_TRAN
D: Command: preset_07 with parameter ihr/IHR_IEDM
D: Starting ESP Version Thu, 24 May 2017 16:55:00 GMT...  Free memory 13040
D: Sketch size 345600, free size 700416
D: Reset VS1053...
D: End reset VS1053...
D: Slow SPI,Testing VS1053 read/write registers...
D: Fast SPI, Testing VS1053 read/write registers again...
D: endFillByte is 0
D: Selected network: All Your Base            
D: Try WiFi All Your Base
D: Connected to All Your Base
D: IP = 10.13.38.65
D: Start server for commands
D: MQTT uses prefix espradio
D: Init MQTT
D: Connecting to MQTT 10.42.42.52, port 1883, user OpenHAB, password OpenH@B01...
D: MQTT Connected to the broker 10.42.42.52, session is not present
D: Subscribing to espradio at QoS 0, packetId = 1 
D: MQTT Subscribe acknowledged, packetId = 1, QoS = 0
D: MQTT message arrived [espradio], length = 57, LOUNGE-RADIO.COM - swiss made - Zero Cult - Back To E
D: Command: lounge-radio.com - swiss made - zero cult - back to earth with parameter 0
D: ESPRadio called with illegal parameter: lounge-radio.com - swiss made - zero cult - back to earth
D: Publish to topic espradio/volume : 0
D: STOP requested
D: MQTT Publish acknowledged, packetId = 2
D: Song stopped correctly after 0 msec
D: New preset/file requested (0/0) from 10.42.42.52:8080/static/mommyradio.mp3
D: Connect to new host 10.42.42.52:8080/static/mommyradio.mp3
D: Connect to 10.42.42.52 on port 8080, extension /static/mommyradio.mp3
D: Connected to server
D: Publish to topic espradio/ip : 10.13.38.65
[... 1409 lines cut ...]
3fff59e0:  5bf8a5a3 11d651cd e06a0281 d0a8ae49  
3fff59f0:  9146a065 d2006994 23bec332 17dd6bba  
3fff5a00:  b1e29dbc faf1431f cdd53935 535d054c  
3fff5a10:  c3ee52ca 171b6824 22091501 d5447ab8  
3fff5a20:  e8b4400f 70f7ac8f 90a4e0c6 ae5413a9  
3fff5a30:  92a26057 54bee718 046564b9 0d58882b  
3fff5a40:  9c97fb3a 866402ec fd1f000c 7261fbb7  
3fff5a50:  fff46283 5cc0b22c 8b4014d6 2d2dfcd2  
3fff5a60:  00a0998f 28cd3980 ba77c39b 53c5e004  
3fff5a70:  3fdd2e54 e372384e b31a38fd 2528d222  
3fff5a80:  15060800 f1ff3b40 ff1e405e d54a01fc  
3fff5a90:  42343100 6f221685 3debcc2f dfbee6f5  
3fff5aa0:  7549ce15 214b41ac 2e5e8121 5e76218b  
3fff5ab0:  fb4d6c0b 9b7b7b6a 38d73813 d88248dc  
3fff5ac0:  da72e1b0 00fda2c4 26f00287 8042d880  
3fff5ad0:  c00d9900 7b33dfc8 08fe467d 78302e56  
3fff5ae0:  728f1797 258a269a f958ca34 5529e3d7  
3fff5af0:  89998df1 49c16fa9 3dedfc97 56899ef8  
3fff5b00:  8c470ed5 e6ca1fb1 8dab0aae cf43fa61  
3fff5b10:  a91fbfaf 127b38e3 7bfe23e8 cefeaa29  
3f⸮
 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```

NonaSuomy, 2017-05-27:
The prior code I made at least didn't crash the processor and would resume the connection after ten mins.
I'm guessing that it crashes now because after it disconnects it doesn't give the wifi library anytime to do anything else.

```
D: Metadata block 80 bytes
D: Streamtitle found, 66 bytes
D: StreamTitle='The Smashing Pumpkins - Bullet With Butterfly Wings';
D: Publish to topic espradio/icy/streamtitle : The Smashing Pumpkins - Bullet With Butterfly Wings
D: Publish to topic espradio/nowplaying : The Smashing Pumpkins - Bullet With Butterfly Wings
D: MQTT Publish acknowledged, packetId = 51
D: MQTT Publish acknowledged, packetId = 52
D: Publish to topic espradio/ip : 10.13.38.65
D: MQTT Publish acknowledged, packetId = 53
D: Metadata block 48 bytes
D: Streamtitle found, 42 bytes
D: StreamTitle='Rise Against - The Violence';
D: Publish to topic espradio/icy/streamtitle : Rise Against - The Violence
D: Publish to topic espradio/nowplaying : Rise Against - The Violence
D: MQTT Publish acknowledged, packetId = 54
D: MQTT Publish acknowledged, packetId = 55
D: MQTT Disconnected from the broker, reason 0, reconnecting...
D: MQTT reconnection attempt: 1
D: MQTT Disconnected from the broker, reason 0, reconnecting...
D: MQTT Disconnected from the broker, reason 0, reconnecting...
D: MQTT Disconnected from the broker, reason 0, reconnecting...
D: MQTT Disconnected from the broker, reason 0, reconnecting...
D: MQTT Disconnected from the broker, reason 0, reconnecting...
D: MQTT Disconnected from the broker, reason 0, reconnecting...
D: MQTT Disconnected from the broker, reason 0, reconnecting...
D: MQTT Disconnected from the broker, reason 0, reconnecting...
D: MQTT Disconnected from the broker, reason 0, reconnecting...
D: MQTT Disconnected from the broker, reason 0, reconnecting...
D: MQTT Disconnected from the broker, reason 0, reconnecting...
D: MQTT Disconnected from the broker, reason 0, reconnecting...

Exception (0):
epc1=0xcd080400 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3fff0bf0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fff0d90:  3fff0df0 40106fd4 3fff04dd 3fff2eac  
...
3fff5490:  7208215e 143eb5e2 6eea1f19 3f84640a  
3f
 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
(Doesn't reset gets stuck)
```

Is there any way you can load my code base from the GIST above without modifications and not using an esp32, to use your spare esp8266 + vs1053b as completely changing the code base and uC is not similar, as the esp8266 doesn't use the same Async MQTT library.

NonaSuomy, 2017-05-28:
I removed all the code from ```void onMqttDisconnect ( AsyncMqttClientDisconnectReason reason )``` and only left the debug line. Seems to have prevented the crashing so far have yet to have a reason 0 error for some reason as well.

Up to ```D: MQTT Publish acknowledged, packetId = 406``` let's see how far we get.

Edzelf, 2017-05-28:
I will react on your github, cause this is not about my software.

NonaSuomy, 2017-05-28:
Guess it was just fluke:

```
D: Publish to topic espradio/nowplaying : Sloan - If It Feels Good Do It
D: MQTT Publish acknowledged, packetId = 3786
D: MQTT Publish acknowledged, packetId = 3787
D: MQTT Disconnected from the broker, reason 0, reconnecting...
D: MQTT Disconnected from the broker, reason 0, reconnecting...
D: MQTT Disconnected from the broker, reason 0, reconnecting...
D: MQTT Disconnected from the broker, reason 0, reconnecting...
D: MQTT Disconnected from the broker, reason 0, reconnecting...
D: MQTT Disconnected from the broker, reason 0, reconnecting...
D: MQTT Disconnected from the broker, reason 0, reconnecting...
D: MQTT Disconnected from the broker, reason 0, reconnecting...
D: MQTT Disconnected from the broker, reason 0, reconnecting...
D: No data input
D: No data input
D: Trying other station/file...
D: STOP requested

Exception (29):
epc1=0x4000e1b2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffd70 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff10:  8f698b15 000002f4 3ffeee50 401004f4  
3fffff20:  3fff242c 3ffefdb0 3ffeee50 40106ed4  
3fffff30:  00000007 00000007 00000000 4021d03e  
3fffff40:  401053e8 1c361d9c 4021d4fe 00000008  
3fffff50:  00000000 00000000 4021d5ab 3ffeef04  
3fffff60:  3ffefdb0 1c361d9c 4021d4c0 402013e0  
3fffff70:  4022ec81 3ffeef04 3ffefdb0 8f698747  
3fffff80:  4022ecc6 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffefdc0 3fffdad0 3fff0f04 40216ef3  
3fffffa0:  40000f49 00005f20 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset
```

NonaSuomy, 2017-06-09:
looks like it was an issue with the asyncmqtt library, he just updated.

https://github.com/marvinroger/async-mqtt-client/issues/45

https://github.com/marvinroger/async-mqtt-client/issues/14

I see if this fixes the issue and report back.

geocristi, 2021-03-08:
Hi there , great job indeed  , but I have question
Does it accept commands via MQTT ?  I can't  send anything to it,  can only  read the messages.
I'm trying to intergate it into a HomeAssistant