Edzelf/Esp-radio #88: replace special / non ASCII characters before displaying on TFT

everyb313, 2017-05-10:
**idea for displaying UTF8 characters in songtitle**
(code stolen at  http://playground.arduino.cc/Main/Utf8ascii)
--> replace UTF8 characters with a lookup table before printing on TFT

--------------------- 
void displayinfo ( const char* str, uint16_t ypos, uint16_t height, uint16_t color )
{
#if defined ( USETFT )
  char s[255]; 
  strcpy(s,str);
  utf8ascii(s);

(....) 
  
tft.println(s);
#endif
}

//******************************************************************************************
// ****** UTF8-Decoder: convert UTF8-string to extended ASCII *******
//******************************************************************************************
static byte c1;  // Last character buffer

// Convert a single Character from UTF8 to Extended ASCII
// Return "0" if a byte has to be ignored
static const byte lut_C3[] = { "AAAAAAACEEEEIIIIDNOOOOO#0UUUU###aaaaaaaceeeeiiiidnooooo##uuuuyyy" };
byte utf8ascii(byte ascii) {
    if ( ascii<128 )   // Standard ASCII-set 0..0x7F handling  
    {   c1=0;
        return( ascii );
    }

    // get previous input
    byte last = c1;   // get last char
    c1=ascii;         // remember actual character

    switch (last)     // conversion depending on first UTF8-character
    {   
//        case 0xC2: return  (ascii);  break;
//        case 0xC3: return  (ascii | 0xC0);  break;
//        case 0x82: if(ascii==0xAC) return(0x80);       // special case Euro-symbol

        case 0xC2: return  ('~');  break;
        case 0xC3: return  (lut_C3[ascii-128]);  break;
        case 0x82: if(ascii==0xAC) return('E');       // special case Euro-symbol
         
    }

    return  (0);                                     // otherwise: return zero, if character has to be ignored
}

//******************************************************************************************
// convert String object from UTF8 String to Extended ASCII
//******************************************************************************************
String utf8ascii(String s)
{      
        String r="";
        char c;
        for (int i=0; i<s.length(); i++)
        {
                c = utf8ascii(s.charAt(i));
                if (c!=0) r+=c;
        }
        return r;
}

//******************************************************************************************
// In Place conversion UTF8-string to Extended ASCII (ASCII is shorter!)
//******************************************************************************************
void utf8ascii(char* s)
{      
        int k=0;
        char c;
        for (int i=0; i<strlen(s); i++)
        {
                c = utf8ascii(s[i]);
                if (c!=0)
                        s[k++]=c;
        }
        s[k]=0;
}

Edzelf, 2017-05-11:
Looks good.  I have never seen these strange characters on the screen.  But I will include the code.

everyb313, 2017-05-11:
Maybe you can try the following radio stations (French, Spanish, Portugese): 
streaming210.radionomy.com:80/Radio-Tango-Velours
streaming210.radionomy.com:80/Colombiasalsarosa
streaming211.radionomy.com:80/CorcovadoRadio

Edzelf, 2017-05-11:
Fixed. See new version.

closed 2017-05-31 by Edzelf