Edzelf/Esp-radio #89: Streamtitle not shown

Sindi1, 2017-05-24:
I have a strange problem. From today to tomorrow the radio does not show the song title, artist or information. It just shows the station name and "Connect to .."

After pressing the “Flash-button” on the NodeMCU Board title and artist were displayed.

Example: 

http://mp3-live.swr3.de/swr3_m.m3u

Display Shows: 
**SWR3-Livestream
Connect to swr-swr3-live.cast.addradio.de on port 80, extension /swr/swr3/live/mp3/128/stream.mp3**

In addition, there are sound disturbances every few seconds

After pressing Flash:

**Am Mikrofon: Marcus Barsch
SWR3**

No sound disturbances anymore.

Serial log: 

: Command: preset with parameter 01
D: Preset set to 1
D: STOP requested
D: Stopping client
D: Song stopped correctly after 0 msec
D: New preset/file requested (0/0) from mp3-live.swr3.de/swr3_m.m3u
D: Connect to new host mp3-live.swr3.de/swr3_m.m3u
D: Playlist request, entry 1
D: Connect to mp3-live.swr3.de on port 80, extension /swr3_m.m3u
D: Connected to server
D: Read from playlist
D: Playlistheader: HTTP/1.1 200 OK
D: Playlistheader: Server: Apache
D: Playlistheader: Content-Disposition: attachment; filename="swr3_mp3_m.m3u"
D: Playlistheader: ETag: "3487cdbeb7c9afc2a11abe4717b1733e"
D: Playlistheader: Content-Length: 241
D: Playlistheader: Content-Type: audio/x-mpegurl
D: Playlistheader: Cache-Control: max-age=86400
D: Playlistheader: Date: Wed, 24 May 2017 20:36:58 GMT
D: Playlistheader: Connection: close
D: Playlistheader: 
D: Switch to PLAYLISTDATA
D: Playlistdata: #EXTM3U
D: Playlistdata: #EXTINF:-1,SWR3-Livestream
D: Playlistdata: http://swr-swr3-live.cast.addradio.de/swr/swr3/live/mp3/128/stream.mp3
D: Entry 1 in playlist found: http://swr-swr3-live.cast.addradio.de/swr/swr3/live/mp3/128/stream.mp3
D: Connect to new host swr-swr3-live.cast.addradio.de/swr/swr3/live/mp3/128/stream.mp3
D: Connect to swr-swr3-live.cast.addradio.de on port 80, extension /swr/swr3/live/mp3/128/stream.mp3
D: Connected to server
D: Switch to DATA, bitrate is 0
D: First chunk:
D: 48 54 54 50 2F 31 2E 30
D: 20 32 30 30 20 4F 4B 0D
D: 0A 43 6F 6E 74 65 6E 74
D: 2D 54 79 70 65 3A 20 61

**After pressing Flash:**

D: Digital button 2 pushed
D: STOP requested
D: Stopping client
D: Song stopped correctly after 0 msec
D: New preset/file requested (1/2) from mp3-live.swr3.de/swr3_m.m3u
D: Connect to new host mp3-live.swr3.de/swr3_m.m3u
D: Playlist request, entry 2
D: Connect to mp3-live.swr3.de on port 80, extension /swr3_m.m3u
D: Connected to server
D: Read from playlist
D: Playlistheader: HTTP/1.1 200 OK
D: Playlistheader: Server: Apache
D: Playlistheader: Content-Disposition: attachment; filename="swr3_mp3_m.m3u"
D: Playlistheader: ETag: "3487cdbeb7c9afc2a11abe4717b1733e"
D: Playlistheader: Content-Length: 241
D: Playlistheader: Content-Type: audio/x-mpegurl
D: Playlistheader: Cache-Control: max-age=86400
D: Playlistheader: Date: Wed, 24 May 2017 20:40:04 GMT
D: Playlistheader: Connection: close
D: Playlistheader: 
D: Switch to PLAYLISTDATA
D: Playlistdata: #EXTM3U
D: Playlistdata: #EXTINF:-1,SWR3-Livestream
D: Playlistdata: http://swr-swr3-live.cast.addradio.de/swr/swr3/live/mp3/128/stream.mp3
D: Entry 1 in playlist found: http://swr-swr3-live.cast.addradio.de/swr/swr3/live/mp3/128/stream.mp3
D: Playlistdata: #EXTINF:-1,SWR3-Livestream
D: Playlistdata: http://swr-mp3-m-swr3.akacast.akamaistream.net/7/720/137136/v1/gnl.akacast.akamaistre
D: Entry 2 in playlist found: http://swr-mp3-m-swr3.akacast.akamaistream.net/7/720/137136/v1/gnl.akaca
D: Connect to new host swr-mp3-m-swr3.akacast.akamaistream.net/7/720/137136/v1/gnl.akacast.akamaistrea
D: Connect to swr-mp3-m-swr3.akacast.akamaistream.net on port 80, extension /7/720/137136/v1/gnl.akaca
D: Connected to server
D: Content-Type: audio/mpeg
D: icy-description:Mehr Hits. Mehr Kicks. Einfach SWR3
D: icy-genre:Pop Music
D: icy-name:SWR3
D: icy-pub:1
D: icy-url:http://www.swr3.de
D: Server: Akacast 1.0
D: Cache-Control: no-cache
D: icy-metaint:16000
D: Switch to DATA, bitrate is 0
D: First chunk:
D: 63 EB 7E BF F4 92 2E 3A
D: 9B C0 00 01 A4 43 41 2C
D: 8B 40 67 79 DB 2A 23 5A
D: D6 51 E6 8D B5 09 7F 5F
D: Metadata block 48 bytes
D: Streamtitle found, 41 bytes
D: StreamTitle='Am Mikrofon: Marcus Barsch';

Edzelf, 2017-05-25:
I can reproduce this problem.  The same error occurs in mu ESP32 version.  The header "icy-metaint:16000" is not seen.  Therefore the sound is disturbed every 16000 bytes.  If you go directly to "swr-swr3-live.cast.addradio.de on port 80, extension /swr/swr3/live/mp3/128/stream.mp3" it works correctly.

I will look in to this problem later.

Edzelf, 2017-05-26:
Fixed.

closed 2017-05-26 by Edzelf

Sindi1, 2017-05-27:
Thanks :+1: