Edzelf/Esp-radio #90: Empty streams & url parsing error

koskee, 2017-05-26:
I just got my radio up and running after figuring out a bunch of library related compile errors. Arduino boards manager is garbage Imo. 

I'm actually quite happy with the radio so far. However, I've run into a couple things that aren't optimal, as well as an ini file parsing issue. 

First, I guess I should mention that on my first trip through the channels, it appears that approximately 1 of 2 of the stations will connect and appear to be functional, (even pulling up track names from the metadata and the first chunk has normal hex values from what I can tell) but there's no output. I would assume that if they were selected to be used in the project as a demo station, that they must work for at least somebody? Why would I be getting either "no data" or a stream with no output? Would it be licensing/country issues? I didn't think that would be an issue with internet radio so just wondering what's up with that.. 

Second, and the real issue I had was adding a station that I found thru the link at the bottom of the web interface. The simplified link is "http://noisefm.ru:8000/play.m3u&t=.m3u" after I removed the first portion of the url (I assumed that's just a wrapper used for launching some kind of player,  probably?) 
The issue that I'm observing with this is that when it tries to connect it truncates anything past the '&' and ultimately fails to connect. Basically when the stream's url is pulled from radio.ini, it tries to connect to "http://noisefm.ru:8000/play.m3u" which not only does not work, but the esp has a fit over it, presumably one of the characters has special significance or possibly just because of the double '.' dot. 
It apparently sees that as an EOF (end of file) because upon adding that to the ini file (I put it as preset 2) all of the other stations that would have been listed after it were no longer available in the preset list on webinterface. 
I know that it should work because I had initially tried entering it into the text entry "play stream/file" box on webinterface to test that I had the correct url, and that worked fine. 

An additional effect, although I'm not certain the two are related because I had an instance when this happened prior to changing the ini, was that when clicking the config tab, I just get the last portion of the html code that would have made up the page appearing as plaintext in the browser window. This began happening consistently after changing the file.  
I haven't identified how to replicate this behaviour, but it seems almost like when the esp is bogged down doing something intensive that maybe the connection is timing out or the response (datatype?)  isn't being sent properly which makes the browser think that what it's receiving is the plaintext body of the page. It's weird because one might think that the <html tags> would still have an effect, especially because it doesn't display <html><head></head><body> or any of the beginning of the html document, only the last maybe ~50 lines. 
 
For resolving the url parsing error, possibly enclosing the url in quotes inside the ini file - or some other special character -  seems like a feasible remedy to me, but I haven't had a lot of time to analyze the code (its relatively a pretty hefty size application - one of the most elaborate that I've seen for the ESP.. Good work!) It's likely you guys would know best what might work, and what might not, since you will be much more intimately familiar with how the code works. 

Any help is appreciated -  sorry if this bug report is a little sparse on initial troubleshooting, but I tried to gather as much I could while still remaining relevant. Let me know if there's any troubleshooting/testing you guys need me to do.

Edzelf, 2017-05-26:
It looks like everything past the ampersand is removed.  I will look in to that.
I will also check which stations in the list are no longer available.

Edzelf, 2017-05-26:
Ampersands are not allowed in a preset station.  In your example "noisefm.ru:8000/play.m3u" will work correctly.  The playinglist is only one entry in this case, so you can also use "noisefm.ru:8000/play" as well.
I have checked the example stations.  Some of them did not work because of a patch I made this morning.  It made distinction between "Content-Type" en "content-type" in the HTTP header.  That's corrected now.

The SPIFFS system is not very reliable, so occasionally it can happen that a part of the info is missed when read from the SPIFFS.

My ESP32 version seems to perform much better.

koskee, 2017-05-26:
Okay, I tried your alternate URL, and it works. So that's excellent. Now I'm wondering where you derived the alternate address from? Is there a procedure for acquiring valid URLs for a given station? or is /play a common address for these sorts of things? Just wondering so I don't have to come back and ask you to figure out why a bunch of stations aren't working and so I know what to do when I encounter more ampersands. I'll leave this open for a day or 2 so you have a chance to respond.

Thanks for your prompt attention/resolution. 

At work I am currently forced to listen to a certain country music station 8hrs/day and its driving me crazy (they play the same songs 2-4 times in a single shift), so this little project will hopefully provide us humble workers with some much needed variety once I sweet talk the manager into adopting this extremely inexpensive and excellent alternative. You're truly a lifesaver and I applaud thee.

Edzelf, 2017-05-27:
Nice story about your music at work!
The best way to see the target URL for a station is by using Winamp.  See the PDF file, page 4.  I have changed the text to clarify the procedure.  Note that URLs ending with ".mu3" are just playlists, not stations.  Often playlist contain just one station, so there's no real need for al list.

koskee, 2017-05-27:
Perfect. I will try that. 

Thank you, you've been most helpful.

closed 2017-05-27 by koskee