Edzelf/Esp-radio #92: Latest .ino commit results in choppy output

koskee, 2017-05-28:
I had originally thought it was my internet connection (lots of people in my house use it, so its sometimes slow) or as a result of one of my modifications to the code, but upon further testing, it seems that the latest commit to the arduino file has made the output from my radio choppy.  The characteristics of this are dependent on the bitrate of the stream. Basically it gives me a fraction of a second of audio followed by a fraction of a second of silence (the silence seems to be longer for higher bitrate streams). What made me believe it was actually the new version is I went back to the previous version, and it picks up the stream just fine (although not all the stations work as I had previously mentioned in another issue).

I hate to be a bother, but..
Since I am not entirely fluent in all of GitHub's features, I haven't been able to find where it will list the changes made from one file to another, so I am uncertain of what was actually changed, but you had said it was something to do with the difference between "content-type" and "Content-Type" I believe.
As you had said, all of the streams appear to be sending audio in this new version, however I can't actually listen to any of them as they are all full of gaps. To me, it seems like the ESP is no longer keeping the buffer full enough for smooth playback, so it seems we may need to find an alternate (faster) fix for the issue? Its unfortunate that the patch didn't work out as I had hoped.

Ideas? Thoughts?

Edzelf, 2017-05-29:
The "26 May 2017 14:35" version should play smoothly.  Al least mine is....
You may use the TEST command to inspect the filling of the ringbuffer.  This should be close to 20000.  Then you also see the number of bytes in the input stream.  If this is zero, the WiFi connection is poor.
The list of changes is kept in the sketch itself.
If you have 7 dollars to spare, switch to the ESP32 version.  It is more stable and has more goodies.

koskee, 2017-05-29:
I think ive figured it out. 

I was getting a lot of "mpeg compression" noises, ie. Blips and bloops and various garbled sounds from the radio that hadn't been apparent in earlier testing. I rearranged my breadboard, thinking I was maybe corrupting the data running power lines beside signal lines, with very limited success. It helped a bit but the odd corruption sounds were completely stubborn and would not go away. I tried adding a hefty capacitor to the 5v power rail, and a smaller one on the 3v3 rail, in case it was a power supply issue - marginal improvement. 

I saw you said elsewhere that the reset circuit could be removed, so I ran the reset line to the reset on the ESP. This prevented both from booting or even flashing the LED. So I ran it to the 3.3v pwr bus, and it was kind of working, but the noise was super prevalent. 

I clearly wasn't thinking, because the mp3 board is not 3v3, it's a 5v. Connected the reset line directly to the 5v bus, and the difference was immediately obvious (tried with a resistor first which didn't really help). My jaw hit the floor. The audio from the module is now crystal clear and everything seems WAY more stable, in terms of the audio anyway. 

So I'm thinking this is the way to go. The esp can only put out 3v3 on a good day, and add a diode drop, even with a smaller shottky drop would put that into the dreadedful "undefined"  voltage range pretty easily. 

So for anyone getting strange output from the 1053 board, giving the reset line a solid 5v really helped. 

Now I'll just have to try the latest version again and see if the problem has been remedied. 

Ps. 
I ordered a couple esp32s dev boards this weekend, just have to wait for them to arrive. Paid for quick shipping, so hopefully by week's end..

Edzelf, 2017-05-29:
3.3 Volt should be enough.  But I found that the reset line of the VS1053 seems to pick-up the high frequency of the ESP transmitter easily.  A very short line to the reset of the ESP should work.  But +5 Volt is also a good solution, then it can be reset by power-off.

koskee, 2017-06-01:
Made some progress with incorporating a reset circuit that was controllable with the esp but still gave solid output. 

What I did was take the two dual diode outputs and feed them into the a annode of an opticoupler with ~100ohm series resistor from cathode to ground. The collector of the PC817 opticoupler was fed 5V from the rail, and the emitter was connected to the XRST. This was still a bit choppy so I reinforced the voltage level coming out of the 817 with a 4.7uF cap between ground and the emitter, which manages to hold the reset line steady at around 4.2V coming out of the opticoupler. 

I also should mention that I've got a 2200uF electrolytic cap and a yellow 10nF multilayer ceramic on my 5V power bus, and a 63uF Electro holding the 3v3 bus stable. 

Overall it works really well, despite being a little component happy. I think that once I get it onto a perf board instead of the solderless breadboard I've been using for testing & experimenting, that all most of the wiring can be kept reply short and soldered in for a superior connection compared to the cheap pressure clips in my breadboard.

koskee, 2017-06-06:
I've now moved on to ESP32, and for anybody that is on the fence about whether the upgrade is worthwhile, there truly is no question about it. 

Where the 8266 just barely covered the bases of running this setup, the ESP32 is all that AND a bag of potato chips. It's clearly got plenty of room to spare, and the webinterface actually works properly!
From what I've seen so far, the new 32 platform can fully handle the task, and without making any compromises along the way. 

For example, Ive upped my ring buffer to 32,000 - to start -  and I'm sure I can go much higher.. I am going to see if I can find the upper limit of that sometime soon. I found the esp would only run reliably with around 21000 - maaaybe 22000 in the buffer before it got unstable. 

Cheers guys, good luck

closed 2017-06-06 by koskee