Edzelf/Esp-radio #94: Feature request: analogue buttons operation

vadpol, 2017-05-29:


vadpol, 2017-05-29:
I've managed to assemble as much compact version as possible in a small box with d-class amplifier and 2500 mAh lipo incorporated. Headphones also could be connected, then the amp is switched off automatically. It works perfectly well without any problem from the very beginning for max 10hrs continuous play on one battery charge. I saw that the newest update has the new resistor configuration of analogue buttons because many esps on the market have a0 input of 1V as a max. Propose just to change the one 10k resistor connected to the ground to 3,6k one as the simplest way. Then the output voltage of the total resistor chain will never exceed 1V whatever button is pushed. Secondly, also propose to configure the buttons as follows (after some practicing): one button just for incrementing (in a circle) of the preset stations, the remained two buttons for volume adjusment up and down. This will be more convenient for using of the device on the run without need to go to a webpage to do so. Many thanks!

Edzelf, 2017-05-30:
If you have only one power source, you may have sound distortion due to the unusual analog output of the VS1053.  Maybe you classs-d amplifier works perfectly with the VS1053.  What type do you use?
The analog buttons in my example will output 139 mV with no button pushed, The buttons will generate 161, 192 or 238 mV.  But you are free to use other values.  The software will search for the nearest value with respect to asw1, asw2, asw3.  asw1, asw2, asw3 are in ADC counts, not mV.
The ESP32 version has up to 14 free definable digital input for this purpose.

vadpol, 2017-05-30:
I use just generic PAM8403 based Chinese made d-class stereo amplifier and didn't find any sound distortion so far even everything is working on one battery. I also modified a power scheme (by using of a very compact 5V operated relay of DPDT type) and two switching groups two position slide switch in such a way that there are two power modes available.  If no USB power connected: - there is total OFF mode or battery power ON mode. If USB power connected (which triggers a relay if so): - power ON mode from USB source only or BATTERY CHARGING mode (no power to the rest of the scheme). I could share the diagram if I manage to draw it nicely :)  As for the analogue buttons - they are working fine with my described above setup and the following values: asw1 = 309, asw2 = 779, asw3 = 446. The preset stations could be incremented or decremented by the buttons, also to set the initial one by dedicated button. No issue with that. My request is a bit different - as I am more to hardware than to software - would it be possible  to assign (in the code) two analogue buttons for the volume adjustment, not for choosing of the preset stations up and down? As to my practical experience with the device  so far one button is well enough to pick up the station if repeatedely pushed and thus going through the presets one by one in a loop. The remained two buttons in this case would be much more convenient to use for the volume adjustment up and down.  Honestly I've tried to do so already in the code but my change causes the full reset when one of the two buttons is pushed :( so suspect that I didn't take everything into consideration. I believe that just small code change is needed in this case, but do not aware  where exactly... It would be nice to have a proposed version I think. Could you work it out or just give a clue on what to change in the code? ESP8266 version is used. Many thanks!

vadpol, 2017-05-30:
Here is a small video on how it sounds however still USB powered and with active external sound boxes
[ESP Web radio.zip](https://github.com/Edzelf/Esp-radio/files/1038470/ESP.Web.radio.zip)
Battery powered it sounds the same, no difference!

vadpol, 2017-05-30:
I do believe that sound distortion problem lies in the connecting cable length in between ESP8266 module and VS1053 module. It should be as short as possible. In my case there is no cable, everything is connected via intermediate small pcb where all the connections are done.

Edzelf, 2017-05-30:
Nice box you made!

Some class-d amplifiers have an analog common connected to GND.  The analog common of the VS1053 is at 1.65 Volt.  That will cause the destortion.

The code in timer100() could be something like this:
```
        dbgprint ( "Analog button %d pushed, v = %d", anewval, v ) ;
        if ( anewval == 1 )                         // Button 1?
        {
          ini_block.newpreset = currentpreset + 1 ; // Yes, goto next preset
        }
        else if ( anewval == 2 )                    // Button 2?
        {
          ini_block.reqvol += 2  ;                  // Increase volume
        }
        else if ( anewval == 3 )                    // Button 3?
        {
          ini_block.reqvol -= 2  ;                  // Decrease volume
        }
```

vadpol, 2017-05-30:
Many thanks, I will give it a try! I still want to encapsulate the box in a carbon film when it's done.  I was changing exactly the same subroutine timer100, but perhaps missed something. Let's see, I will come back soon.

vadpol, 2017-05-31:
Just tested with above correction - it works just fine! Good job! This will help to avoid a variable volume resistor at the side of the box, so even smaller size box then! If I am not so nasty - just thought that it also might be a good idea to implement a volume indication on the display in the bottom line, something like 'VOLUME = **' were ** could be just numbers from 00 to 99 (or 100) showing the present value. Will it be possible? Thanks a lot!

vadpol, 2017-05-31:
Made a small update of above code to keep the volume value within 0-100 range as it was jumping over:

dbgprint ( "Analog button %d pushed, v = %d", anewval, v ) ;
        if ( anewval == 1 )                         // Button 1?
        {
          ini_block.newpreset = currentpreset + 1 ; // Yes, goto next preset
        }
        else if ( anewval == 2 )                    // Button 2?
        {
          ini_block.reqvol += 2  ;                  // Increase volume
          if ( ini_block.reqvol > 100 )             // Keeping of the volume within upper limit
          ini_block.reqvol = 100 ;
        }
        else if ( anewval == 3 )                    // Button 3?
        {
          ini_block.reqvol -= 2  ;                  // Decrease volume
          if ( ini_block.reqvol < 0)                // Keeping of the volume within lower limit
          ini_block.reqvol = 0 ;
        }

Edzelf, 2017-05-31:
Volume indicator added in new version.

vadpol, 2017-05-31:
Everything works fine! Thanks a lot!

closed 2017-05-31 by vadpol

kolivi5, 2021-11-21:
Hi.
Please help with analog keys. 
I made an esp radio with one button, and it works. I'm trying to insert 3 buttons but the buttons don't work. 
I changed in the code:
 #define asw1    252
#define asw2    334
#define asw3    499
//#define asw1    2000
//#define asw2    2000
//#define asw3    2000
Do you need anything else
change?
Where is the mistake?

Edzelf, 2021-11-22:
This looks okay.   I would suggest to add some debug lines in the "anagetsw" function, like this:
```
//******************************************************************************************
//                                  A N A G E T S W                                        *
//******************************************************************************************
// Translate analog input to switch number.  0 is inactive.                                *
// Note that it is adviced to avoid expressions as the argument for the abs function.      *
//******************************************************************************************
uint8_t anagetsw ( uint16_t v )
{
  int      i ;                                    // Loop control
  int      oldmindist = 1000 ;                    // Detection least difference
  int      newdist ;                              // New found difference
  uint8_t  sw = 0 ;                               // Number of switch detected (0 or 1..3)

  if ( v > analogrest )                           // Inactive level?
  {
    for ( i = 0 ; i < NUMANA ; i++ )
    {
      dbgprint ( "ana switch seen" ) ;
      newdist = analogsw[i] - v ;                  // Compute difference
      newdist = abs ( newdist ) ;                  // Make it absolute
      if ( newdist < oldmindist )                  // New least difference?
      {
        oldmindist = newdist ;                     // Yes, remember
        sw = i + 1 ;                               // Remember switch
        dbgprint ( "sw may be %d", sw ) ;
      }
    }
    dbgprint ( "sw %d detected", sw ) ;
  }
  return sw ;                                      // Return active switch
}

```

kolivi5, 2021-11-23:
Hello,
thank you very much for your prompt response.

I've tried this code and it works (with R4=3k3)! Thank you again, you
helped a lot.

All the best,
Ivica

napisao je: