Edzelf/Esp-radio #95: Pinout for TFT lcd with ST7735S controller

Mantikor, 2017-05-30:
Pinout for TFT lcd with ST7735S controller
[image]

vadpol, 2017-05-31:
Still a new display library should be used due to chip change  ILI9163C --> ST7735C. Also suspect different display commands then. Any recommendation on that?

Edzelf, 2017-05-31:
I think any display will do.  But since only a few functions are used, it will not be too difficult to implement a different display.

Mantikor, 2017-05-31:
no any changes needed for this type of display, different only pinout markup

srspinho, 2018-07-20:
Thank You Mantikor ! ! 

The ST7735 display worked like a Charm here !  No need to change anything on the code.

JOliverasC, 2020-08-04:
I changed the library and few lines... 

//#include <TFT_ILI9163C.h>
#include <Adafruit_ST7735.h> 

//TFT_ILI9163C     tft = TFT_ILI9163C ( TFT_CS, TFT_DC ) ;
Adafruit_ST7735 tft = Adafruit_ST7735(TFT_CS, TFT_DC, TFT_RST);   // TFT_RTS defined as -1

  //tft.begin();          // Init TFT interface
  tft.initR(INITR_BLACKTAB);                           // Init ST7735S chip, black tab
  tft.setRotation ( 3 ) ;                                    // Use landscape format
  tft.fillRect ( 0, 0, 160, 128, BLACK ) ;             // Clear screen does not work when rotated
  //tft.clearScreen() ; 
[image]