Edzelf/Esp-radio #98: Geeeteck VS1053 icw ESP12E dev board DOIT

ilioss, 2017-06-24:
Hello All,
[list-opstartesp8266-12E dev.txt](https://github.com/Edzelf/Esp-radio/files/1099817/list-opstartesp8266-12E.dev.txt)

After first building the internet radio with the ESP32 ( succesfull ) I started again with the original internet radio with the 12E.
Install. the ESP12E DEV KIT ( DOIT) on a geeeteck VS1053 arduino shield. No succes. After some looking around on internet I tested the board as mp3 player with succes. I was able to have song played from the sd card.
Again I connect to my ESP12E no succes.
So I have two doubts,
1) Is my ESP12 E still healty?
2) Howe do I check my geeetech shield or must it set for the correct operating mode?
The serial monotor list gives also as SV1053 test all 00000 aperntly I think it is not tested.
See enclosed list. On my ESP32 list some figures are filled with data other than 00000.
Like to hear,
Greetings,
Theo``

Edzelf, 2017-06-24:
As the debug logging shows only bad reads from VS1053, I'm pretty sure that the wiring is not correct.

ilioss, 2017-06-24:
Hi Ed,
Thanks for your quick reply.
Indeed the list show bad VS1053.
Either wrong and or bad connections between ESP12E dev kit and VS1053.
Menwhile I tested the ESP12 solo it works ok.
The VS1053 also tested as mp3 player looks ok.
Now i would like to test the i/o from both to start with the ESP12.
Can i test all the outputs like
pinMode (2, OUTPUT);

digitalWrite(2, high);
delay(1000);
digitalWrite(2, low);
delay(1000);

For all gpio,s?
Than i can check output for theire function.
Apart from that I will double check my connections.
Regards,
Theo

Edzelf, 2017-06-25:
Yes, good plan.  All input bits can be tested likewise.

closed 2017-07-03 by ilioss