Edzelf/Esp-radio #99: Allways errors from VS1053 on startup

ManfredEP, 2017-07-10:
Hello,
I tried to build the whole device and then load software and SPIFFS image but on start the device on serial interface allways reports a lot of errors. All belongs to VS1053 and looks like: 
D: VS1053 error retry SB:000C R1:FFFF R2:FFFF
All the lines I could not copy here since the copy&paste doesn't work for unknown reason on the whole stuff ... only on parts like a single line. The hex data behind the SB: are different in different lines but not the R!: and R": parts bind it. What's wrong here? Are there chances it's only a missing connection or some other build-up error or is the VS1053 damaged?

greetings and thanks to all with any help (or any information where to go to get help if needed)

Edzelf, 2017-07-10:
Looks like a wiring error.  The error message come from a test.  Data is written to the VS1053 and then the data is read back twice and compared with the original data.  "SB"stands for "Should Be", "R1" for "Read back 1st try", "R2" for "Read back 2nd try".
So check the wiring of all 9 wires that go to the VS1053.  The reset line (pin 3) may be connected to the reset pin of the ESP8266.  Then you don't need the reset circuit.

Cut and past of the debug output works correct if you do not select the garbage lines at the beginning of the output.  So start with the first line containing "D:".

ManfredEP, 2017-07-10:
Thanks for your fast answer. You are right: the connections were wrong I found out by myself at 3 o'clock in the morning. Now I corrected the error but now the message is much shorter: vs1053 not properly installed. Hope the device is not damaged (due to the error one pin was shorted to V(DD)). I will check this now. 

Thanks too for the help on cut&paste. Next time it would help I hope. 

greetings

ManfredEP, 2017-07-10:
Taken wrong button, excuse me. 

greetings

texynz, 2017-08-11:
Im getting similar:

D: Slow SPI,Testing VS1053 read/write registers...
D: VS1053 error retry SB:0000 R1:FFFF R2:FFFF
D: VS1053 error retry SB:012C R1:FFFF R2:FFFF
D: VS1053 error retry SB:0258 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0384 R1:FFFF R2:FFFF
D: VS1053 error retry SB:04B0 R1:FFFF R2:FFFF
D: VS1053 error retry SB:05DC R1:FFFF R2:FFFF
D: VS1053 error retry SB:0708 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0834 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0960 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0A8C R1:FFFF R2:FFFF
D: VS1053 error retry SB:0BB8 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0CE4 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0E10 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0F3C R1:FFFF R2:FFFF
D: VS1053 error retry SB:1068 R1:FFFF R2:FFFF
D: VS1053 error retry SB:1194 R1:FFFF R2:FFFF
D: VS1053 error retry SB:12C0 R1:FFFF R2:FFFF
D: VS1053 error retry SB:13EC R1:FFFF R2:FFFF
D: VS1053 error retry SB:1518 R1:FFFF R2:FFFF
D: VS1053 error retry SB:1644 R1:FFFF R2:FFFF
D: VS1053 not properly installed!

Using this module

http://g01.a.alicdn.com/kf/HTB1UPydJVXXXXclXFXXq6xXFXXX1/Free-Shipping-1PCS-VS1003B-VS1053-MP3-Module-font-b-Development-b-font-Board-Onboard-font-b.jpg

Wiring is correct.

Tried bridging pins 31/32 just in case that board suffers from stuck in midi mode. 

Gets the first chunk from the stream and then nothing. NOthing i heard.

Edzelf, 2017-08-12:
Looks like a problem with the SPI bus.  MISO seems to be HIGH all the time.  Has nothing to do with midi mode, bacause the test is performed on the volume register of the VS1053.

texynz, 2017-08-12:
Chucked that board and got a new one and it ran straight away. Buy cheap and you bound to get some bad ones in the batch.

Edzelf, 2017-08-12:
I managed to repair one faulty VS1053.  It had a solder bridge between 2 pins.