Edzelf/Esp-radio #101: can only process 64kBit streams

student1704, 2017-07-23:
I have build this project with NODEMCU 0.9 and VS1053 without a TFT. 
But only 64kBit streams are processed properly. With faster streams the tone is unclear.

What to do?

Edzelf, 2017-07-23:
Please give the URL of a failing stream.  Can you check if this stream works on a Windows machine?

student1704, 2017-07-24:
I have tried many - alwas the same error. So, it is not a matter of one URL only.
I can hear voices from faster streams but it is not understandable.
Furthermore it seems that volume controll doesn't work or the output signal is extrem stong.
I use the VS1053B. Are the control commands different to the VS1053?

student1704, 2017-07-25:
I use the standard radio.ini and replaced item 6 and 7 with 64kbit streams. Those work any other not.
I think the problem comes from signal XSC. Because when I connect it as proposed - nothing happen. When I tie XCS with 10 K to GND then 64 kbit work. It seems - control commands are not accepted.

Edzelf, 2017-07-25:
Remove the reset circuit (the 2 diodesand the 1 k resistor).  Then tie the RST pin of the VS1053 to the RST of the NodeMcu or with a pull-up resistor to +3.3V

student1704, 2017-07-25:
Ok, I made the modifications, but without no success.
I attached 2 screenshots measured with my slow logic analyzer.
[image]
[image]

Edzelf, 2017-07-25:
Yes, XCS seems to be the problem.  It should be low during the clockpulses.

KabuloKans, 2018-01-19:
Hi Edzelf and congrats for your cool project ! 

Got same issues as student1704 (and the same chip V1053b).  Only streams below 64k are working correctly (without resistor tied to GND),  volume/tones settings aren't working but I can see commands executed in the serial terminal. (Other settings are working on the website version : Next/Prev/Stop/Resume/status/Test)

Got rid of breadboards, removed the reset circuits, etc... 
Nothing worked and XCS always HIGH too.

Any idea ? (Maybe a small difference between V1053 and V1053B ?)

KabuloKans, 2018-01-19:
Don't really know why but this is working now...
Changed a jack adapter... (don't understand the link between amp and sound distortion...)

Thanks for your work anyway !

KabuloKans, 2018-01-19:
Is it possible that a capacitor (in ESP) is being in cause ?

While testing I had suddenly no sound from VS1053b output so I thought the problem was amp related.
I checked my cables (and tried a new one) and removed the ESP from my shield (DIY). Unfortunately (and shame on me!) I replaced the ESP upside/down for a second. 
When I replaced the ESP correctly, everything was working.  Maybe I helped a capacitor to discharge.

Edzelf, 2018-01-19:
In the IDE: Select V1.4 Prebuilt as IwIP Variant. 256 kbps stations will work now.

KabuloKans, 2018-01-19:
Thank you for helping !