Edzelf/Esp-radio #102: Again error on ESP8266 with vs1053

ManfredEP, 2017-07-26:
Since the other one sends a lot of error messages bake to master I bought a new vs1053. I had made some mistake on wiring SPI so the fault was not strange to me. But now it doesn't work too. SPI works fine (as I think on the messages from ESP). Wb Interface works too and a file playing I can start ... but there's nothing played. Allways the same: no data input. 
`D: FS Total 2949250, used 1286626
FS Total 2949250, used 1286626
D: /radio.ini                       -     887
D: /Test.mp3                        - 1274727
D: Added SSID 00 = Rx5dsT to acceptable networks
D: * Scan Networks *
D: Number of available networks: 6
D:  1 - DIRECT-QLCLX-3300 Series  Signal: -85 dBm Encryption WPA2  
D:  2 - Rx5dsTG                   Signal: -77 dBm Encryption WPA2  
D:  3 - Gastzugang                Signal: -85 dBm Encryption Auto  
D:  4 - FRITZ!Box Fon WLAN 7390   Signal: -76 dBm Encryption WPA2  
D:  5 - Speedlink-967             Signal: -84 dBm Encryption WPA2  
D:  6 - Rx5dsT                    Signal: -68 dBm Encryption Auto  Acceptable
D: --------------------------------------
D: Command: mqttbroker (without parameter)
D: Command: mqttport with parameter 1883
D: Command: mqttuser with parameter none
D: Command: mqttpasswd with parameter none
D: Command: mqtttopic with parameter kuechenradio
D: Command: mqttpubtopic with parameter kuechenradioIP
D: Command: wifi_00 with parameter Rx5dsT/--------------------------------
D: Command: volume with parameter 99
D: Command: toneha with parameter 0
D: Command: tonehf with parameter 0
D: Command: tonela with parameter 0
D: Command: tonelf with parameter 0
D: Command: preset with parameter 0
D: Command: preset_00 with parameter swr-swr4-rp.cast.addradio.de/swr/swr4/rp/mp3/128/stream.mp3
D: Starting ESP Version Wed, 31 May 2017 12:35:00 GMT...  Free memory 12992
Starting ESP Version Wed, 31 May 2017 12:35:00 GMT...  Free memory 12992
D: Sketch size 345264, free size 700416
Sketch size 345264, free size 700416
D: Reset VS1053...
D: End reset VS1053...
D: Slow SPI,Testing VS1053 read/write registers...
D: Fast SPI, Testing VS1053 read/write registers again...
D: endFillByte is 0
D: Selected network: Rx5dsT                   
Selected network: Rx5dsT                   
D: Try WiFi Rx5dsT
D: IP = 192.168.178.27
D: Start server for commands
D: STOP requested
D: Song stopped correctly after 0 msec
D: New preset/file requested (-1/0) from http://swr-swr4-rp.cast.addradio.de/swr/swr4/rp/mp3/128/strea
D: Connect to new host http://swr-swr4-rp.cast.addradio.de/swr/swr4/rp/mp3/128/stream.mp3
D: Connect to http on port 0, extension //swr-swr4-rp.cast.addradio.de/swr/swr4/rp/mp3/128/stream.mp3
D: Request http://swr-swr4-rp.cast.addradio.de/swr/swr4/rp/mp3/128/stream.mp3 failed!
D: No data input
D: No data input
D: Trying other station/file...
D: STOP requested
D: Song stopped correctly after 0 msec
D: New preset/file requested (0/0) from 
`
Please excuse me: The password of my WiFi I have deleted with '-'
Now the ini-file of the device: 
`# radio.ini
# Initialization file for Esp-radio
#
# MQTT broker, credentials and topic  to subscribe
#
mqttbroker =  			# Broker to connect with
mqttport = 1883					            # Portnumber (1883 is default)
mqttuser = none					            # (No) username for broker
mqttpasswd = none				            # (No) password for broker
mqtttopic = kuechenradio				        # Topic for receiving commands
mqttpubtopic = kuechenradioIP			      # IP will be pubished here
#
# WiFi network credentials, may be more than on entry.  Strongest signal will be selected.
wifi_00 = Rx5dsT/Eqb73nDfmPasWcCgqz81XtHReyuB

#
# VS1053 settings
volume = 99
toneha = 0
tonehf = 0
tonela = 0
tonelf = 0
#
# Presets
preset = 0					                               # Start with preset 0
preset_00 = http://swr-swr4-rp.cast.addradio.de/swr/swr4/rp/mp3/128/stream.mp3 #  0 - SWR4
# End of file
This is one station from here. The radio has a special purpose so only this is needed. If the one file on SPIFFS stored is requested through web interface this is the reaction:
`D: FileRequest received /index.html
D: Response sent
D: FileRequest received /favicon.ico
D: Response sent
D: list request from browser
D: FileRequest received /radio.css
D: Response sent
D: Command: station with parameter Test.mp3
D: Connect to new host Test.mp3
D: Connect to  on port 80, extension /
D: Request Test.mp3 failed!
D: No data input
D: Trying other station/file...
D: STOP requested
D: Song stopped correctly after 0 msec
D: New preset/file requested (0/0) from 
`
So the question is: what is happening here? Or: Why nothing correct is happening here?

Thanks for suggestions or possible corrections

Manfred

Edzelf, 2017-07-27:
The problem lies in this line:
D: Connect to http on port 0, extensi....
Port 0 is wrong, it should be should be 80.
I will look into it.
A workaround may be to specify the preset like:
preset_00 = swr-swr4-rp.cast.addradio.de/swr/swr4/rp/mp3/128/stream.mp3:80     # 0 - SWR4

Edzelf, 2017-07-27:
For files on SPIFFS, you should specify the file like:
localhost/Test.mp3
Note that the name is case-sensitive.

Edzelf, 2017-07-27:
Found the problem.  Just remove the "http://" from the beginning of the line:
preset_00 = swr-swr4-rp.cast.addradio.de/swr/swr4/rp/mp3/128/stream.mp3 # 0 - SWR4

ManfredEP, 2017-07-27:
Hello,
first I checked with file, since for the radio check I have to change the ini file. That's the reaction on it:
`D: New preset/file requested (0/0) from 
D: Command: station with parameter localhost/Test.mp3
D: First chunk:
D: 49 44 33 03 00 00 00 00
D: 01 68 54 50 45 31 00 00
D: 00 04 00 00 00 4D 43 32
D: 54 53 53 45 00 00 00 4C
D: Recovered from dataloss
D: No data input
D: No data input
D: Trying other station/file...
D: STOP requested
D: Song stopped correctly after 0 msec
D: New preset/file requested (0/0) from 
`
As one can see it tries but it doesn't work. Is here anyone using this radio so I can check if there is anything other wrong (for example a problem in the software)? The error message looks for me like a try to read but any problem stops it. 
Now I have changed the ini file. First the http:// at the beginning is out and :80 I have added at the end of the line. 
That's the reaction:
`D: Start server for commands
D: STOP requested
D: Song stopped correctly after 0 msec
D: New preset/file requested (-1/0) from swr-swr4-rp.cast.addradio.de/swr/swr4/rp/mp3/128/stream.mp3:8
D: Connect to new host swr-swr4-rp.cast.addradio.de/swr/swr4/rp/mp3/128/stream.mp3:80
D: Connect to swr-swr4-rp.cast.addradio.de/swr/swr4/rp/mp3/128/stream.mp3 on port 80, extension /swr/s
D: Request swr-swr4-rp.cast.addradio.de/swr/swr4/rp/mp3/128/stream.mp3:80 failed!
D: No data input
D: No data input
`
You can see: It doesn't work. Any other ideas?

Manfred

ManfredEP, 2017-07-27:
Hello,
one addition: What's that:  on port 80, extension /swr/s? I mean the 'extension' and so on. On the file it wrote: extension //Test.mp3. That looks like a copy of a part of the name I keid in. But on the radion line this is not so since there is only one part liking simular: ....de/swr/swr4.... the part with 6 letters behind .de looks like this. But if so this is senceless. Do anyone know where this come from and what is the meaning of it? Why this is written? Or is it something wrong like a buffer overflow for example. The example lines for the ini file in the package are a lot shorter than this. 

Manfred

Edzelf, 2017-07-27:
This is in my ini-file:
# Presets
preset = 0 # Start with preset 0
preset_00 = swr-swr4-rp.cast.addradio.de/swr/swr4/rp/mp3/128/stream.mp3 # 0 - SWR4

And this is the debug output:
D: New preset/file requested (-1/0) from swr-swr4-rp.cast.addradio.de/swr/swr4/rp/mp3/128/stream.mp3
D: Connect to new host swr-swr4-rp.cast.addradio.de/swr/swr4/rp/mp3/128/stream.mp3
D: Connect to swr-swr4-rp.cast.addradio.de on port 80, extension /swr/swr4/rp/mp3/128/stream.mp3
D: Connected to server
D: Switch to HEADER
D: Content-Type: audio/mpeg
D: audio/mpeg seen.
D: icy-description:SWR4. Da sind wir daheim
D: icy-genre:Varied
D: icy-name:SWR4 RP
D: icy-pub:1
D: icy-url:http://www.swr.de/swr4/rp
D: Server: dg-swr-http_dus-dtag_edge_40708f628e109c6bd708e8b067467adc
D: Cache-Control: no-cache, no-store
D: Access-Control-Allow-Origin: *
D: Access-Control-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type
D: -Allow-Headers: Origin, Accept, X-Requested-With, Content-Type seen.
D: Access-Control-Allow-Methods: GET, OPTIONS, HEAD
D: Connection: Close
D: Expires: Mon, 26 Jul 1997 05:00:00 GMT
D: icy-metaint:16000
D: Switch to DATA, bitrate is 0, metaint is 16000
D: First chunk:
D: FF FB 94 60 C4 00 03 25
D: 47 DE 61 E3 33 6A 48 E1
D: 2B CD 25 23 25 0D 21 21
D: 77 86 2C 6D A9 12 87 AF
D: Metadata block 48 bytes
D: Streamtitle found, 47 bytes
D: StreamTitle='"Sternenregen" von Thomas Anders';

Edzelf, 2017-07-28:
Tip: you can change the ini-file through the web interface on the config page.

ManfredEP, 2017-08-11:
I've got it. It works now. There was one faulty connection on SPI. Now all is fine. 

Thanks to all.

Manfred

closed 2017-08-11 by ManfredEP