Edzelf/Esp-radio #103: using oled 0.92" display

blotfi, 2017-08-09:
Hi
Thank you for sharing your project
Are they a version that uses the small oled 0.92" display?
Has anyone fork the project to use it?
thanks
Lotfi

Edzelf, 2017-08-09:
I think the display is too small to be useful for this project.

fredericplante, 2017-08-09:
To small unless you scroll the text.

blotfi, 2017-08-10:
just to display the radio canal and scroll the content information .
it is an I2C interface, and D3/D4 are available
What are the routines I have to rewrite ?
thanks
by the way your radio works great
what aout the reset circuit ? can I juste plug the RST to GND via 1k directly without the GPIO diodes

Edzelf, 2017-08-10:
Look at the code fragments after "#if defined (USETFT)".  Those are the parts that must be changed.
Consider to switch to the ESP32 version; it is much better.
The RST pin of the VS1053 can be tied to the reset pin of the ESP8266 or via a 1 k resistor to +3.3 Volt.
See also [https://github.com/Edzelf/Esp-radio/issues/32](https://github.com/Edzelf/Esp-radio/issues/32)

blotfi, 2017-08-10:
Ok, I share my code that use OLED : check defined ( USEOLED )
`
// https://github.com/olikraus/u8g2/wiki/u8g2setupcpp
U8G2_SSD1306_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ D4, /* data=*/ D3, /* reset=*/ U8X8_PIN_NONE);   // All Boards without Reset of the Display
`

but I disabled the scroll (look in the loop routine)  coz it makes the VC1053 buffer to be empty.. take too much ressources :-(
perhaps on ESP32 it will work better

here is the code
best

`//******************************************************************************************
//*  Esp_radio -- Webradio receiver for ESP8266, 1.8 color display and VS1053 MP3 module,  *
//*               by Ed Smallenburg (ed@smallenburg.nl)                                    *
//*  With ESP8266 running at 80 MHz, it is capable of handling up to 256 kb bitrate.       *
//*  With ESP8266 running at 160 MHz, it is capable of handling up to 320 kb bitrate.      *
//******************************************************************************************
// ESP8266 libraries used:
//  - ESP8266WiFi       - Part of ESP8266 Arduino default libraries.
//  - SPI               - Part of Arduino default libraries.
//  - Adafruit_GFX      - https://github.com/adafruit/Adafruit-GFX-Library
//  - TFT_ILI9163C      - https://github.com/sumotoy/TFT_ILI9163C
//  - ESPAsyncTCP       - https://github.com/me-no-dev/ESPAsyncTCP
//  - ESPAsyncWebServer - https://github.com/me-no-dev/ESPAsyncWebServer
//  - FS - https://github.com/esp8266/arduino-esp8266fs-plugin/releases/download/0.2.0/ESP8266FS-0.2.0.zip
//  - ArduinoOTA        - Part of ESP8266 Arduino default libraries.
//  - AsyncMqttClient   - https://github.com/marvinroger/async-mqtt-client
//  - TinyXML           - Fork https://github.com/adafruit/TinyXML
//
// A library for the VS1053 (for ESP8266) is not available (or not easy to find).  Therefore
// a class for this module is derived from the maniacbug library and integrated in this sketch.
//
// See http://www.internet-radio.com for suitable stations.  Add the stations of your choice
// to the .ini-file.
//
// Brief description of the program:
// First a suitable WiFi network is found and a connection is made.
// Then a connection will be made to a shoutcast server.  The server starts with some
// info in the header in readable ascii, ending with a double CRLF, like:
//  icy-name:Classic Rock Florida - SHE Radio
//  icy-genre:Classic Rock 60s 70s 80s Oldies Miami South Florida
//  icy-url:http://www.ClassicRockFLorida.com
//  content-type:audio/mpeg
//  icy-pub:1
//  icy-metaint:32768          - Metadata after 32768 bytes of MP3-data
//  icy-br:128                 - in kb/sec (for Ogg this is like "icy-br=Quality 2"
//
// After de double CRLF is received, the server starts sending mp3- or Ogg-data.  For mp3, this
// data may contain metadata (non mp3) after every "metaint" mp3 bytes.
// The metadata is empty in most cases, but if any is available the content will be presented on the TFT.
// Pushing the input button causes the player to select the next preset station present in the .ini file.
//
// The display used is a Chinese 1.8 color TFT module 128 x 160 pixels.  The TFT_ILI9163C.h
// file has been changed to reflect this particular module.  TFT_ILI9163C.cpp has been
// changed to use the full screenwidth if rotated to mode "3".  Now there is room for 26
// characters per line and 16 lines.  Software will work without installing the display.
// If no TFT is used, you may use GPIO2 and GPIO15 as control buttons.  See definition of "USETFT" below.
// Switches are than programmed as:
// GPIO2 : "Goto station 1"
// GPIO0 : "Next station"
// GPIO15: "Previous station".  Note that GPIO15 has to be LOW when starting the ESP8266.
//         The button for GPIO15 must therefore be connected to VCC (3.3V) instead of GND.

//
// For configuration of the WiFi network(s): see the global data section further on.
//
// The SPI interface for VS1053 and TFT uses hardware SPI.
//
// Wiring:
[... 3077 lines cut ...]
      }
    }
  }
  else if ( argument.startsWith ( "list" ) )            // List all presets?
  {
    dbgprint ( "list request from browser" ) ;
    request->send ( 200, "text/plain", presetlist ) ;   // Send the reply
    return ;
  }
  else
  {
    reply = analyzeCmd ( argument.c_str(),              // Analyze it
                         value.c_str() ) ;
  }
  request->send ( 200, "text/plain", reply ) ;          // Send the reply
  //t = millis() - t ;
  // If it takes too long to send a reply, we run into the "LmacRxBlk:1"-problem.
  // Reset the ESP8266.....
  //if ( t > 8000 )
  //{
  //  ESP.restart() ;                                   // Last resource
  //}
}

`

blotfi, 2017-08-12:
I also switched to ESP32 web radio version, now I can scroll the text and play radio !
ESP32 has a native I2C (not like the ESP8266 with Software emulated I2C) so my OLED is being fetched with data at lightening speed ;-)