Edzelf/Esp-radio #104: no response from web interface

PeterFF, 2017-08-11:
Dear Mr. Smallenburg

thank you upfront for your great project work. The MP3 ESP-Radio plays all my streams nicely, with TFT display and configuration via the Android App.

The issue why I am contacting you is that the web interface of the esp-radio isn’t responding.

Neither the index.html nor config.html page complete and when looking into the source-code-of-the-page it seems randomly only one or most of the time no TCP packet
arrived with the/all browser/s running into a timeout. The esp-radio log shows successful delivery of the requested html page.  Even the tiny content of the about.html is not delivered complete.
I can GET and SET individual parameters via the webbrowser only.

I did test it on ESP8266-12E, NodeMCU ESP8266-12, WeMos ESP8266 boards, (80 and 160MHz) with always same behaviour
I use Arduino 1.8.3, ESP8266 core 2.3.0 (same problem with “your” 2.2.0 core) with latest ESPAsyncTCP and ESPAsyncWebserver from github.
Surprisingly the ESPAsyncWebserver example is working fine.

Could you please advise how to get the webinterface running and might share your version of ESPAsyncTCP/Webserver (as zip), which I suspect as the source of this issue ?

Best regards
  Peter

Edzelf, 2017-08-11:
Peter,
This is a known problem, see [https://github.com/Edzelf/Esp-radio/issues/67](https://github.com/Edzelf/Esp-radio/issues/67)
Consider to switch to my ESP32 version of the radio.  The ESP32 seem to have better handling of the TCP input buffers and is much more stable.
You can find the ESPAsyncTCP webserver at [https://github.com/me-no-dev](https://github.com/me-no-dev).  I use the most recent version.

PeterFF, 2017-08-11:
Dear Ed,

thank you for the response. I have read issue #67 and thought only sometimes the webinterface fails, which would be acceptable.

Since I saw your screenshots of the webpages I thought I can use it for initial setup or updates, even if it sometime fails. But on my ESP8266 it never worked. So i will give up on the webinterface on EPS8266.

I did order an ESP32 already and will continue with this one. Nevertheless I will use the running ESP8266 esp-radio OLED in my kitchen, without web interface.

Btw.: On my second esp-radio I did attach an OLED 0'92 SH1106 via SPI to it with a frame buffer switching between different screen content every 5secs and NTP. If you like I can provide you my INO file.

Again thank you very much !

Best regards

  Peter