Edzelf/Esp-radio #105: Command timings whilst streaming a file directly.

texynz, 2017-08-16:
If playing a MP3 from a file (HTTP served), using 

?station = http:\\myserver\mymp3.mp3
or
?preset = 1 (where the saved preset is http:\\myserver\mymp3.mp3

You cannot stop the file being played by forcing another file to be streamed by setting another preset/station. No stop or forced stop is recognised. 

I have tried setting two different files in the preset, but if i play preset1, and lets say it is 3min long, and 2minutes along i say to myself, i sick of this song, lets jump to the next track. I cannot change to preset2, and force it to play preset2. I seems like the code is not looking for/of has enough time to see another "command" whilst it is streaming preset1.

No issues if you are using proper streams, rather than direct file streams.

Is this a processing issue with the es8266 which can be resolved with the higher spec esp32 or more to do wit the way the direct mp3 file is streamed.

Edzelf, 2017-08-16:
I tested it with:
preset_10 = www.smallenburg.nl/kontiki.mp3
preset_11 = www.smallenburg.nl/friendly.mp3
That works correctly.  Maybe the bitrate of your MP3 is very high.
Anyway, consider to switch to ESP32.  It is much better.

texynz, 2017-08-16:
Ok I tried it with your presets and it works, and then tried it again via and external network and it worked prefectly, but not when using it within the same environment.

using
preset_10 = 192.168.1.5/mymp3.mp3
doesnt work for forcing it to stop mid playing

but using 
preset_10 = myexternaldns/mymp3.mp3
works fine

Going to do a bit more testing. Dont think it is a bitrate issue cos same bit rate on both tests. Might be a router problem, but going to try a few things.

texynz, 2017-08-21:
Update on this. Still not sure what the actual issue is, but looks like it might have something to do with the router. I changed have tried it on a few different network setups/configs and sort of hit and miss and there is not specific pattern. 

One of the things I am seeing when it doesnt work, is when I am watching the serial logs, I dont see the "stop" command when a new preset is set, also the track seems to loop itself, it is like the loop has detected a new preset command, but hasnt got enough time to set it, so just replays the last.