Edzelf/Esp-radio #108: Problem with compilation

bnq2, 2017-10-10:
Hi everyone, i have a problem with compilation this script. I installed all librarys and i put all files in my project folder. The compiler sees the all html.h files, but during installation, the IDE tries to compile HTML (so I think).  How can I fix it? Please help, I really want to finish this great project ;] 

These are my logs  [http://wklej.org/id/3268443/](url)

Edzelf, 2017-10-10:
The link does not work.

bnq2, 2017-10-10:
sorry, its problem with this serwise, it works when you put it in browser

Edzelf, 2017-10-10:
Can you list the first 10 lines of the file about_html.h please?

bnq2, 2017-10-10:
<!DOCTYPE html>
<html lang="en">
  <head>
    <meta charset="utf-8">
  <link rel="dns-prefetch" href="https://assets-cdn.github.com">
  <link rel="dns-prefetch" href="https://avatars0.githubusercontent.com">
  <link rel="dns-prefetch" href="https://avatars1.githubusercontent.com">
  <link rel="dns-prefetch" href="https://avatars2.githubusercontent.com">
  <link rel="dns-prefetch" href="https://avatars3.githubusercontent.com">
  <link rel="dns-prefetch" href="https://github-cloud.s3.amazonaws.com">
  <link rel="dns-prefetch" href="https://user-images.githubusercontent.com/">

<!DOCTYPE html>
<html lang="en">
  <head>
    <meta charset="utf-8">
  <link rel="dns-prefetch" href="https://assets-cdn.github.com">
  <link rel="dns-prefetch" href="https://avatars0.githubusercontent.com">
  <link rel="dns-prefetch" href="https://avatars1.githubusercontent.com">
  <link rel="dns-prefetch" href="https://avatars2.githubusercontent.com">
  <link rel="dns-prefetch" href="https://avatars3.githubusercontent.com">
  <link rel="dns-prefetch" href="https://github-cloud.s3.amazonaws.com">
  <link rel="dns-prefetch" href="https://user-images.githubusercontent.com/">

bnq2, 2017-10-10:
The file starts with 8 new lines, and then :
http://wklej.org/id/3268502/

skakri, 2017-10-10:
Did you copy about_html.h or did you clone this repo? Error is related to invalid characters – https://stackoverflow.com/questions/19198332/compilation-error-stray-302-in-program-etc

Edzelf, 2017-10-10:
The file about_html.h is only 38 lines long.  You are using the wrong method to download the sourse files.

bnq2, 2017-10-10:
Bingo! I copied the text from a file in a repo to a notepad and saved it to *.h in ANSI ;] now compilation is correctly ;] tommorow i try upload it to ESP. Realy THX ;]

closed 2017-10-10 by bnq2