Edzelf/Esp-radio #109: Current version with latest libraries fails to compile

Uksa007, 2017-10-18:
Using a WeMos D1 mini getting the following compile errors.
Any ideas?

C:\Users\Uksa007\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp: In constructor 'AsyncClient::AsyncClient(tcp_pcb*)':

C:\Users\Uksa007\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:77:28: error: invalid conversion from 'long int (*)(void*, tcp_pcb*, pbuf*, long int)' to 'tcp_recv_fn {aka signed char (*)(void*, tcp_pcb*, pbuf*, signed char)}' [-fpermissive]

     tcp_recv(_pcb, &_s_recv);

                            ^

In file included from C:\Users\Uksa007\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:27:0:

C:\Users\Uksa007\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include/lwip/tcp.h:318:18: error:   initializing argument 2 of 'void tcp_recv(tcp_pcb*, tcp_recv_fn)' [-fpermissive]

 void             tcp_recv    (struct tcp_pcb *pcb, tcp_recv_fn recv) ICACHE_FLASH_ATTR;

                  ^

C:\Users\Uksa007\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:78:28: error: invalid conversion from 'long int (*)(void*, tcp_pcb*, uint16_t) {aka long int (*)(void*, tcp_pcb*, short unsigned int)}' to 'tcp_sent_fn {aka signed char (*)(void*, tcp_pcb*, short unsigned int)}' [-fpermissive]

     tcp_sent(_pcb, &_s_sent);

                            ^

In file included from C:\Users\Uksa007\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:27:0:

C:\Users\Uksa007\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include/lwip/tcp.h:319:18: error:   initializing argument 2 of 'void tcp_sent(tcp_pcb*, tcp_sent_fn)' [-fpermissive]

 void             tcp_sent    (struct tcp_pcb *pcb, tcp_sent_fn sent)ICACHE_FLASH_ATTR;

                  ^

C:\Users\Uksa007\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:79:28: error: invalid conversion from 'void (*)(void*, long int)' to 'tcp_err_fn {aka void (*)(void*, signed char)}' [-fpermissive]

     tcp_err(_pcb, &_s_error);

                            ^

In file included from C:\Users\Uksa007\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:27:0:

C:\Users\Uksa007\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include/lwip/tcp.h:321:18: error:   initializing argument 2 of 'void tcp_err(tcp_pcb*, tcp_err_fn)' [-fpermissive]

 void             tcp_err     (struct tcp_pcb *pcb, tcp_err_fn err)ICACHE_FLASH_ATTR;

                  ^

C:\Users\Uksa007\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:80:31: error: invalid conversion from 'long int (*)(void*, tcp_pcb*)' to 'tcp_poll_fn {aka signed char (*)(void*, tcp_pcb*)}' [-fpermissive]

     tcp_poll(_pcb, &_s_poll, 1);

                               ^

In file included from C:\Users\Uksa007\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:27:0:

C:\Users\Uksa007\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include/lwip/tcp.h:320:18: error:   initializing argument 2 of 'void tcp_poll(tcp_pcb*, tcp_poll_fn, u8_t)' [-fpermissive]

 void             tcp_poll    (struct tcp_pcb *pcb, tcp_poll_fn poll, u8_t interval)ICACHE_FLASH_ATTR;

                  ^

C:\Users\Uksa007\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp: In member function 'bool AsyncClient::connect(IPAddress, uint16_t)':

C:\Users\Uksa007\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:128:25: error: invalid conversion from 'void (*)(void*, long int)' to 'tcp_err_fn {aka void (*)(void*, signed char)}' [-fpermissive]

   tcp_err(pcb, &_s_error);

                         ^

In file included from C:\Users\Uksa007\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:27:0:
[... 102 lines cut ...]

C:\Users\Uksa007\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include/lwip/tcp.h:320:18: error:   initializing argument 2 of 'void tcp_poll(tcp_pcb*, tcp_poll_fn, u8_t)' [-fpermissive]

 void             tcp_poll    (struct tcp_pcb *pcb, tcp_poll_fn poll, u8_t interval)ICACHE_FLASH_ATTR;

                  ^

C:\Users\Uksa007\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp: In member function 'void AsyncServer::begin()':

C:\Users\Uksa007\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:839:30: error: invalid conversion from 'long int (*)(void*, tcp_pcb*, long int)' to 'tcp_accept_fn {aka signed char (*)(void*, tcp_pcb*, signed char)}' [-fpermissive]

   tcp_accept(_pcb, &_s_accept);

                              ^

In file included from C:\Users\Uksa007\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:27:0:

C:\Users\Uksa007\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include/lwip/tcp.h:317:18: error:   initializing argument 2 of 'void tcp_accept(tcp_pcb*, tcp_accept_fn)' [-fpermissive]

 void             tcp_accept  (struct tcp_pcb *pcb, tcp_accept_fn accept) ICACHE_FLASH_ATTR;

                  ^

exit status 1
Error compiling for board WeMos D1 R2 & mini.

Uksa007, 2017-10-18:
To answer my own question, this issuse was the library, 
I have changed the library.json back to commit 9b0cc37 but still had error found I need to implement c2d4c16 to resolve.
Why are they committing PR for 2.4_RC when they don't work with the current production release is beyone me !

pathakhcst, 2017-10-19:
I am facing this issue with NodeMCU and PlatformIO. I am using esp8266_stage. 
.piolibdeps\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp:839:30: error: invalid conversion from 'int8_t ()(void, tcp_pcb*, int8_t) {
aka signed char ()(void, tcp_pcb*, signed char)}' to 'tcp_accept_fn {aka long int ()(void, tcp_pcb*, long int)}' [-fpermiss
ive]
tcp_accept(pcb, &s_accept);
^
In file included from .piolibdeps\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp:27:0:
C:\Users\verizon.platformio\packages\framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910\tools\sdk\lwip\include/lwip/tcp.h:317:18: error: initializing argument 2 of 'void tcp_accept(tcp_pcb*, tcp_accept_fn)' [-fpermissive]
void tcp_accept (struct tcp_pcb *pcb, tcp_accept_fn accept) ICACHE_FLASH_ATTR;
^
*** [.pioenvs\nossl\lib\ESPAsyncTCP_ID305\ESPAsyncTCP.o] Error 1

Edzelf, 2017-10-19:
Did you check  https://github.com/me-no-dev/ESPAsyncTCP/issues/60 ?

pathakhcst, 2017-10-19:
I saw it but I don't understand what to do to resolve the issue? I am using platformio ide.

Edzelf, 2017-11-13:
I used [this](https://github.com/me-no-dev/ESPAsyncTCP/commit/c2d4c168d41b78d9abd71a074c80eb8087c33561) to correct it.