Edzelf/Esp-radio #110: Station playback stops every 30 minutes

RonWei, 2017-10-30:
Every 30 minutes the playback of the current radiostation stops/freeze. After a moment the firmware switches to the next radiostation in the programmlist and continue for another 30 minutes of playback. And so on ...

Edzelf, 2017-10-30:
The radio will switch over if the data stream stops for a longer time.  Maybe your internet connection is very busy every 30 minutes.  You may have some clues in the logging on the serial output.

RonWei, 2017-10-30:
Thx for your answer. I think here is the code part that skips the station ever 30 minutes.

 I will investigate my traffic to find out why

RonWei, 2017-10-30:
Thx for your answer. I think this code part skips the station after 30 minutes.
```

      if ( ( morethanonce > 0 ) ||                // Happened more than once?
           ( playlist_num > 0 ) )                 // Or playlist active?
      {
        datamode = STOPREQD ;                     // Stop player
//        ini_block.newpreset++ ;                 // Yes, try next channel

        dbgprint ( "Trying other station/file..." ) ;
      }
      morethanonce++ ;                            // Count the fails
```

I have to investigate my traffic to find out why the datastream stops or i have to rewrite the code to reconnect to the same (not the next) radio station. Should a " hostreq = true ;  " is enough to reconnect? 

Best regards

Edzelf, 2017-10-30:
Ron, If any data is received, morethanonce will be reset to 0.  You will see in the logging:
"Recovered from dataloss" and the radio will keep playing the same station.
Normally a reconnect is not the solution, as in most cases the station is not reachable.  That's why the radio will try the next station.  Setting hostreq to true will not reconnect as the host has not changed.
So again, look into the logging to see what's going on.
You will probably see "No data input" several times before the radio gives up.

RonWei, 2017-10-30:
Hello Edzelf, thats right, i checked the log. It seems to be a problem with the radio station i heard. Sometimes no data can be received for a couple of seconds from this station and then your firmware get a "timeout" and switch to the next station. 
I modified the code so the radio will continue to play the same channel when new data is received after a few seconds.  In my opinion thats a better option than changing the channel.

closed 2017-10-30 by RonWei

Edzelf, 2017-10-30:
In that case, just increase the number 0 in line 962.