Edzelf/Esp-radio #111: VS1053 error retry SB:4X R1:4X R2:4X

Barrois, 2017-11-15:
Hello,
I want to do an cheap web radio with the "Minimal version".
First I want to test with your : SimpleMp3PlayerWithDebug.ino 
but the return is   :

N: Playing sound... 0⸮~?⸮4⸮!⸮⸮⸮⸮_SAA⸮⸮N: Hello VS1053!
N: Reset VS1053...
N: End reset VS1053...
N: Slow SPI,Testing VS1053 read/write registers...
E: VS1053 error retry SB:4X R1:4X R2:4X
E: VS1053 error retry SB:4X R1:4X R2:4X
E: VS1053 error retry SB:4X R1:4X R2:4X
E: VS1053 error retry SB:4X R1:4X R2:4X
E: VS1053 error retry SB:4X R1:4X R2:4X
E: VS1053 error retry SB:4X R1:4X R2:4X
E: VS1053 error retry SB:4X R1:4X R2:4X
E: VS1053 error retry SB:4X R1:4X R2:4X
E: VS1053 error retry SB:4X R1:4X R2:4X
E: VS1053 error retry SB:4X R1:4X R2:4X
E: VS1053 error retry SB:4X R1:4X R2:4X
E: VS1053 error retry SB:4X R1:4X R2:4X
E: VS1053 error retry SB:4X R1:4X R2:4X
E: VS1053 error retry SB:4X R1:4X R2:4X
E: VS1053 error retry SB:4X R1:4X R2:4X
E: VS1053 error retry SB:4X R1:4X R2:4X
E: VS1053 error retry SB:4X R1:4X R2:4X
E: VS1053 error retry SB:4X R1:4X R2:4X
E: VS1053 error retry SB:4X R1:4X R2:4X
E: VS1053 error retry SB:4X R1:4X R2:4X
N: Fast SPI, Testing VS1053 read/write registers again...
E: VS1053 error retry SB:4X R1:4X R2:4X
E: VS1053 error retry SB:4X R1:4X R2:4X
E: VS1053 error retry SB:4X R1:4X R2:4X
E: VS1053 error retry SB:4X R1:4X R2:4X
E: VS1053 error retry SB:4X R1:4X R2:4X
E: VS1053 error retry SB:4X R1:4X R2:4X
E: VS1053 error retry SB:4X R1:4X R2:4X
E: VS1053 error retry SB:4X R1:4X R2:4X
E: VS1053 error retry SB:4X R1:4X R2:4X
E: VS1053 error retry SB:4X R1:4X R2:4X
E: VS1053 error retry SB:4X R1:4X R2:4X
E: VS1053 error retry SB:4X R1:4X R2:4X
E: VS1053 error retry SB:4X R1:4X R2:4X
E: VS1053 error retry SB:4X R1:4X R2:4X
E: VS1053 error retry SB:4X R1:4X R2:4X
E: VS1053 error retry SB:4X R1:4X R2:4X
E: VS1053 error retry SB:4X R1:4X R2:4X
E: VS1053 error retry SB:4X R1:4X R2:4X
E: VS1053 error retry SB:4X R1:4X R2:4X
E: VS1053 error retry SB:4X R1:4X R2:4X
N: endFillByte is 0xFF
N: Playing sound... 

can you help me ?

Edzelf, 2017-11-15:
SimpleMp3PlayerWithDebug.ino is not mine....

Barrois, 2017-11-17:
Yes you are right ,  désolé et encore merci.
Have you a small .ino for only a ESP and a VS1053 just to ear some web radio  with 5 botton : + -, stop , radio+, radio - ?
Something little for the "Minimal version"

Edzelf, 2017-11-17:
The current version can do just that, so I do not see the need for a special version.

Barrois, 2017-11-20:
Merci.
but I got a error  on ESPAsyncTCP.cpp  
sure is not yourse !

C:\Users\Bernard\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp: In constructor 'AsyncClient::AsyncClient(tcp_pcb*)':

C:\Users\Bernard\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:78:28: error: invalid conversion from 'long int (*)(void*, tcp_pcb*, pbuf*, long int)' to 'tcp_recv_fn {aka signed char (*)(void*, tcp_pcb*, pbuf*, signed char)}' [-fpermissive]

     tcp_recv(_pcb, &_s_recv);

                            ^

In file included from C:\Users\Bernard\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:27:0:

C:\Users\Bernard\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2/tools/sdk/lwip/include/lwip/tcp.h:318:18: error:   initializing argument 2 of 'void tcp_recv(tcp_pcb*, tcp_recv_fn)' [-fpermissive]

 void             tcp_recv    (struct tcp_pcb *pcb, tcp_recv_fn recv) ICACHE_FLASH_ATTR;

                  ^

C:\Users\Bernard\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:79:28: error: invalid conversion from 'long int (*)(void*, tcp_pcb*, uint16_t) {aka long int (*)(void*, tcp_pcb*, short unsigned int)}' to 'tcp_sent_fn {aka signed char (*)(void*, tcp_pcb*, short unsigned int)}' [-fpermissive]

     tcp_sent(_pcb, &_s_sent);

                            ^

In file included from C:\Users\Bernard\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:27:0:

C:\Users\Bernard\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2/tools/sdk/lwip/include/lwip/tcp.h:319:18: error:   initializing argument 2 of 'void tcp_sent(tcp_pcb*, tcp_sent_fn)' [-fpermissive]

 void             tcp_sent    (struct tcp_pcb *pcb, tcp_sent_fn sent)ICACHE_FLASH_ATTR;

                  ^

C:\Users\Bernard\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:80:28: error: invalid conversion from 'void (*)(void*, long int)' to 'tcp_err_fn {aka void (*)(void*, signed char)}' [-fpermissive]

     tcp_err(_pcb, &_s_error);

                            ^

In file included from C:\Users\Bernard\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:27:0:

C:\Users\Bernard\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2/tools/sdk/lwip/include/lwip/tcp.h:321:18: error:   initializing argument 2 of 'void tcp_err(tcp_pcb*, tcp_err_fn)' [-fpermissive]

 void             tcp_err     (struct tcp_pcb *pcb, tcp_err_fn err)ICACHE_FLASH_ATTR;

                  ^

C:\Users\Bernard\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:81:31: error: invalid conversion from 'long int (*)(void*, tcp_pcb*)' to 'tcp_poll_fn {aka signed char (*)(void*, tcp_pcb*)}' [-fpermissive]

     tcp_poll(_pcb, &_s_poll, 1);

                               ^

In file included from C:\Users\Bernard\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:27:0:

C:\Users\Bernard\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2/tools/sdk/lwip/include/lwip/tcp.h:320:18: error:   initializing argument 2 of 'void tcp_poll(tcp_pcb*, tcp_poll_fn, u8_t)' [-fpermissive]

 void             tcp_poll    (struct tcp_pcb *pcb, tcp_poll_fn poll, u8_t interval)ICACHE_FLASH_ATTR;

                  ^

C:\Users\Bernard\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp: In member function 'bool AsyncClient::connect(IPAddress, uint16_t)':

C:\Users\Bernard\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:130:25: error: invalid conversion from 'void (*)(void*, long int)' to 'tcp_err_fn {aka void (*)(void*, signed char)}' [-fpermissive]

   tcp_err(pcb, &_s_error);

                         ^

[... 103 lines cut ...]

C:\Users\Bernard\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2/tools/sdk/lwip/include/lwip/tcp.h:320:18: error:   initializing argument 2 of 'void tcp_poll(tcp_pcb*, tcp_poll_fn, u8_t)' [-fpermissive]

 void             tcp_poll    (struct tcp_pcb *pcb, tcp_poll_fn poll, u8_t interval)ICACHE_FLASH_ATTR;

                  ^

C:\Users\Bernard\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp: In member function 'void AsyncServer::begin()':

C:\Users\Bernard\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:848:30: error: invalid conversion from 'long int (*)(void*, tcp_pcb*, long int)' to 'tcp_accept_fn {aka signed char (*)(void*, tcp_pcb*, signed char)}' [-fpermissive]

   tcp_accept(_pcb, &_s_accept);

                              ^

In file included from C:\Users\Bernard\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:27:0:

C:\Users\Bernard\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2/tools/sdk/lwip/include/lwip/tcp.h:317:18: error:   initializing argument 2 of 'void tcp_accept(tcp_pcb*, tcp_accept_fn)' [-fpermissive]

 void             tcp_accept  (struct tcp_pcb *pcb, tcp_accept_fn accept) ICACHE_FLASH_ATTR;

                  ^

exit status 1
Erreur de compilation pour la carte WeMos D1 R2 & mini

Edzelf, 2017-11-20:
Solved, see issue 109.