Edzelf/Esp-radio #113: Compilation error

lkomlosi, 2017-11-18:
I've tried to compile everything on Arduino 1.8.5 and 2.3.0 ESP core.
But there are still errors.
Can somebody help me.

Leon
C:\Users\Leon\Documents\Arduino\esp_radio\Esp_radio.ino: In function 'void listNetworks()':

Esp_radio:925: error: request for member 'c_str' in 'WiFi.WiFiClass::SSID(((int)((uint8_t)i)))', which is of non-class type 'char*'

                i + 1, WiFi.SSID ( i ).c_str(), WiFi.RSSI ( i ),

                                       ^

C:\Users\Leon\Documents\Arduino\esp_radio\Esp_radio.ino: In function 'bool connectwifi()':

Esp_radio:1387: error: 'class WiFiClass' has no member named 'softAPdisconnect'

   WiFi.softAPdisconnect(true) ;                        // still keep the old connection

        ^

Esp_radio:1388: error: invalid conversion from 'const char*' to 'char*' [-fpermissive]

   WiFi.begin ( ini_block.ssid.c_str(),

                                     ^

In file included from C:\Users\Leon\Documents\Arduino\esp_radio\Esp_radio.ino:1:0:

C:\Program Files (x86)\Arduino\libraries\WiFi\src/WiFi.h:79:9: error:   initializing argument 1 of 'int WiFiClass::begin(char*, const char*)' [-fpermissive]

     int begin(char* ssid, const char *passphrase);

         ^

Esp_radio:1391: error: 'class WiFiClass' has no member named 'waitForConnectResult'

   if (  WiFi.waitForConnectResult() != WL_CONNECTED )  // Try to connect

              ^

Esp_radio:1394: error: 'class WiFiClass' has no member named 'softAP'

     WiFi.softAP ( NAME, NAME ) ;                       // This ESP will be an AP

          ^

C:\Users\Leon\Documents\Arduino\esp_radio\Esp_radio.ino: In function 'void setup()':

Esp_radio:1800: error: 'class WiFiClass' has no member named 'persistent'

   WiFi.persistent ( false ) ;                          // Do not save SSID and password

        ^

Esp_radio:1802: error: 'class WiFiClass' has no member named 'mode'

   WiFi.mode ( WIFI_STA ) ;                             // This ESP is a station

        ^

Esp_radio:1802: error: 'WIFI_STA' was not declared in this scope

   WiFi.mode ( WIFI_STA ) ;                             // This ESP is a station

koskee, 2017-11-19:
I would guess this as a library issue. It looks like it might be including the wifi library from another architecture, such as the default arduino library provided with the IDE perhaps. 

If you enable verbose compilation output in the preferences, it should spit out a few lines in there somewhere telling you which libraries it is selecting.

I'll let someone else chime in on the best way to force specific library selection during compilation, as I've not ever known an elegant way of dealing with this, so I might learn something as well.

lkomlosi, 2017-11-19:
Verbose output
L.

Compiling sketch...
"C:\Users\Leon\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Leon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\Leon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\Leon\AppData\Local\Temp\arduino_build_328935/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=160000000L -DLWIP_OPEN_SRC   -DARDUINO=10805 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-IC:\Users\Leon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Users\Leon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\generic" "-IC:\Program Files (x86)\Arduino\libraries\WiFi\src" "-IC:\Users\Leon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src" "-IC:\Users\Leon\Documents\Arduino\libraries\ESPAsyncTCP-master\src" "-IC:\Users\Leon\Documents\Arduino\libraries\ESPAsyncWebServer-master\src" "-IC:\Users\Leon\Documents\Arduino\libraries\async-mqtt-client-master\src" "-IC:\Users\Leon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\SPI" "-IC:\Users\Leon\Documents\Arduino\libraries\Adafruit-GFX-Library-master" "-IC:\Users\Leon\Documents\Arduino\libraries\TFT_ILI9163C-master" "-IC:\Users\Leon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\Ticker" "-IC:\Users\Leon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ArduinoOTA" "-IC:\Users\Leon\Documents\Arduino\libraries\TinyXML-master" "-IC:\Users\Leon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\Hash\src" "-IC:\Users\Leon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266mDNS" "C:\Users\Leon\AppData\Local\Temp\arduino_build_328935\sketch\Esp_radio.ino.cpp" -o "C:\Users\Leon\AppData\Local\Temp\arduino_build_328935\sketch\Esp_radio.ino.cpp.o"
C:\Users\Leon\Documents\Arduino\esp_radio\Esp_radio.ino: In function 'void listNetworks()':

Esp_radio:925: error: request for member 'c_str' in 'WiFi.WiFiClass::SSID(((int)((uint8_t)i)))', which is of non-class type 'char*'

                i + 1, WiFi.SSID ( i ).c_str(), WiFi.RSSI ( i ),

                                       ^

C:\Users\Leon\Documents\Arduino\esp_radio\Esp_radio.ino: In function 'bool connectwifi()':

Esp_radio:1387: error: 'class WiFiClass' has no member named 'softAPdisconnect'

   WiFi.softAPdisconnect(true) ;                        // still keep the old connection

        ^

Esp_radio:1388: error: invalid conversion from 'const char*' to 'char*' [-fpermissive]

   WiFi.begin ( ini_block.ssid.c_str(),

                                     ^

In file included from C:\Users\Leon\Documents\Arduino\esp_radio\Esp_radio.ino:1:0:

C:\Program Files (x86)\Arduino\libraries\WiFi\src/WiFi.h:79:9: error:   initializing argument 1 of 'int WiFiClass::begin(char*, const char*)' [-fpermissive]

     int begin(char* ssid, const char *passphrase);

         ^

Esp_radio:1391: error: 'class WiFiClass' has no member named 'waitForConnectResult'

   if (  WiFi.waitForConnectResult() != WL_CONNECTED )  // Try to connect

              ^

Esp_radio:1394: error: 'class WiFiClass' has no member named 'softAP'

     WiFi.softAP ( NAME, NAME ) ;                       // This ESP will be an AP

          ^

C:\Users\Leon\Documents\Arduino\esp_radio\Esp_radio.ino: In function 'void setup()':

Esp_radio:1800: error: 'class WiFiClass' has no member named 'persistent'

   WiFi.persistent ( false ) ;                          // Do not save SSID and password

        ^

Esp_radio:1802: error: 'class WiFiClass' has no member named 'mode'

   WiFi.mode ( WIFI_STA ) ;                             // This ESP is a station

        ^

Esp_radio:1802: error: 'WIFI_STA' was not declared in this scope

   WiFi.mode ( WIFI_STA ) ;                             // This ESP is a station

               ^

Using library WiFi at version 1.2.7 in folder: C:\Program Files (x86)\Arduino\libraries\WiFi 
Using library ESP8266WiFi at version 1.0 in folder: C:\Users\Leon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi 
Using library ESPAsyncTCP-master at version 1.1.0 in folder: C:\Users\Leon\Documents\Arduino\libraries\ESPAsyncTCP-master 
Using library ESPAsyncWebServer-master at version 1.1.0 in folder: C:\Users\Leon\Documents\Arduino\libraries\ESPAsyncWebServer-master 
Using library async-mqtt-client-master at version 0.8.1 in folder: C:\Users\Leon\Documents\Arduino\libraries\async-mqtt-client-master 
Using library SPI at version 1.0 in folder: C:\Users\Leon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\SPI 
Using library Adafruit-GFX-Library-master at version 1.2.2 in folder: C:\Users\Leon\Documents\Arduino\libraries\Adafruit-GFX-Library-master 
Using library TFT_ILI9163C-master at version 0.9 in folder: C:\Users\Leon\Documents\Arduino\libraries\TFT_ILI9163C-master 
Using library Ticker at version 1.0 in folder: C:\Users\Leon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\Ticker 
[... cut ...]
Using library TinyXML-master at version 1.0.0 in folder: C:\Users\Leon\Documents\Arduino\libraries\TinyXML-master 
Using library Hash at version 1.0 in folder: C:\Users\Leon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\Hash 
Using library ESP8266mDNS in folder: C:\Users\Leon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266mDNS (legacy)
exit status 1
request for member 'c_str' in 'WiFi.WiFiClass::SSID(((int)((uint8_t)i)))', which is of non-class type 'char*'

Edzelf, 2017-11-19:
I would try to  remove C:\Program Files (x86)\Arduino\libraries\WiFi.

koskee, 2017-11-19:
Yeah I'm pretty sure it should be using the ESP8266WiFi library. I wonder why it's also including the arduino version. I don't see it included in the code, maybe one of the other libraries contains an include for it? They're completely different file names, so that's strange.

I have come across weird errors like this before with the esp32 version (arduino IDE couldn't locate libraries that it was previously able to find, and nothing had changed). In the end,  I ended up reinstalling everything just to be able to flash my board again. Pretty sure that issue is something to do with the IDE itself, and not the code.

lkomlosi, 2017-11-19:
Remove C:\Program Files (x86)\Arduino\libraries\WiFi.
Change TCPAsync.cpp.
Compiled.

Thanks Leon

closed 2017-11-19 by lkomlosi