Edzelf/Esp-radio #114: ESP- radio with Nextion display

middelbh, 2017-12-13:
I succeeded to use your program with my Wemos mini d1. Works great!
Now I want to use a Nextion display as a user interface, allowing touch control. Whatever I try, the Nextion will not react..
if I use the program below separately it works, but including it in Esp-radio there is no reaction in my textbox (t0).

#include <SoftwareSerial.h>
#include <Nextion.h>
SoftwareSerial nextion(d3,d4);// Nextion TX to pin d3 and RX to pin d4 of Wemos d1 mini
Nextion myNextion(nextion, 9600);

in Setup:
myNextion.init(); 
myNextion.setComponentText("t0","4321");//symply to test if info is sent to the display

Hope anyone can help!

Edzelf, 2017-12-13:
Did you remove the references to d3 and d4 in the radio software?
Does SoftwareSerial work in combination with the normal Serial I/O?

middelbh, 2017-12-13:
As far as I know I removed all references to d3 and d4, and normally SoftwareSerial works together with normal serial.. I will do a separate test for this lateron..

middelbh, 2017-12-13:
By the way, what possibilities exist to change your system into a library? That would make programs more clear to debug...

middelbh, 2017-12-13:
@Edzelf, I tested, Serial and SoftwareSerial do work in combination...

Edzelf, 2017-12-13:
I ordered one, so I can test (in a couple of weeks).
I'm not planning to turn the sketch into a library.

Edzelf, 2017-12-13:
Which Nection library are you using?

Edzelf, 2017-12-13:
The description of this library does not mention the ESP8266 as a supported board.

middelbh, 2017-12-13:
Correct, but I used it in several other test situations without problems... just check the right GPIO’s and it works. What I will try tomorrow is attach another arduino to the serial ports that go to my nextion and see if teh softwareserial works...

Edzelf, 2017-12-13:
Good idea.

middelbh, 2017-12-14:
Tested, but looks like softwareserial does not work in Esp-radio. I will try to find the cause..

middelbh, 2017-12-18:
Got it working now, I will have to cleanup my sketch, I will then show what has to be done to the program to get it working.
Grtz, Hans

middelbh, 2017-12-27:
in the initialisation of the program I added:
int d1=5;
int d2=4;
int d3=0;
int d4=2;
int d5=14;
int d6=12;
int d7=13;
int d8=15;
and
#include <SoftwareSerial.h>
#include <Nextion.h>
SoftwareSerial displayPort = SoftwareSerial(d4,d3);// Nextion RX to pin d3 and TX to pin d4 of 	Arduino
Nextion myNextion(displayPort, 9600);

Change Pins for VS1053 module: as my wemos d1 with battery does not have a d0 but a d8
#define VS1053_DCS    d8 

Delete:
#define TFT_CS 15
#define TFT_DC 2
and
#define BUTTON1 2
#define BUTTON2 0
#define BUTTON3 15

In TIMER100  remove all regarding BUTTON1, BUTTON2 and BUTTON3

In SHOWSTREAMTITLE comment out:
  if ( ( p1 = strstr ( streamtitle, " - " ) ) ) // look for artist/title separator
  {

    *p1++ = '  ' ; // Found: replace 3 characters by newline

    p2 = p1 + 2 ;
    if ( *p2 == ' ' )                           // Leading space in title?
    {
      p2++ ;
    }
    strcpy ( p1, p2 ) ;                         // Shift 2nd part of title 2 or 3 places
  }

  displayinfo ( streamtitle, 20, 40, CYAN ) ;   // Show title at position 20

And replace by:

  myNextion.setComponentText("g0",streamtitle);  //I've a scrolling text on g0 in Nextion
  myNextion.setComponentText("t0",String(vs1053player.getVolume()));//I've a text at t0
  myNextion.setComponentValue("z0",vs1053player.getVolume());  //I've a dial at z0

In SETUP :

add in the beginning:

displayPort.begin(9600);
myNextion.init();  
myNextion.setComponentText("g0","Radio started");

change system_update_cpu_freq ( 160 ) ; 
to
system_update_cpu_freq ( 80 ) ; 

and delete the lines:

pinMode ( BUTTON2, INPUT_PULLUP ) ;  
pinMode ( BUTTON1, INPUT_PULLUP ) ; // Input for control button 1
pinMode ( BUTTON3, INPUT_PULLUP ) ;// Input for control button 3

manddar, 2018-05-15:
Hi @middelbh ,

I recently set up ESP Radio without display, but I have one Nextion display in storage. Is it possible for you to upload ino file and describe changes made in pin connections?

TIA - Mandar