Edzelf/Esp-radio #115: Web Interface Issues in AP mode

jgrulich, 2018-01-08:
I've found several issues when there is no correct WiFi setting and it switches into AP mode.
1) The web server doesn't work at all in secure AP mode, need to be used only open mode.
2) The radio.ini file is limited to 30 lines only, otherwise it is note saved and is cleared and the serial reports the following and freezes:

_D: /radio.ini saved

Exception (28):
epc1=0x4000bf0e epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffc20 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffdc0:  00000001 3fff989c 3ffffdf0 3ffec608  
3ffffdd0:  00000001 3fff989c 3ffffe08 40215b4a  
3ffffde0:  00000001 3fff989c 3fff98d0 4020dc43  
3ffffdf0:  00000000 00000000 00000000 00000000  
3ffffe00:  00000000 00000000 3fffb3fc 0000067f  
3ffffe10:  00000675 3fffa0f4 0000000f 00000007  
3ffffe20:  3fff40e4 3fff9774 3fff9774 00000030  
3ffffe30:  3ffec839 3ffec83a 3fff989c 4020e4d7  
3ffffe40:  00000000 00000006 3fffacfc 4023a1e0  
3ffffe50:  3fffa29c 00000000 3fff9b54 401074e4  
3ffffe60:  3fff9774 3fff9778 00000006 00000001  
3ffffe70:  00000000 b1190000 3fff9774 3fff9d88  
3ffffe80:  00000001 3fff40e4 3fff9d34 4020e5a9  
3ffffe90:  00000001 3fff40e4 3fff9d34 4020c1ac  
3ffffea0:  3fff9774 00000b68 3fff9774 40237b76  
3ffffeb0:  3fff3104 3fff3110 00000006 3fff2f18  
3ffffec0:  3fff9774 3fff2f20 3fff2f1c 4020c1ff  
3ffffed0:  3fff9774 3fff2f20 3fff2f1c 4023b9fc  
3ffffee0:  00000232 0104a8c0 00000018 00000232  
3ffffef0:  00000018 00000000 000000fc 402363f1  
3fffff00:  40220000 00000000 00000064 3fff3108  
3fffff10:  3ffec5e0 3fff3110 3fff40e4 40239ef9  
3fffff20:  3fff2e14 3fffa29c 3fffa29c 4021e8d2  
3fffff30:  4021edc6 3ffef130 3ffec5c0 3fffa29c  
3fffff40:  3ffec5d2 00000000 3fff40e4 402392f5  
3fffff50:  0204a8c0 00000001 4010510a 3ffef868  
3fffff60:  3ffeee10 40101ab1 00000000 00000005  
3fffff70:  00000002 40101843 00000002 00000000  
3fffff80:  4010187e 00000002 00000000 3fff3a74  
3fffff90:  3fffdc90 00000000 3fff40e4 4022ddaf  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)_

3) The reset command reports this and freezes:

_D: Command: reset (without parameter)

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset_

I've tested on several EPS modules with the same result.
I'm using Arduino 1.8.5. , ESP8266 v. 2.3.0, the rest libs are the latest version.
May someone double check this?

Edzelf, 2018-01-09:
See issue #59

jgrulich, 2018-01-09:
The issue #59 is related to the CLI mode when the MP3 is played.
But I'm reporting issues which happens in AP mode, when the MP3 is not played. But seems to me that in this AP mode, even more when the secure AP is used, there is no more resources left for the web, even when only one connection is open.

jgrulich, 2018-01-09:
Basically, this is following of the issue #74 with additional not working points.
I know that there is ESP-32 version, but it's too expensive.
I like this ESP8266 based version, which is simple and looks reasonably and therefor like to bring it working.