Edzelf/Esp-radio #116: Problem with presets

amigafan, 2018-01-09:
Excellent project, however, I have a problem with presets in the config file.
When I add a station eg "http://www.classicfm.nl:80/player/classicfm.m3u"
to the preset list and select then it from Control page, it does not work.

This some of the error from the serial monitor:

: New preset/file requested (6/0) from http://www.classicfm.nl:80/player/classicfm.m3u
D: Connect to new host http://www.classicfm.nl:80/player/classicfm.m3u
D: Playlist request, entry 1
D: Connect to http on port 0, extension //www.classicfm.nl:80/player/classicfm.m3u
D: Request http://www.classicfm.nl:80/player/classicfm.m3u failed!

However, if I enter the station manual in the Control page and press PLAY it works.

So every time I want to listen to this station I have to type
http://www.classicfm.nl:80/player/classicfm.m3u and press PLAY.

Are there any who have a solution?

Edzelf, 2018-01-09:
I added:
preset_11 = www.classicfm.nl/player/classicfm.m3u
To the preset list, so without the "http:" and the ":80" and it works normally.

amigafan, 2018-01-09:
Thanks, now it works perfectly.
But I can not help wondering why it worked!