Edzelf/Esp-radio #118: ESP8266-2.4.0 core libs are not working

jgrulich, 2018-01-09:
I've updated the ESP8266 core libs to the 2.4.0 and found that it loads the ESP too much and it results in the poor WiFi connection and stream data loss.
I've tested even both rc1 and rc2 with the same result.

mischaka, 2018-01-16:
I have the same problem

Edzelf, 2018-01-19:
I have the problem also.  I will try to solve it.

Edzelf, 2018-01-19:
In the IDE: Select V1.4 Prebuilt as IwIP Variant.  256 kbps stations will work now.

jgrulich, 2018-01-19:
Not really. I've already tested all the IwIP variants. With all the 2.4.0 versions is the connection very instable and with little bit lower signal level, or in the overloaded WiFi area it fails to connect to some stations at all.
And on the other hand the 2.4.0 lib results in the larger bin file and it loads the CPU significantly more.

gitpeut, 2018-05-18:
I upgraded to ESP8266 v2.4.1. In 2.4.1 V1.4 Prebuilt is not available. 
After some struggling, I found that compiling with lwip V2.0 Lower Memory worked. 
I also set cpu frequency to 160, but I am not sure how much of a difference that makes.