Edzelf/Esp-radio #119: Posion of display text

amigafan, 2018-01-09:
I have a 128 x 160 pixel display connected and it works fine. But sometimes the metadata text overlaps the station's name. Is it possible to move the station name down, there is plenty of space?

Edzelf, 2018-01-09:
Yes, it is easy to change the position.

amigafan, 2018-01-09:
How?

jgrulich, 2018-01-10:
Already solved in my version. There different TFT Lib and therefor better text font and changed completely the text layout.
Look here: https://github.com/jgrulich/Esp-radio.

amigafan, 2018-01-10:
I I really like your layout, but for some reason I can not connect to any stations! It keeps trying but it does not succeed.

jgrulich, 2018-01-10:
This happened to me too with the ESP8266 core 2.4.0. Working is only 2.3.0.
Or reflash the radio.ini once again, sometimes it needs hard reset, or reflash program ...
Really don't know what happens there in this "Black box" called ESP.
I've also found very useful disabling of ArduinoOTA, which makes it much more stable and web response faster. Try this version: https://github.com/jgrulich/Esp-radio/tree/OTA-Disabled.

amigafan, 2018-01-11:
Thank you, I changed to version 2.3.0 and now everything works perfectly

closed 2018-01-12 by amigafan