Edzelf/Esp-radio #120: Esp reset

mat3511, 2018-01-17:
I have this problem:
D: Starting ESP Version Wed, 31 May 2017 12:35:00 GMT...  Free memory 10504
D: Sketch size 387840, free size 659456
D: Reset VS1053...
D: End reset VS1053...
D: Slow SPI,Testing VS1053 read/write registers...
D: Fast SPI, Testing VS1053 read/write registers again...
D: endFillByte is 0
D: Selected network: ZTE-6ZKV3X               
D: Try WiFi ZTE-6ZKV3X
D: IP = 192.168.1.12
D: Start server for commands

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld

Does anyone know what can cause it?

Edzelf, 2018-01-17:
Is your power supply strong enough?

ajitam, 2018-02-15:
@mat3511 do you by any chance running macOS High Sierra?
A have the same problem.
But when I upload on my notebook (macOS Sierra) everything works.

I know it's a long shot but still... I'll do some further analysis and post here

ajitam, 2018-02-16:
0K, here is what I found out so far.

What I did:

* a check what was the difference between computer (alts call it *HS*) on which I get a restart (macOS **High** Sierra) and computer (lets call it *S*) I can (macOS Sierra)

* od computer *S* I had board version `2.3.0` so I downgraded version on computer *HS* ☛ uploading error

* I then searched solution for uploading error on High Sierra and I found out that there could be a FTDI driver problem ☛ [https://github.com/esp8266/Arduino/issues/3711](Issue)

* I updated the [http://www.ftdichip.com/Drivers/VCP.htm](driver), restart ESP, restart Arduino IDE ,
 reuploaded `data.ini`, recompile and reupload sketch ☛ it worked!

Few more things:
* uploading `data.ini` with old drivers took about 5min, whit new driver - few seconds
* ESP always crashed after line `D: Start server for commands`. I'm guessing this is the point where MQTT connection is made.
* I will try to go back to `2.4.0` and see if problem comes back

I have no idea if this is OS / board version / FTDI driver or combination of any of those.

Edzelf, 2018-02-16:
Try to use a powersupply that can deliver at least 1 ampere.

ajitam, 2018-02-16:
I don't think that the power supply is the issue here. I'm running on the same setup for a few months with now problems. Problems started to emerge when I switch env/configuration

mat3511, 2018-02-16:
Hello, I am runnig win 10, but I slolved this problem by putting comment symbols in this part of code:
 **if ( NetworkFound )                                  // OTA and MQTT only if Wifi network found
  {
    ArduinoOTA.setHostname ( NAME ) ;                  // Set the hostname
    ArduinoOTA.onStart ( otastart ) ;
    ArduinoOTA.begin() ;                               // Allow update over the air
    if ( ini_block.mqttbroker.length() )               // Broker specified?
    {
      // Initialize the MQTT client
      WiFi.hostByName ( ini_block.mqttbroker.c_str(),
                        mqtt_server_IP ) ;             // Lookup IP of MQTT server
      mqttclient.onConnect ( onMqttConnect ) ;
      mqttclient.onDisconnect ( onMqttDisconnect ) ;
      mqttclient.onSubscribe ( onMqttSubscribe ) ;
      mqttclient.onUnsubscribe ( onMqttUnsubscribe ) ;
      mqttclient.onMessage ( onMqttMessage ) ;
      mqttclient.onPublish ( onMqttPublish ) ;
      mqttclient.setServer ( mqtt_server_IP,           // Specify the broker
                             ini_block.mqttport ) ;    // And the port
      mqttclient.setCredentials ( ini_block.mqttuser.c_str(),
                                  ini_block.mqttpasswd.c_str() ) ;
      mqttclient.setClientId ( NAME ) ;
      dbgprint ( "Connecting to MQTT %s, port %d, user %s, password %s...",
                 ini_block.mqttbroker.c_str(),
                 ini_block.mqttport,
                 ini_block.mqttuser.c_str(),
                 ini_block.mqttpasswd.c_str() ) ;
      mqttclient.connect();
    }
  }
  else
  {
    currentpreset = ini_block.newpreset ;              // No network: do not start radio
  }**
It worked, I am not using OTA or MQTT anyway

Ps: I also did change my pover supply, becouse the old one wasn't strong enough.

ajitam, 2018-02-16:
Hmm ok I guess this problem is all over the place :) Should we just close this issue for now?

mat3511, 2018-02-16:
Yes, we should. :)

Edzelf, 2018-02-16:
The latest version of the library (rc2) has a working OTA.

nixnuex, 2018-03-03:
Had the same issue with Arduino 1.8.5 on macOS High Sierra with Arduino ESP board version 2.4.0 and a china ESP8266 NodeMCU v3 with CH340G chip.

The solution was to downgrade the Arduino ESP board version to 2.3.0