Edzelf/Esp-radio #122: adding oled vu meters

perlix, 2018-01-26:
Just built this radio and it works flawlessly! As a finishing touch, I would like to add two small oled displays for emulating analog vu meters. I'm already using a second (serial connected) Esp for driving the TFT display, so there's enough free pins for the (I2C) oleds.

I have a working sketch for the vu meters, and know how to wire an external audio signal with a voltage divider for DC bias. However, applying this wiring for reading the VS1053's output would require its audio GND to be connected to the common Esp/VS1053 GND. Something tells me that this is not a good idea (and might even cause damage). Any suggestions?

Edzelf, 2018-01-26:
You may use some 600:600 ohm transformers.  They are cheap on Aliexpress.

perlix, 2018-01-28:
Thanks, that looks like exactly what I need.