Edzelf/Esp-radio #124: A Status LED would be nice

Primus007, 2018-01-31:
Support for a status led would be nice. A blink code shows problems. like fast blinking = no connection to the wlan. slow blinking = no connection to the radio station(s) .... #

Edzelf, 2018-01-31:
You may try to connect a LED (with a 1k resistor in series) from +5 Volt to the DREQ pin.  That will give some info.

Primus007, 2018-02-01:
that's not the same but thanks anyway.
Maybe you'll want to make a blinkcode later if you have time :)

I would like to thank you at this point for your good work! Esp-radio is great and easy to use.

fredericplante, 2018-02-01:
Ain't the TFT supposed to display the status, if you use a TFT, I see no point to use a blinking LED to tel the state.