Edzelf/Esp-radio #125: Not possible to listen to the radio station

amigafan, 2018-02-03:
Suddenly it is not possible to listen to my favorite radio station: http://www.classicfm.nl/player/classicfm.m3u.
Here is a dump a console output:

D: New preset/file requested (1/0) from www.classicfm.nl/player/classicfm.m3u
D: Connect to new host www.classicfm.nl/player/classicfm.m3u
D: Playlist request, entry 1
D: Connect to www.classicfm.nl on port 80, extension /player/classicfm.m3u
D: Connected to server
D: Read from playlist
D: Playlistheader: HTTP/1.1 404 Not Found
D: Playlistheader: Server: nginx
D: Playlistheader: Date: Sat, 03 Feb 2018 18:26:09 GMT
D: Playlistheader: Content-Type: text/html
D: Playlistheader: Transfer-Encoding: chunked
D: Playlistheader: Connection: close
D: Playlistheader: Vary: Accept-Encoding
D: Playlistheader: Set-Cookie: PHPSESSID=gsosaqf9iq07gfiio93d4ocju0; path=/
D: Playlistheader: Expires: Thu, 19 Nov 1981 08:52:00 GMT
D: Playlistheader: Cache-Control: no-store, no-cache, must-revalidate, post-check=0, pre-check=0
D: Playlistheader: Pragma: no-cache
D: Playlistheader: 
D: Switch to PLAYLISTDATA
D: Playlistdata: 1ef8
D: Playlistdata: 1ef8<!DOCTYPE html><html lang="nl">
D: Entry 1 in playlist found: 1ef8<!DOCTYPE html><html lang="nl">
D: Playlistdata: <head>
D: Entry 2 in playlist found: <head>
D: Stopping client
D: Connect to new host <head>
D: Connect to  on port 80, extension /
D: Request <head> failed!
D: Switch to HEADER

For some reason, Content-Type is text / html
I hope some can help. Thanks in advance

Edzelf, 2018-02-03:
404 not found is a server error.  You have to look for a different server hosting classicfm.

amigafan, 2018-02-03:
Oddly, it's a link from their own website

ampecs, 2018-05-17:
http://19393.live.streamtheworld.com/CLASSICFM.mp3
I extracted this link from the web page player with web developer tools in chrome . Give it a try