Edzelf/Esp-radio #128: Other VS1053 Modules possible?

Primus007, 2018-02-14:
The blue VS1053 Module from LC Technology is shown in the hardware parts list. But are other Modules possible and tested? What is with the geetech vs1053 http://www.geeetech.com/wiki/index.php/VS1053_MP3_breakout_board_with_SD_card
??

This board has a CS pin, where to connect this? And where to connect the 8k1 PullUp ?
I am using the minimal version with an Wemos D1 mini.

thank you.

Edzelf, 2018-02-14:
I guess it will work with this board.
There is a CS and a XCS pin on this board.  I think the XCS is for the VS1053 and the CS is for the SD-card.
The 8k2 resistor is a pull-up for the XRST line of the VS12053.  So connect it to the XRESET of the VS1053 and +3.3V or + 5V.

Primus007, 2018-02-16:
thank you. i try this if i receive the board.