Edzelf/Esp-radio #129: Stop playing after command/change

dupcakgmailcom, 2018-02-24:
Hi, it's great project!

It plays well right after reset. 
But it will stop when I change something (vol+-, station).

It will continue playing after disconnect and connect RST wire from VS1053. (probably reseting)
I've checked wiring multiple times. I tried to connect VS1053 RST to ESP8266 RST (WemosD1) or to 5V - it doesn't matter.

I'm aware of one change - I had to change following code:
from system_update_cpu_freq ( 160 ) ; to system_update_cpu_freq ( 80 ) ;   
If it was set to 160 then tho sound was disruptive. For 80MHz is good.
   
This is part of debug output:
D: Reset VS1053...
D: End reset VS1053...
D: Slow SPI,Testing VS1053 read/write registers...
D: Fast SPI, Testing VS1053 read/write registers again...
D: endFillByte is 88

Could you advise ?
Thanks

Edzelf, 2018-02-24:
Looks like there is something wrong with the web interface.  Could you try to enter some commands like "upvolume=2" on the serial interface in the IDE?  Use the "Enter" key after each command.
Could you also try a different (stronger) powersupply?

RobertSVK, 2018-02-24:
I appreciate your quick response.

I've tried different stronger powersupply and also command via serial interface - same behaviour :(
I believe it's not related to web GUI.

I don't have "reset circuit". I've just connected V1053 RST to ESP RST or 5V or via 10k resistor to 5V.
Could missing "reset circuit" affect functionality ?
I've read in different thread that it's not really necessary. 

Also I've tried my VS1053 board in different project without problems. So I believe my hardware is ok.

Edzelf, 2018-02-24:
The reset circuit is not critical.  Are there unusual lines in the logging when you change the volume?

RobertSVK, 2018-02-24:
I don't think so. It looks normal for me 

D: Starting ESP Version Wed, 31 May 2017 12:35:00 GMT...  Free memory 13152
D: Sketch size 348080, free size 700416
D: Reset VS1053...
D: End reset VS1053...
D: Slow SPI,Testing VS1053 read/write registers...
D: Fast SPI, Testing VS1053 read/write registers again...
D: endFillByte is 88
...
D: FileRequest received /index.html
D: Response sent
D: list request from browser
D: FileRequest received /radio.css
D: Response sent
D: FileRequest received /favicon.ico
D: Response sent
**D: Command: upvolume with parameter 2** 

> then I need to diconnect/connect VS1053 RST  manually 

D: Command: upvolume with parameter 2 

> diconnect/connect VS1053 RST

D: Volume is now 34  

> same after command via serial - diconnect/connect VS1053 RST

D: Metadata block 64 bytes
D: Streamtitle found, 56 bytes
D: StreamTitle='Urban Jazz Coalition - Better Days Comin'';

Edzelf, 2018-02-24:
I see something strange: The endFillByte is 88.  Normally this would be 00.  Are you sure you chip is a VS1053?.  I know that some modules have a different chip, although it has been sold as a VS1053.  Please check this.

Edzelf, 2018-02-25:
Yes, the software is for VS1053 only.