Edzelf/Esp-radio #131: Esp-radio reveals private WiFi password in access point mode

nixnuex, 2018-03-14:
If the connection to the preset WiFis from the .ini file fails, Esp-radio opens it's own WiFi with name and password default:

https://github.com/Edzelf/Esp-radio/blob/2dbd3945f6b28e967706043dc246acba18ec195a/Esp_radio.ino#L199

https://github.com/Edzelf/Esp-radio/blob/2dbd3945f6b28e967706043dc246acba18ec195a/Esp_radio.ino#L1389

This mechanism allows an external party to log into the AP, go to the configuration page and read out the users private WiFi key.

I consider this as a serious issue as it might open up the users WiFi without him even noticing that this information was accessible (as it happened in my case). I propose to deactivate the AP mode by default or block access to the content of the .ini file if the AP mode password has not been customized by the user.

Edzelf, 2018-03-14:
In the ESP32 version of the radio, the passwords are hidden in the webinterface and the logging.  But even then it it possible to hack the ESP and read the passwords.  A solution might be to encode the password.

fredericplante, 2018-03-19:
But the bottom line question is, do you hide something that is so secret that you want it?  Worst case scenario, you reburn the thing with a diffenrant password, that all... Are you the FBI or the CIA? Do you put your banking info on the ESP8266?

nixnuex, 2018-03-20:
@fredericplante I don't know if that came across, but this is primarily about making my home's (not Esp-radio's) WiFi password accessible through Esp-radio in AP mode after it failed to connect for whatever reason. And yes, I consider this as a security risk, especially if it happens as default and even unnoticed for most users.

@Edzelf Thank you very much for your reply. I like your implementation and first level of defense on the Esp32-radio. And I agree: as long as you get access to the Esp-radio, there is a chance that you can extract all information.

So, is a non-default / user-defined AP mode password the best protection? Along with some information that your home's WiFi password might be visible from the .ini when switching to the AP mode?

Edzelf, 2018-03-20:
I added some comment of changing default network name and password in the documentation.