Edzelf/Esp-radio #135: extract of VS1053

DerMega, 2018-04-19:
Hi Edzelf,

I hope my question ist not to cheeky but, is it possible to get an extract of the VS1053 Class and the neccesary code to get Webstreams running on the ESP? :)
U seem to be the only person on the WEB who made this running on ESP and I´m very impressed.
I´m not a bad coder but this is a whole new level for me.
I have a private project for a daylight alarm clock and everithing is perfect with it except the alarm via webstream. I cant get it to work, so I am hoping you could help me. :)

thx Tobi

Edzelf, 2018-04-19:
What do you mean by "extract".  All the sourcecode for the VS1053 class is present in the sketch.  You may use it freely.

DerMega, 2018-04-19:
I had the idea that you possibly have the class and the neccesary adittional sourcecode for streaming in a single file, for testing purposes or so, wich you could provide.
If not its okay.
I already startet to seperate your class from the entire radio code. Its a lot of work and I just thought I could save some time :D
Thanks anyways, especially for sharing your code with us!

Edzelf, 2018-04-19:
I do not have a separate file.  But it should not be so hard to isolate the class.  Maybe you have to remove the dbgwrite calls.

vadpol, 2018-04-23:
Adafruit has made a vs1053 library
https://github.com/adafruit/Adafruit_VS1053_Library/archive/master.zip

DerMega, 2018-04-24:
Thx for the input vadpol. Even though I´m well advanced in extracting the lib from edzelf I´ll look at it.

perlix, 2018-04-24:
Isn't the library at https://github.com/baldram/ESP_VS1053_Library a fork of Ed's library?

There's also a basic radio example for Adafruit's library at https://learn.adafruit.com/adabox004/internet-radio. It can be changed to use the above library with very little effort.

baldram, 2018-08-01:
@perlix Exaclty, this is true. The https://github.com/baldram/ESP_VS1053_Library is fork of @Edzelf Esp-radio which contains only the code needed to drive VS1053 chip. So it's just a driver for this audio device. Probably it's exactly what @DerMega is looking for. 
PS: If you like ESP_VS1053_Library, please do not forget to click the "star" ;-)

baldram, 2018-08-03:
One more thing, the https://github.com/baldram/ESP_VS1053_Library already has an Basic Web Radio example like the one from Adafruit mentioned by @perlix . So no need to adopt it, already done. So this is exactly what @fredericplante is looking for. 
 https://github.com/baldram/ESP_VS1053_Library/tree/master/examples/WebRadioDemo