Edzelf/Esp-radio #136: wifi key default?

Nurbull, 2018-04-26:
First of all I wanted to thank you with a lot of work done.
I interested the project and decided to repeat it, as I'm new to this case, I ask you to prompt me.
The problem is that after uploading the program, the "Esp-radio" network appeared, but it is locked with a password. tell me please what password is the default or where can I look at the wikifey in the program esp-radio.ino

the start log:
```
D: FS Total 2949250, used 0
D: File /radio.ini not found!
D: * Scan Networks *
D: Number of available networks: 1
D:  1 - KN group                  Signal: -37 dBm Encryption Auto  
D: --------------------------------------
D: File /radio.ini not found, use save command to create one!
D: Starting ESP Version Wed, 18 Apr 2018 10:42:00 GMT...  Free memory 13728
D: Sketch size 355024, free size 692224
D: Reset VS1053...
D: End reset VS1053...
D: Slow SPI,Testing VS1053 read/write registers...
D: VS1053 error retry SB:0000 R1:FFFF R2:FFFF
D: VS1053 error retry SB:012C R1:FFFF R2:FFFF
D: VS1053 error retry SB:0258 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0384 R1:FFFF R2:FFFF
D: VS1053 error retry SB:04B0 R1:FFFF R2:FFFF
D: VS1053 error retry SB:05DC R1:FFFF R2:FFFF
D: VS1053 error retry SB:0708 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0834 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0960 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0A8C R1:FFFF R2:FFFF
D: VS1053 error retry SB:0BB8 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0CE4 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0E10 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0F3C R1:FFFF R2:FFFF
D: VS1053 error retry SB:1068 R1:FFFF R2:FFFF
D: VS1053 error retry SB:1194 R1:FFFF R2:FFFF
D: VS1053 error retry SB:12C0 R1:FFFF R2:FFFF
D: VS1053 error retry SB:13EC R1:FFFF R2:FFFF
D: VS1053 error retry SB:1518 R1:FFFF R2:FFFF
D: VS1053 error retry SB:1644 R1:FFFF R2:FFFF
D: Fast SPI, Testing VS1053 read/write registers again...
D: VS1053 error retry SB:0000 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0003 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0006 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0009 R1:FFFF R2:FFFF
D: VS1053 error retry SB:000C R1:FFFF R2:FFFF
D: VS1053 error retry SB:000F R1:FFFF R2:FFFF
D: VS1053 error retry SB:0012 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0015 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0018 R1:FFFF R2:FFFF
D: VS1053 error retry SB:001B R1:FFFF R2:FFFF
D: VS1053 error retry SB:001E R1:FFFF R2:FFFF
D: VS1053 error retry SB:0021 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0024 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0027 R1:FFFF R2:FFFF
D: VS1053 error retry SB:002A R1:FFFF R2:FFFF
D: VS1053 error retry SB:002D R1:FFFF R2:FFFF
D: VS1053 error retry SB:0030 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0033 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0036 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0039 R1:FFFF R2:FFFF
D: endFillByte is FF
D: Selected network: <null>                   
D: Try WiFi <null>
D: WiFi Failed!  Trying to setup AP with name Esp-radio and password Esp-radio.
D: IP = 192.168.4.1
D: Start server for commands
```

skakri, 2018-04-26:
> D: WiFi Failed!  Trying to setup AP with name Esp-radio and password Esp-radio.

Nurbull, 2018-04-26:
solved thanks

johnnyrockswell, 2020-03-29:
Hey Guys, how did you solve it? I have the same problem. I entered the wifi credentials to defaultpresets.h but it doesn't connect. thanks for your help!