Edzelf/Esp-radio #139: Succesfully added a Nextion Display to your Espradio

Manu-J, 2018-06-16:
Thanks for making your code an opensource, I just modified it to interact it with my Nextion display,

More details on my project find [Here](https://www.youtube.com/watch?v=1q8UP5HhdpM&t=36s). Once again thanks a lot.

https://www.youtube.com/watch?v=1q8UP5HhdpM&t=36s

[image]

Edzelf, 2018-06-16:
Looks good!

Manu-J, 2018-06-16:
Yes I am using the Rx tx pins of Nodemcu.I initially tried to use software serial pins but it was throwing a lot of issues..but when I used like this no issues,it can communicate with Nextion as well as with PC serial port.The only problem is that the debug output will show also the Nextion display commands. The only thing to remeber is that to make sure that the Nextion and Nodemcu baudrates are set for same value. For this project I set the baudrates as 115200 ans it worked like charm.