Edzelf/Esp-radio #140: Can not receive data from streaming radio

srspinho, 2018-07-22:
Hello Ed Smallenburg ! ! !

My name name is Sergio, and I am from Brazil.

I have just built the ESP-Radio and it worked very well. I also followed the suggestion to include the ST7735 (Pinout for TFT lcd with ST7735S controller #95) display, and it worked from the first try.

But, I am trying to listen to a Brazilian classic Rock station (Kiss FM) via the address :  

stm50.sateg.com.br:26840

When I put it on the Winamp, I am able to listen to the radio without problems : 

Network received: 420745 bytes
Server: Ultravox/2.1 SHOUTcast v2.5.1.724/posix(linux x64)
Content-Type: audio/aacp
Ultravox sync: 756 messages, 0 desyncs
Ultravox Data Message: 0x8003
Ultravox SID/AvgBR/MaxBR: 1/96000/96000
Stream name: Orban Opticodec-PC Encoder
Current title: Hi-Fi Internet Stream

But, when I try to do the same on ESP-Radio, I can not receive data : 

D: New preset/file requested (6/0) from 
D: New preset/file requested (6/0) from stm50.sateg.com.br:26840
D: Connect to new host stm50.sateg.com.br:26840
D: Connect to stm50.sateg.com.br on port 26840, extension /
D: Connected to server
D: Switch to HEADER
D: Accept-Ranges:none
D: Access-Control-Allow-Origin:*
D: Cache-Control:no-cache,no-store,must-revalidate,max-age=0
D: Connection:close
D: icy-name:Orban Opticodec-PC Encoder
D: icy-genre:Various
D: icy-br:96
D: icy-sr:22050
D: icy-url:http://www.kiissfm.com.br
D: icy-pub:1
D: content-type:audio/aacp
D: udio/aacp seen.
D: icy-metaint:32768
D: X-Clacks-Overhead:GNU Terry Pratchett
D: Switch to DATA, bitrate is 96, metaint is 32768
D: First chunk:
D: FF F9 5C 80 45 A1 E8 21
D: 1B 55 25 B9 86 42 81 20
D: E4 30 36 24 09 89 01 61
D: 20 88 4B 70 53 9B 68 CE
D: No data input

have you faced this problem or something similar to this ?

One detail : I can receive data from this service, but from another radios. 

Thank you very Much !

Sergio Pinheiro

Edzelf, 2018-07-22:
The station broadcasts an AACP stream, not MP3.  Make sure you have a VS1053 chip installed.  Other decodeder chips will not work with AACP.

srspinho, 2018-07-22:
Hello Ed !
Thank you very much for the fast reply.
Well, my MP3 board has a VS1053B chip on it (please, see the picture attached).

I was reading a pdf from Sparkfun and found the following 

**_VS1053b contains AAC technology (ISO/IEC 13818-7 and ISO/IEC 14496-3) which cannot be used
without a proper license from Via Licensing Corporation or individual patent holders.
VS1053b contains spectral band replication (SBR) and parametric stereo (PS) technologies developed by
Coding Technologies. Licensing of SBR is handled within MPEG4 through Via Licensing Corporation.
Licensing of PS is handled with Coding Technologies.
See
http://www.codingtechnologies.com/licensing/aacplus.htm
for more information.
To the best of our knowledge, if the end product does not play a specific format that otherwise would
require a customer license: MPEG 1.0/2.0 layers I and II, WMA, or AAC, the respective license should
not be required. Decoding of MPEG layers I and II are disabled by default, and WMA and AAC format
exclusion can be easily performed based on the contents of the SCI
HDAT1 register.  Also PS and SBR
decoding can be separately disabled._**

as I undestood, the AAC would be available by default. (the link wit details about AAC is not available anymore).

Or, should I enable another register in order to enable it ?

[image]

Thank You !

Sergio Pinheiro

PS. I am building the ESP-32 version too ! !