Edzelf/Esp-radio #143: Using ESp8266+VS1053 as an USB DAC

Arduinoslfan, 2018-08-23:
Great project Edzelf.
This idea came to my mind and didnt found any information about a similar project.
Is it possible to use ESP8266+VS1053 combo as an USB DAC?
I mean we connect esp8266 to pc usb port and get audio through the VS1053 headphone out.
Do you think its possible? 
Thank you

Edzelf, 2018-08-23:
I guess so, but the PC is also capable to generate analog audio.

Arduinoslfan, 2018-08-31:
@Edzelf Thank you. you are right PC can output analog audio through its inbuilt DAC. However its quality is quite average.Any idea how should I proceed ?

@fredericplante you are right. serial port needs to handle 1 million plus baud rate. But esp8266 seems to be capable of that. Sadly my knowledge is quite limited about this area. So needs to do more research. Thank you.

Edzelf, 2018-08-31:
Why using a serial port? In combination with an ESP8266 WiFi is a better option.
In combination with an STM32, the USB port may be the better way.

fredericplante, 2018-08-31:
Oh you mean not using the ESP8266 at all if your connected on the USB.