Edzelf/Esp-radio #144: WiFI password not working , SAVE option not working

radegast-rado, 2018-08-24:
Hi ED

I have few issues with my build.

1 - Wifi password is not recognized - only works with open WIFI without password.
question is where to put password for my WIFI?
is it inside DATA.INI file 
wifi_00 = SSID/PASSWORD
I am not sure if this is correct.

2 - Everytime when I try to SAVE changes on DATA.INI file via webserver - everything stop, also when click on Restart button everything stop working.
how can I restart my radio without uploading sketch again and again?

BTW - brilliant project !!!

Cheers 

RAD

Edzelf, 2018-08-24:
1 - this method (wifi_00=ssid/password) is correct.
2 - the SPIFFS filesystem is not very reliable.  You may try to add an extra dummy file of 100 kB or so to the SPIFFS.  The ini-file will be stored in a different part of the SPIFFS.  That will help if you are lucky.  Be sure that your power supply is powerful enough.
The best solution is to switch to the ESP32 version.

radegast-rado, 2018-09-05:
I have noticed that reloading DATA.INI after sketch fixed the issue - unfortunately updating DATA file from webserver cause glitches.
Reset issue only occurs when USB cable is connected to computer - on external power everything run ok.

I am about to change esp8266 to esp32 as suggested.

Thanks ED for all your help.

I will keep you updated about ESP32 build - all the best!

closed 2018-09-05 by radegast-rado