Edzelf/Esp-radio #150: How to install the ESP-Radio to ESP32 board?

wieb18, 2018-11-24:
Hello,

I am very very new to Arduino, just few days ago bought new ESP32 and VS1053 boards.

[image]

[image]

Try installing ESP-Radio but never passed Arduino Compiler (although already choosed DUMMYTFT, didn't buy any LCD yet).

[image]

**The error message:**

```
Arduino: 1.8.7 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default, QIO, 80MHz, 4MB (32Mb), 115200, Info"

In file included from C:\Users\wieb\Documents\Arduino\libraries\SoftwareSerial-master/avr/interrupt.h:38:0,

                 from C:\Users\wieb\Documents\Arduino\libraries\SoftwareSerial-master\SoftwareSerial.cpp:41:

C:\Users\wieb\Documents\Arduino\libraries\SoftwareSerial-master/avr/io.h:635:6: warning: #warning "device type not defined" [-Wcpp]

 #    warning "device type not defined"

      ^

In file included from C:\Users\wieb\Documents\Arduino\libraries\SoftwareSerial-master\SoftwareSerial.cpp:43:0:

C:\Users\wieb\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\cores\esp32/Arduino.h:83:0: warning: "sei" redefined

 #define sei()

 ^

In file included from C:\Users\wieb\Documents\Arduino\libraries\SoftwareSerial-master\SoftwareSerial.cpp:41:0:

C:\Users\wieb\Documents\Arduino\libraries\SoftwareSerial-master/avr/interrupt.h:81:0: note: this is the location of the previous definition

 # define sei()  __asm__ __volatile__ ("sei" ::: "memory")

 ^

In file included from C:\Users\wieb\Documents\Arduino\libraries\SoftwareSerial-master\SoftwareSerial.cpp:43:0:

C:\Users\wieb\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\cores\esp32/Arduino.h:84:0: warning: "cli" redefined

 #define cli()

 ^

In file included from C:\Users\wieb\Documents\Arduino\libraries\SoftwareSerial-master\SoftwareSerial.cpp:41:0:

C:\Users\wieb\Documents\Arduino\libraries\SoftwareSerial-master/avr/interrupt.h:99:0: note: this is the location of the previous definition

 # define cli()  __asm__ __volatile__ ("cli" ::: "memory")

 ^

In file included from C:\Users\wieb\Documents\Arduino\libraries\SoftwareSerial-master\SoftwareSerial.cpp:43:0:

C:\Users\wieb\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\cores\esp32/Arduino.h:106:0: warning: "_BV" redefined

 #define _BV(b) (1UL << (b))

 ^

In file included from C:\Users\wieb\Documents\Arduino\libraries\SoftwareSerial-master/avr/io.h:99:0,

                 from C:\Users\wieb\Documents\Arduino\libraries\SoftwareSerial-master/avr/interrupt.h:38,
[... 504 lines cut ...]

C:\Users\wieb\Documents\Arduino\libraries\SoftwareSerial-master\SoftwareSerial.cpp:590:26: error: 'XMIT_START_ADJUSTMENT' was not declared in this scope

   tunedDelay(_tx_delay + XMIT_START_ADJUSTMENT);

                          ^

Multiple libraries were found for "SD.h"
 Used: C:\Users\wieb\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\SD
 Not used: C:\Program Files (x86)\Arduino\libraries\SDx
exit status 1
Error compiling for board ESP32 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

This ESP-Radio using tons of library and I managed fill all the missing library files one by one. But the last compiler error above , I don't know how to solve it.

**How to install the ESP-Radio to my ESP32 board?** 

Maybe my question is very dumb because I could see a lot people done installing without any issues. But I am sure there is many people like me who are very very new to Arduino and has the same issue like me.

Any help very appreciated and thanks in advance.

Edzelf, 2018-11-24:
All errors point to softserial.  That is not used in this program.  Remove you softserial library and try to compile a simple example like "blinky" first.

wieb18, 2018-11-25:
Edzelf... thanks a lot for the guidance, finally ESP32-Radio able correctly compiled. 

Previous error because of my mistake which include Nextion default display on the first run, then changed to DUMMYTFT on the last minute. That is why the compiler producing error because its still loaded with Nextion Display library which doesn't use anymore.

Maybe anyone experiencing issue like myself, here is the step for installing ESP32-Radio:

[image]

1. First... choose your display to prevent compiler error. If no display (like myself) then choose DUMMYTFT.

[image]

2. Load Esp32_radio_init.ino to Arduino IDE

[image]

3. Change Esp32_radio_init.ino Wifi credential to yours own Wifi network

[image]

4. Upload Esp32_radio_init.ino to ESP32

[image]

5. Finally... upload ESP32-Radio.ino to ESP32

**Here is my Serial output:**

```
D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Thu, 04 Oct 2018 07:22:32 GMT.  Free memory 233752
D: Display type is DUMMYTFT
D: Partition nvs found, 20480 bytes
D: Read 31 keys from NVS
D: pin_ir set to -1
D: pin_enc_clk set to -1
D: pin_enc_dt set to -1
D: pin_enc_sw set to -1
D: pin_tft_cs set to -1
D: pin_tft_dc set to -1
D: pin_tft_scl set to -1
D: pin_tft_sda set to -1
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to -1
D: pin_vs_cs set to -1
D: pin_vs_dcs set to -1
D: pin_vs_dreq set to -1
D: pin_shutdown set to -1
D: pin_shutdownx set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is HIGH
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
[... 88 lines cut ...]
D: New preset/file requested (7/0) from 205.164.62.15:10032
D: Connect to new host 205.164.62.15:10032
D: Connect to 205.164.62.15 on port 10032, extension /
D: Connected to server
D: STOP requested
D: Stopping client
D: New preset/file requested (8/0) from skonto.ls.lv:8002/mp3
```

But now I have a new issue:  **VS1053 not properly installed!**

Here is my wiring (following ESP32-Radio.ino guidance):

[image]

WIRING:
1.  SCK to GPIO 18
2.  MISO to GPIO 19
3.  MOSI to GPIO 23
4.  XRESET to RESET
5.  XDCS to GPIO 32
6.  XCS to GPIO 5
7.  DREG to GPIO 4

Please help...

ybourdon, 2019-03-06:
Hello,
This is a nice job !
Finaly were you able to make esp-Radio work with ESP-32 CPU ?
Regards.

Yves

Edzelf, 2019-03-06:
This issue is on the wrong repository (for ESP8266 version).  The ESP32 version has its own repository. See https://github.com/Edzelf/ESP32-Radio.