Edzelf/Esp-radio #156: Play music from cloud storage

FreeElement, 2019-01-24:
I would like to see in your project an analogue of the functionality: Music Player for Google Drive.
To play music from the cloud, flash cards will no longer be needed in the future.
Or if it takes a lot of your time, tell me where to fix the code, I'll take it on myself and for this I will learn the programming language.

Edzelf, 2019-01-24:
Main problem is: how to read from a google drive without having Windows.
It is probably easier to install a shoutcast server on a Windows machine that reads from the Google drive.

FreeElement, 2019-01-24:
There is a plugin MyClouds to listen to music from the clouds on the player AIMP:
http://www.aimp.ru/forum/index.php?topic=58962.0

I asked the developer about the principle of its action.

I quote his answer: "in the case of MP3 you can listen to it as a regular Internet radio station. The difference between clouds and Internet radio stations is that the link to the file is temporary and it does not live long. In order to get it, you need to log in."
I am not familiar with the technology of data transmission over networks and this information has not clarified much for me. 
But if the analogy: AIMP and ESP-radio it players, differs only in the software environment code execution. AIMP somehow plays files from the cloud. Maybe it's all about Windows...

I understand correctly-ESP-radio (ESP8266) is a client and makes a request to the streaming server, and that on request already gives the mp3 stream?

Edzelf, 2019-01-24:
Yes the server delivers a stream or a file.  The contents is the same.  So if you can read from a Google drive in a microcontroller environment, it will work.