Edzelf/Esp-radio #157: D: /radio.ini - 1987

netrosec, 2019-01-29:
I'm testing with the VS1053 and a wemos nodemcu without a tft screen but i included the tft library in the sketch. I get this error immediately after uploading i dont know what it means. 
```
D: FS Total 2949250, used 2259
D: /radio.ini                       -    1987
D: Added SSID 00 = (:) to acceptable networks
D: Added SSID 01 = ADSL-11 to acceptable networks
D: * Scan Networks *
D: Number of available networks: 1
D:  1 - (:)                       Signal: -40 dBm Encryption WPA2  Acceptable
D: --------------------------------------
D: Command: mqttbroker with parameter broker.hivemq.com
D: Command: mqttport with parameter 1883
D: Command: mqttuser with parameter none
D: Command: mqttpasswd with parameter none
D: Command: mqtttopic with parameter espradio
D: Command: mqttpubtopic with parameter espradioIP
D: Command: wifi_00 with parameter (:)/switchfoot
D: Command: wifi_01 with parameter ADSL-11/yyyyyyy
D: Command: volume with parameter 72
D: Command: toneha with parameter 0
D: Command: tonehf with parameter 0
D: Command: tonela with parameter 0
D: Command: tonelf with parameter 0
D: Command: preset with parameter 6
D: Command: preset_00 with parameter 109.206.96.34:8100
D: Command: preset_01 with parameter airspectrum.cdnstream1.com:8114/1648_128
D: Command: preset_02 with parameter us2.internet-radio.com:8050
D: Command: preset_03 with parameter airspectrum.cdnstream1.com:8000/1261_192
D: Command: preset_04 with parameter airspectrum.cdnstream1.com:8008/1604_128
D: Command: preset_05 with parameter us1.internet-radio.com:8105
D: Command: preset_06 with parameter icecast.omroep.nl:80/radio1-bb-mp3
D: Command: preset_07 with parameter 205.164.62.15:10032
D: Command: preset_08 with parameter skonto.ls.lv:8002/mp3
D: Command: preset_09 with parameter 94.23.66.155:8106
D: Command: preset_10 with parameter ihr/IHR_IEDM
D: Command: preset_11 with parameter ihr/IHR_TRAN
D: Command: preset_12 with parameter ihr/WQTR
D: Command: preset_13 with parameter ihr/CKMXAM
D: FS Total 957314, used 2259
D: /radio.ini                       -    1987
D: Added SSID 00 = (:) to acceptable networks
D: Added SSID 01 = ADSL-11 to acceptable networks
D: * Scan Networks *

```

netrosec, 2019-01-29:
I have a 5v 2amp power bank. But the esp8266 is connected to my computer.

Edzelf, 2019-01-29:
Better connect it to the powerbank.

netrosec, 2019-01-29:
@edzelf I did but nothing changed. The nodemcu light pulses about every second a network search shows esp radio is not on my network.

Edzelf, 2019-01-29:
Maybe it has something to do with the weird name of the network.

netrosec, 2019-01-29:
Unlikely. The network name worked fine for the esp32 radio project. Maybe i'll stick with esp32 radio.