Edzelf/Esp-radio #159: Getting errors: VS1053 SPI error. SB:15EA R1:15E7 R2:15E7 ....

AaronNGray, 2019-02-10:
I am getting the following errors I am suspecting the VS1053 board is faulty ?

D: Slow SPI, Testing VS1053 read/write registers...
D: Fast SPI, Testing VS1053 read/write registers again...
D: VS1053 SPI error. SB:15EA R1:15E7 R2:15E7
D: VS1053 SPI error. SB:1ED8 R1:1ED5 R2:1ED5
D: VS1053 SPI error. SB:2A36 R1:2A33 R2:2A33
D: VS1053 SPI error. SB:2B6E R1:0000 R2:0000
D: VS1053 SPI error. SB:2B71 R1:FF00 R2:0000
D: VS1053 SPI error. SB:2B74 R1:FE00 R2:0000
D: VS1053 SPI error. SB:2B77 R1:FF00 R2:0000
D: VS1053 SPI error. SB:2B7A R1:FF00 R2:0000
D: VS1053 SPI error. SB:2B7D R1:FE00 R2:0000
D: VS1053 SPI error. SB:2B80 R1:FF00 R2:0000
D: VS1053 SPI error. SB:2B83 R1:FF00 R2:0000
D: VS1053 SPI error. SB:2B86 R1:FF00 R2:0000
D: VS1053 SPI error. SB:2B89 R1:FF00 R2:0000
D: VS1053 SPI error. SB:2B8C R1:FF00 R2:0000
D: VS1053 SPI error. SB:2B8F R1:FF00 R2:0000
D: VS1053 SPI error. SB:2B92 R1:FE00 R2:0000
D: VS1053 SPI error. SB:2B95 R1:FF00 R2:0000
D: VS1053 SPI error. SB:2B98 R1:FF00 R2:0000
D: VS1053 SPI error. SB:2B9B R1:FF00 R2:0000
D: VS1053 SPI error. SB:2B9E R1:FF00 R2:0000

I have the following wiring :-

|ESP32|Colour|VS1053|Rotary
|-----|------|------|--
|EN   |Blue  |XRST  |
|32   |Grey  |XDCS  |
|5    |White |XCS   |
|4    |Purple|DREQ  |
|18   |Yellow|SCK   |
|19   |Orange|MISO  |
|23   |Green |MOSI  |
|25   |White |      |CLK
|26   |Grey  |      |DT
|27   |Purple|      |SW

Edzelf, 2019-02-10:
The slow SPI test seems to work.  Probably a wiring problem or a wek powersupply.

AaronNGray, 2019-02-10:
AFAICS the wiring is fine. I am using Two USB connections from the same computer one for power and the other for serial connection with the grounds connected. I have seemed to been having what looks like RF field problems to some degree when I was first trying to get it to work.

It did not work at first after setting the WIFI SSID and Password then I did a reset and it sprung into action working for several hours. Then stopped and it was late so I presumed it was a network issue and packed in for the night. Next morning there was no audio but the serial output looked totally correct.

I am now with another ESP32 board and a new VS1053 board thats now giving the above error with the following output but no sound.

D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Tue, 05 Jan 2019 19:48:00 GMT.  Free memory 291184
D: Display type is BLUETFT
D: Partition nvs found, 20480 bytes
D: Read 46 keys from NVS
D: pin_ir set to 35
D: pin_enc_clk set to 25
D: pin_enc_dt set to 26
D: pin_enc_sw set to 27
D: pin_tft_cs set to 15
D: pin_tft_dc set to 2
D: pin_tft_scl set to -1
D: pin_tft_sda set to -1
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to 21
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 32
D: pin_vs_dreq set to 4
D: pin_shutdown set to -1
D: pin_shutdownx set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is LOW, probably no PULL-UP
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is LOW, probably no PULL-UP
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: GPIO39 is LOW, probably no PULL-UP
D: gpio_00 will execute uppreset = 1
D: gpio_12 will execute upvolume = 2
D: gpio_13 will execute downvolume = 2
D: gpio_14 will execute stop
D: gpio_17 will execute resume
D: gpio_34 will execute station = icecast.omroep.nl:80/radio1-bb-mp3
D: Enable pin 35 for IR
D: Start display
D: SD Card Mount Failed!
D: Create list with acceptable WiFi networks
D: Added TP-LINK_9789 to list of networks
D: Added SSID2 to list of networks
D: End adding networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 11
D:  1 - FreeBeer                  Signal: -55 dBm, Encryption WPA_WPA2_PSK, 
D:  2 - TP-LINK_9789              Signal: -57 dBm, Encryption WPA_WPA2_PSK, Acceptable
[... 95 lines cut ...]
D: Headerline: Content-Type: audio/mpeg
D: audio/mpeg seen.
D: Headerline: icy-br:192
D: Headerline: ice-audio-info: samplerate=48000;channels=2;bitrate=192
D: Headerline: icy-br:192
D: Headerline: icy-genre:Talk
D: Headerline: icy-metadata:1
D: Headerline: icy-name:NPO Radio1
D: Headerline: icy-pub:0
D: Headerline: icy-url:http://www.radio1.nl
D: Headerline: Server: Icecast 2.4.0-kh10
D: Headerline: Cache-Control: no-cache, no-store
D: Headerline: Access-Control-Allow-Origin: *
D: Headerline: Access-Control-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type
D: l-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type seen.
D: Headerline: Access-Control-Allow-Methods: GET, OPTIONS, HEAD
D: Headerline: Connection: Close
D: Headerline: Expires: Mon, 26 Jul 1997 05:00:00 GMT
D: Headerline: icy-metaint:16000
D: Switch to DATA, bitrate is 192, metaint is 16000
D: Duration mp3loop 71
D: Duration mp3loop 137
D: Metadata block 64 bytes
D: Streamtitle found, 52 bytes
D: StreamTitle='NPO Radio 1 - NOS Langs de Lijn - NOS';

AaronNGray, 2019-02-10:
There seems to be an nvs error in there too

```nvs_get_str failed 1102 for key basicauth, keylen is 9, len is 150!```

AaronNGray, 2019-02-10:
And I am getting 4.97 Volts.

Edzelf, 2019-02-10:
The basicauth key is new.  I will check this.
Strange thing is that gpio2 is low.
But there is still something wrong with the SPI bus.

AaronNGray, 2019-02-10:
okay I have checked the powerline with a scope and it looks solid.

AaronNGray, 2019-02-10:
GPIO2 is not connected to anything. And I have checked my breakout board and it should be floating.

AaronNGray, 2019-02-10:
I dont have another VS1053 board to substitute :)

AaronNGray, 2019-02-10:
Sorry I have posted this on the wrong repo !!!

closed 2019-04-16 by AaronNGray