Edzelf/Esp-radio #162: SWR Preset won't work

lautrarer, 2019-03-27:
Some Presets work but swr 3 won't. In winamp its normally playing.

D: New preset/file requested (-1/0) from swr-swr3-live.cast.addradio.de/swr/swr3/live/mp3/128/stream.m
D: Connect to new host swr-swr3-live.cast.addradio.de/swr/swr3/live/mp3/128/stream.mp3
D: Connect to swr-swr3-live.cast.addradio.de on port 80, extension /swr/swr3/live/mp3/128/stream.mp3
D: Connected to server
D: Switch to HEADER
D: Server: nginx
D: Content-Length: 0
D: Connection: close
D: Cache-Control: no-cache, no-store
D: Expires: Mon, 26 Jul 1997 05:00:00 GMT
D: Access-Control-Allow-Origin: *
D: Access-Control-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type
D: -Allow-Headers: Origin, Accept, X-Requested-With, Content-Type seen.
D: Access-Control-Allow-Methods: GET, OPTIONS, HEAD
D: Location: http://dg-swr-http-fra-dtag-cdn.cast.addradio.de/swr/swr3/live/mp3/128/stream.mp3
D: Switch to DATA, bitrate is 0, metaint is 0
D: No data input
D: No data input

This is all what i get

Edzelf, 2019-03-27:
You will be redirected in winamp if you use that.
Use dg-swr-http-dus-dtag-cdn.cast.addradio.de/swr/swr3/live/mp3/128/stream.mp3 instead.

lautrarer, 2019-03-27:
Works now ty

closed 2019-03-27 by lautrarer