Edzelf/Esp-radio #163: Equalizer setting in .ini file

PrimLem, 2019-04-21:
Hi,
I need to slightly compensate the frequency response (+4 dB above 1 kHz) and I tried the following settings in the radio.ini file:

toneha = 4
tonehf = 1000

and

toneha = 4 dB
tonehf = 1000 Hz

and ( as raw SCI_BASS value should be)
toneha = 8
tonehf = 1

But none of these settings seems to be properly read from the radio.ini file at boot time. However, when I set the values in the web GUI, the setting is changed, but are not saved into the radio.ini file. What could be the problem or what is the correct format fore these two/four tone settings?

Thanks,
Primoz

Edzelf, 2019-04-22:
In index.html you can find the acceptable values for the settings.  I will check later if the value are saved.

Edzelf, 2019-04-23:
Tone settings from the webinterface are not automatically saved.  Settings in the ini-file will be used to set the initial values of the tone settings.

PrimLem, 2019-04-24:
Thanks for checking how it is working. I'll edit the .ini file and use it this way.

closed 2019-04-24 by PrimLem