Edzelf/Esp-radio #170: Get error while compileing

diwo13, 2019-08-18:
Hi

i try to compile but i get following error. I don't know what to do.
Can anyone please help me?

`/home/USER/Arduino/libraries/Adafruit_Circuit_Playground/Adafruit_Circuit_Playground.h:100:3: error: 'Adafruit_CPlay_FreeTouch' does not name a type
   Adafruit_CPlay_FreeTouch     cap[7]; ///< the array of capacitive touch sensors`

I think the problem starts here. But i don't know what to do.

`
#ifdef __AVR__ // Circuit Playground 'classic'
  CPlay_CapacitiveSensor cap[8]; ///< the array of capacitive touch sensors
#else
  Adafruit_CPlay_FreeTouch     cap[7]; ///< the array of capacitive touch sensors
  IRrecvPCI              irReceiver; ///< the IR receiver object
  IRdecode               irDecoder; ///< the IR decoder object
  IRsend                 irSend; ///< the IR send object
#endif
`

I use arduino IDE 1.8.9 but I tried 1.8.5 else.

The needed libries i downloaded by from links in the head of the file ESP_RADIO.ino

Thanks for help
If you need additional informations i try give this.

Edzelf, 2019-08-18:
Did you select the right board?  It looks like an error while compiling for Atmega instead of ESP8266.

diwo13, 2019-08-18:
Hi Edzelf

thank you for your comment.

I use a WEMOS D1 mini. 
I can't select this board, but I can (and did) select a LOLIN(WEMOS) D1 R2 & mini. I also tried to select Generic ESP8266 Module.

The problem stays.

Edzelf, 2019-08-18:
Could you try to remove the adafruit-circuit-playground library?

diwo13, 2019-08-18:
I remove the complete Adafruit_Circuit_Playground folder from the libraries folder.
Now i got this message

`Esp_radio:1:41: error: Adafruit_Circuit_Playground.h: No such file or directory
 #include <Adafruit_Circuit_Playground.h>`

Edzelf, 2019-08-18:
But there is no include for this library in my sketch.  Are you sure you have the right sketch, the one with the 3129 lines?

diwo13, 2019-08-18:
In my sketch is
```
#include Adafruit_CircuitPlayground
#include Adafruit_Circuit_Playground
```

I commend them out 

```
//#include Adafruit_CircuitPlayground
//#include Adafruit_Circuit_Playground
```

now there error message while compiling and I can load the sketch

Thank you for your help

closed 2019-08-20 by diwo13