Edzelf/Esp-radio #171: I can't add 'RadioBOB' can you please help me?

diwo13, 2019-08-20:
Hi

the Radio works fine. But I have some problems to add new Radio-Stations.

For example "Radio Bob"

I got these links from Radio Bob website: [https://streams.radiobob.de/](url)
`https://streams.radiobob.de/bob-80srock/mp3-192/streams.radiobob.de/`
`https://streams.radiobob.de/bob-80srock/mp3-192/streams.radiobob.de/play.pls`
`https://streams.radiobob.de/bob-80srock/mp3-192/streams.radiobob.de/play.m3u`

I put the first link into winamp an get this: (it's the same string)

`https://streams.radiobob.de/bob-80srock/mp3-192/streams.radiobob.de/`

Unfortunately no one works in ESP-Radio.

Can you please help me?

Thanks

Edzelf, 2019-08-20:
streams.radiobob.de/bob-80srock/mp3-192/streams.radiobob.de works in my setup.
I entered it in the webinterface next to the PLAY button.
And I made a preset for it:
```
preset_012 = streams.radiobob.de/bob-80srock/mp3-192/streams.radiobob.de  # 12 - Radio Bob
```
Did you remove the "https://" prefix?

diwo13, 2019-08-20:
Hi Edzelf

Thank you for Your fast answer.

Yes I removed the "https://" 

I tried to put the string next to the play-button. No Radio Bob is playing.

I copied exacly your preset-String and put it into the radio.ini and saved it. Then I restarted the esp-radio. But in no way Radio Bob is playing here :-(

I have a PiHole working here, but deaktivated it for 5 minutes. But this is not the problem. No Radio Bob at all.

I don't find my mistake

Edzelf, 2019-08-20:
I tried this station on my ESP32 radio.  I will test it on the ESP8266 version...

Edzelf, 2019-08-20:
You are right.  It does not play on the ESP8266 version.   The URL is in fact a link to:
bob.hoerradar.de/radiobob-80srock-mp3-hq?sABC=5q5p28p9%230%23po3994944q0n932r5qp091r95r3q6059%23fgernzf.enqvbobo.qr&amsparams=playerid:streams.radiobob.de;skey:1566320841
This works on my set-up.  Without any interrupts.

diwo13, 2019-08-20:
Hi

thank you for your help.
Unfortunately it works on my side only with interrupts.

Can You try if this link `bob.hoerradar.de/radiobob-80srock-mp3-hq` works on your ESP8266? It's still your link cutted after `.... mp3-hq`

on my side it's the same effect, if use the long link you have posted, or the short link.

Edzelf, 2019-08-20:
Yes, **bob.hoerradar.de/radiobob-80srock-mp3-hq** works correctly.

diwo13, 2019-08-20:
Okay 

then I have to search for the problem in my installation. Do you have an idea, where I can start the search?

WIFI is 300 MBit
Every other Station works good
I have an 100MBit DSL-connection (Speedtest 5 minutes ago confirmed 100MBit)
The ESP has an fixed ip-adress in my network
When the sound stops an I i click the button TEST i get this message `Free memory is 14864, ringbuf 0, stream 0`
When the sound is correctly and I hit the TEST-button i get this message `Free memory is 14808, ringbuf 2085, stream 1861` (for example)

Thank you

Edzelf, 2019-08-20:
Your internet connection is much better than mine.  Under normal circumstances the ringbuf should have 19000+ bytes.  The stream should have 1000+ bytes.

diwo13, 2019-08-22:
Hi

can you take a look on that photo? 
It shows the configuration I used. Is there everything correct?

Thank you
 
[image]

Edzelf, 2019-08-22:
Configuration looks good.  Can you test it on a different WiFi network?

diwo13, 2019-08-23:
Thats a good idea.

I think I can try this tomorrow.

Thanks

diwo13, 2019-08-31:
Hi

Sorry, I've just managed to test it.
On a other WIFI th problem with RadioBOB stays on.

I checked the message from the arduino-monitor. Maybe this can help to find the error.

`
D: Command: preset with parameter 12

D: STOP requested
D: Stopping client
D: Song stopped correctly after 0 msec
D: New preset/file requested (3/0) from streams.radiobob.de/bob-80srock/mp3-192/streams.radiobob.de
D: Connect to new host streams.radiobob.de/bob-80srock/mp3-192/streams.radiobob.de
D: Connect to streams.radiobob.de on port 80, extension /bob-80srock/mp3-192/streams.radiobob.de
D: Connected to server
D: Switch to HEADER
D: Access-Control-Allow-Origin: *
D: Cache-Control: no-cache, must-revalidate
D: Content-Type: audio/mpeg
D: audio/mpeg seen.
D: Expires: Sat, 31 Aug 2019 07:41:28 GMT
D: Location: http://bob.hoerradar.de/radiobob-80srock-mp3-hq?sABC=5q6n24n8                            
D: Server: Caddy
D: Status: 302 Moved Temporarily
D: X-Powered-By: PHP/7.3.6
D: Content-Length: 0
D: Connection: close
D: Switch to DATA, bitrate is 0, metaint is 0
D: No data input
D: No data input
D: Trying other station/file...
D: STOP requested
D: Song stopped correctly after 0 msec
D: New preset/file requested (12/0) from ihr/CKMXAM
D: Connect to new iHeartRadio host: CKMXAM
D: GET /api/livestream?version=1.5&mount=CKMXAMAAC&lang=en
D: Connected!
D: XML parser processing...
D: Bad xml status-code 453
D: Closing XML connection.
D: Connect to new host 
D: Connect to  on port 80, extension /
D: Request  failed!
D: No data input
`

Thank You

diwo13, 2019-08-31:
:-)

YEESSS i git it. 
I put the string from the monitor output
`bob.hoerradar.de/radiobob-80srock-mp3-hq?sABC=5q6n24n8`
into th config file.

And it plays.......GREAT

Thank you for your help.

AND MANY THANKS FOR THIS PROJEKT. IT'S A GREAT RADIO. :+1:

diwo13, 2019-09-02:
:-(

unfortunately it is not perfect yet. every now and then there are still interruptions. but at tolerable intervals, so several minutes to hours between individual interruptions