Edzelf/Esp-radio #175: Display error

Hans8266, 2019-09-06:
I have the radio up and running (actually build two with the same problem).
But the diplay shows garbage 7 mm to the right.
I changed the TFT_ILI9163C.cpp file in line 1096 as described, but this did not change anything.
What am i missing?

Edzelf, 2019-09-06:
What display are you using?  How many pixels is 7 mm?Are you us

Edzelf, 2019-09-06:
What display are you using? How many pixels is 7 mm?

Edzelf, 2019-09-06:
What is the complete path of the TFT__ILI9163C.cpp file you had changed?  Are you sure that this one is used as a library bij IDE?

Hans8266, 2019-09-07:
I use the 1.8 SPI TFT 128x160 .
I have found the problem after reading your pdf again :-),
in the file TFT_ILI9163C_settings.h I changed line 24 from 128 to 160.
I did not find the “blue 1.8 SPI 128x160 board” setting and so changed the default __144_RED_PCB__.
Although at boot up it still shows a little garbagde to the right, but when running it is all fine.
Thank you for the support and the great application!

Bye the way, I had to change the analog values for the 3 keys on the analog port, for me this worked:
asw1 - 588, asw2 - 697, asw3 - 883. My ESP-07 has reading 1024 for 1V input voltage.
I build exactly your circuit.

clparrish, 2019-09-27:
Hi Hans,
I have the same problem with the TFT that you described. I read with interest your solution to the problem, but I don't completely understand it. The lines in my file TFT_ILI9163C_settings.h are not numbered as yours obviously are. Could you please provide more detail about your solution?
Thank you, Curtis

Hans8266, 2019-09-28:
Hi Curties.
The file does not have linenumbers, but the Editor you use may display them, I use Notepad++.
Here I show line 24 and 25:
	#define _TFTWIDTH  		160//the REAL W resolution of the TFT
	#define _TFTHEIGHT 		128//the REAL H resolution of the TFT
I still have some display errors at boot up, but this is of no importance I think.

clparrish, 2019-09-28:
Hans,
Thank you very much for the information.
Curtis