Edzelf/Esp-radio #177: Edzelf

clparrish, 2019-09-27:
Hi,
I have the radio working nicely except for the display. It shows the information correctly, but everything is shifted to the left, with about 13% of "video noise" on the right. Implementing the instructions for modifying the referenced libraries does not change anything (I have the blue 128x160 TFT board). Also, my library lines are not numbered, which makes it difficult. Could you please elaborate on the procedure to correct this problem?
Thank you, Curtis

JOliverasC, 2020-08-04:
I have a similar (or same) problem, and I solved with the modifications published in <<Pinout for TFT lcd with ST7735S controller #95>>  
I have a 1,8" TFT ST7736S SPI  160 x128, red one, and I guess that you have the same

clparrish, 2020-08-04:
Hi, thanks for your tip regarding the TFT. I will try this when I receive
my new ESP8266.
Curt